espressif/esp-adf #538: Question about LyraTD MSC V2.2's codec (ZL38063) and flash (W25Q16JVSNIQ).

utkutpcgl, 2020-12-11:
As far as I know, ZL38063's firmware is in esp-adf\components\audio_hal\driver\zl38063\firmware and it is loaded by the host (WROVER-B chip of MSC V2.2). It is also stated in the hardware reference (A):

> 5.Apply a 3.3 K? resistor from GPIO_2 to DVDD33,can set auto-boot from
>  external flash;if GPIO_2=0,Host boot(default).

And I assume that by default the AEC firmware works on the DSP (rather than ASR) as it is stated here:

> Hi @theronic the current library has firmware for AEC on the DSP. We will provide the abstraction layer on how you can flash 
> your own firmware and configuration on the DSP in the next release. 

(https://github.com/espressif/esp-va-sdk/issues/6#issuecomment-428117487)

1. What is the use of W25Q16JVSNIQ on the board? 
2. Is the AEC firmware loaded on ZL38063 the flash (W25Q16JVSNIQ)? 
3. Or is the flash empty by default, and if so is it used to switch between operation modes (AEC or ASR)?

Also, could you please correct me, if I am wrong on my previous assumptions.

Kind regards.

HengYongChao, 2020-12-24:
Hi @utkutpcgl 
ZL38063 has 4 work modes,  here :https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/zl38063/example_apps/tw_spi_access.c#L38
```
* input arg: mode:  0 - load both firmware and confing
*                   1 - load firmware only
*                   2 - load config only
*                  -1 - Force loading
*/
```
And now we work on mode 0,  Everytime power on esp32 will upload firmware and config to dsp in this mode.

So DSP chip save the firmware in Flash(W25Q16JVSNIQ) is OK. DSP also has its owner ASR hot word.

utkutpcgl, 2020-12-24:
Dear @HengYongChao ,
Thanks for the answer.
Best wishes.

HengYongChao, 2020-12-25:
We will close this issue, feel free to reopen it if any question, Thanks!  @jason-mao

closed 2021-02-23 by jason-mao

liftena, 2021-03-23:
Dear @HengYongChao,
May you share information about firmware that is flashed via spi. 
We want to use device for two way communication.
We are not satisfied with its AEC performance.
Does flashed Firmware has AEC capability?
If not; how can we flash the firmware(how can we find) for two way communication?

HengYongChao, 2021-03-23:
> Dear @HengYongChao,
> May you share information about firmware that is flashed via spi.
> We want to use device for two way communication.
> We are not satisfied with its AEC performance.
> Does flashed [...]

Hi @liftena 
The default firmware does not contain AEC function, you may need to contact MSC for official support.

liftena, 2021-03-23:
Thank you @HengYongChao,
I contacted with Espressif support. 
Waiting response.
If I find solution I will share.

expresspotato, 2021-04-13:
Hi @liftena were you able to find a solution? I am also looking at the combination of ESP32 + ZL38063, but have zero documentation from Microchip Technology. No way to set AEC, Gain, Noise Supression, nothing.

Even if I buy the evaluation kit, I think that's only demo software not individual tuning / a configuration I can then write using tw_spi_access.c.

HengYongChao, 2021-04-13:
> Hi @liftena were you able to find a solution? I am also looking at the combination of ESP32 + ZL38063, but have zero documentation from Microchip Technology. No way to set AEC, Gain, Noise [...]

Please contact MSC official technical support directly, Email: perry.wang#microchip.com

liftena, 2021-04-16:
> Hi @liftena were you able to find a solution? I am also looking at the combination of ESP32 + ZL38063, but have zero documentation from Microchip Technology. No way to set AEC, Gain, Noise [...]

@expresspotato 
We haven't found exact solutions yet.
Working on it.

kristian32, 2023-04-06:
Hi, 
has anybody found a solution for this? We are also not satisfied with echo and are wondering if this is the best option to get good AEC on VoIP calls.
Best wishes!