espressif/esp-adf #540: Algorithm stream issues

michalber, 2020-12-14:
Hey,
I am currently working in implementing in my own solution an AEC, basing on algorithm stream example. I may find few issues in that example. I am using LyraT board - stereo mics + stereo audio playback.

- First issue is that when using `ALGORITHM_STREAM_INPUT_TYPE2` and getting input ringbuffer from algorithm element, returned `ringbuffer_handle` is NULL.

- Second issue is accessing the mp3_decoder multi output ringbuffer. How can you know that it is set before? MP3 decoder is precompiled and nowhere is written that audio element's `max_rb_num` is greater than 0.

> https://github.com/espressif/esp-adf/blob/master/components/audio_pipeline/audio_element.c#L1305

Code snippet according to these issues:

```
ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
audio_element_set_multi_output_ringbuf(mp3_decoder, input_rb, 0);
```

> https://github.com/espressif/esp-adf/blob/master/examples/advanced_examples/algorithm/main/algorithm_examples.c#L133

- Third issue is changing i2s clock when new audio is playing. You are changing algorithm element's record and reference rate but `i2s_stream` clk is only set to writer's element. Will it affect also i2s recording element? (by default it is set to 44.1kHz) If not i2s is recording with 44.1kHz but algorithm element will be set that it is recording with `music.info.sample_rates` rate (eg. 16kHz) 

```

 i2s_stream_set_clk(i2s_stream_writer, music_info.sample_rates, music_info.bits, music_info.channels);

#if defined CONFIG_ESP_LYRAT_V4_3_BOARD
            algo_stream_set_record_rate(element_algo, music_info.channels, music_info.sample_rates);
            algo_stream_set_reference_rate(element_algo, music_info.channels, music_info.sample_rates);
#endif
```

> https://github.com/espressif/esp-adf/blob/master/examples/advanced_examples/algorithm/main/algorithm_examples.c#L183

Thank you to clarify
Best regards
Michael

jason-mao, 2020-12-15:
@michalber Hi Michael,

> First issue is that when using ALGORITHM_STREAM_INPUT_TYPE2 and getting input ringbuffer from algorithm element, returned ringbuffer_handle is NULL.
Please confirm the `algo_stream_init`  parameter `config.input_type = ALGORITHM_STREAM_INPUT_TYPE2`.
https://github.com/espressif/esp-adf/blob/e4a1c52dab7a951ec4894d8aa082e831a3fcd608/components/audio_stream/algorithm_stream.c#L601

If yes, please add more debug message at https://github.com/espressif/esp-adf/blob/e4a1c52dab7a951ec4894d8aa082e831a3fcd608/components/audio_pipeline/audio_element.c#L942.

> Second issue is accessing the mp3_decoder multi output ringbuffer. How can you know that it is set before? MP3 decoder is precompiled and nowhere is written that audio element's max_rb_num is greater than 0.

I suggest that you can use the i2s_stream multi output, not mp3_decoder.

> Third issue is changing i2s clock when new audio is playing. You are changing algorithm element's record and reference rate but i2s_stream clk is only set to writer's element. Will it affect also [...]

The LYRAT_V4_3_BOARD use one I2S0 in full duplex，recoding and playback have same rate and channels. So we just modify the `i2s_stream_writer` CLK.

michalber, 2020-12-15:
@jason-mao Thank you for reply

I have made few changes to example to find what is happening with the `ringbuffer_handle_t`. I added few debug lines to `audio_element.c` file with ESP_LOGW (because LOGI or LOGD does not work for some reason). First I changed the algorithm stream configuration to this:

```
    algorithm_stream_cfg_t algo_config = ALGORITHM_STREAM_CFG_DEFAULT();
#if defined CONFIG_ESP_LYRAT_MINI_V1_1_BOARD
    algo_config.input_type = ALGORITHM_STREAM_INPUT_TYPE1;
#elif defined CONFIG_ESP_LYRAT_V4_3_BOARD
    algo_config.input_type = ALGORITHM_STREAM_INPUT_TYPE2;
    algo_config.rec_linear_factor = 1;
    algo_config.ref_linear_factor = 1;
    algo_config.rec_ch = 2;
    algo_config.ref_ch = 2;
    algo_config.rec_sample_rate = 16000;
    algo_config.ref_sample_rate = 16000;
    algo_config.algo_mask = (ALGORITHM_STREAM_USE_AEC | ALGORITHM_STREAM_USE_AGC | ALGORITHM_STREAM_USE_NS);
#else    
    ESP_LOGE(TAG, "[ * ] Do not support current board");
    return;
#endif
    element_algo = algo_stream_init(&algo_config);
```
 In logs I receieved:
```
W (696) ALGORITHM_STREAM: Algorithm stream multi in RB num: 1
W (700) AUDIO_ELEMENT: Allocate multi-in RB of num 1
```

> I suggest that you can use the i2s_stream multi output, not mp3_decoder.

Also I have changed I2S Stream configuration to:
```
    ESP_LOGI(TAG, "[4.3] Create i2s stream to write data to codec chip");
    i2s_stream_cfg_t i2s_wd_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_wd_cfg.type = AUDIO_STREAM_WRITER;
    i2s_wd_cfg.i2s_config.sample_rate = 16000;
    i2s_wd_cfg.multi_out_num = 1;
    i2s_stream_writer = i2s_stream_init(&i2s_wd_cfg);
```

And in logs i receieved:
```
I (771) ALGORITHM_EXAMPLES: [4.3] Create i2s stream to write data to codec chip
W (779) I2S: I2S driver already installed
W (784) AUDIO_ELEMENT: Allocate multi-out RB of num 1
```

Getting algorithm stream multi input RB and setting it to I2S Stream looks like this:
```
#if defined CONFIG_ESP_LYRAT_V4_3_BOARD
    //Please reference the way of ALGORITHM_STREAM_INPUT_TYPE2 in "algorithm_stream.h"
    ESP_LOGI(TAG, "Get algorithm stream multi input RB");
    ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
    if(input_rb == NULL)
    {
        ESP_LOGE(TAG, "Cannot get multi input RB from algorithm stream");
    }
    ESP_LOGI(TAG, "Set new multi output RB");
    if(audio_element_set_multi_output_ringbuf(i2s_stream_writer, input_rb, 0) != ESP_OK)
    {
        ESP_LOGE(TAG, "Cannot set multi output RB to i2s writer element");
    }
#endif
```

And logs:
```
I (819) ALGORITHM_EXAMPLES: Get algorithm stream multi input RB
W (826) AUDIO_ELEMENT: audio_element_get_multi_input_ringbuf RB max num: 1
W (833) AUDIO_ELEMENT: Requested index 0 is smaller that max_rb_num. Return RB handle
E (842) ALGORITHM_EXAMPLES: Cannot get multi input RB from algorithm stream
I (849) ALGORITHM_EXAMPLES: Set new multi output RB
W (855) AUDIO_ELEMENT: audio_element_set_multi_output_ringbuf RB max num: 1
W (863) AUDIO_ELEMENT: New output RB is NULL
E (867) ALGORITHM_EXAMPLES: Cannot set multi output RB to i2s writer element
```

I found that the problem here is getting null-pointing `ringbuffer_handle_t` from `algo_stream_get_multi_input_rb` but it is previously successfully initialized in
https://github.com/espressif/esp-adf/blob/e4a1c52dab7a951ec4894d8aa082e831a3fcd608/components/audio_pipeline/audio_element.c#L945
Do I need to get `ringbuffer_handle_t` and init it manually? (something like this). In that case it seems to work.

```
    ESP_LOGI(TAG, "Get algorithm stream multi input RB");
    ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
    input_rb = rb_create(2, 512 * 8);
    if(input_rb == NULL)
    {
        ESP_LOGE(TAG, "Cannot get multi input RB from algorithm stream");
    }
    ESP_LOGI(TAG, "Set new multi output RB");
    if(audio_element_set_multi_output_ringbuf(mp3_decoder, input_rb, 0) != ESP_OK)
    {
        ESP_LOGE(TAG, "Cannot set multi output RB to i2s writer element");
    }
```

closed 2021-04-29 by michalber