espressif/esp-adf #543: ESP adf - How to access an mp3 file from sd card via spi mode?

sudheeshsmadhav, 2020-12-16:
Working on a project that uses mp3 playback from sd card. esp-idf for file operation works with sd card spi interface. SD 1line and 4 line modes works well with mp3 playback from sd card but not spi. I am using ESP32 Wroom. Wrover also available.

HengYongChao, 2020-12-22:
The example https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_sdcard_mp3 is you wanted. But I think you  focus is on spi interface fatfs file system.

ADF amost example use SDIO as interface, I suggest that you put spi mode fatfs in IDF as a component in your ADF project .

HengYongChao, 2021-02-23:
@sudheeshsmadhav 

SDIO  -->  SPI , YOU only need change the sdcard init mode like this:

```
    // audio_board_sdcard_init(set, SD_MODE_1_LINE);
    audio_board_sdcard_init(set, SD_MODE_SPI);
```

periph sdcard has support 3 modes:
```
typedef enum {
    SD_MODE_SPI    = 0x0,  /*!< sd_card SPI*/
    SD_MODE_1_LINE = 0x1,  /*!< sd_card 1-line SD mode*/
    SD_MODE_4_LINE = 0x2,  /*!< sd_card 4-line SD mode*/
    SD_MODE_MAX,
} periph_sdcard_mode_t;
```

jason-mao, 2021-02-23:
Closed by duplicate issue #552

closed 2021-02-23 by jason-mao