espressif/esp-adf #544: Memory leak in periph_sdcard.c

raldone01, 2020-12-16:
I marked the two lines that leak memory. I didn't find a function that cleans up the memory allocated by esp_periph_create. esp_periph_destroy only calls the destroy function but doesn't clean up the memory allocated by esp_periph_create.
periph_sdcard.c:
```c
esp_periph_handle_t periph_sdcard_init(periph_sdcard_cfg_t *sdcard_cfg)
{
    esp_periph_handle_t periph = esp_periph_create(PERIPH_ID_SDCARD, "periph_sdcard");
    AUDIO_MEM_CHECK(TAG, periph, return NULL);

    periph_sdcard_t *sdcard = audio_calloc(1, sizeof(periph_sdcard_t));
    AUDIO_MEM_CHECK(TAG, sdcard, return NULL); // THIS LEAKS the memory allocated by esp_periph_create
    if (sdcard_cfg->root) {
        sdcard->root = audio_strdup(sdcard_cfg->root);
    } else {
        sdcard->root = audio_strdup("/sdcard");
    }
    AUDIO_MEM_CHECK(TAG, sdcard->root, {  // THIS LEAKS the memory allocated by esp_periph_create
        audio_free(sdcard);
        return NULL;
    });

    sdcard->card_detect_pin = sdcard_cfg->card_detect_pin;
    sdcard->sd_mode = sdcard_cfg->mode;
    esp_periph_set_data(periph, sdcard);
    esp_periph_set_function(periph, _sdcard_init, _sdcard_run, _sdcard_destroy);
    return periph;
}
```

HengYongChao, 2020-12-17:
@raldone01 
The first area of defensive code should look like this:
`    AUDIO_MEM_CHECK(TAG, sdcard, {
        audio_free(periph);
        return NULL;
    });
`
The second :
`    AUDIO_MEM_CHECK(TAG, sdcard->root, {
        audio_free(sdcard);
        audio_free(periph);
        return NULL;
    });
`

raldone01, 2020-12-17:
I didn't check the other peripherals for similar errors. (Shouldn't you provide a function instead of audio_free(periph) if an additional allocation is needed sometime in the future. Maybe document how to free the the peripherals.) 

Will you fix this is issue in the sdcard peripheral and possibly in the other peripherals too?

closed 2020-12-22 by espressif-bot