espressif/esp-adf #545: bluetooth_service_destroy() usage

iloop2020, 2020-12-16:
Hi ,

What is the correct way to disable/turn off bluetooth A2DP, even esp32 is connected to SmartPhone.

Thank you.
Best Regards,
Leo

HengYongChao, 2020-12-24:
@iloop2020  Are you having any problems using the bluetooth_service_destroy()  API?
Normally, the service needs to be stopped before it is destroyed.

iloop2020, 2020-12-24:
@HengYongChao Thank you for your support.

And yes, when Bluetooth connection is established, calling destroy () API can't terminate the connection, and the host still steaming audio to ESP32.

Would you please give me some direction ?
It is urgent.

Thank you.

HengYongChao, 2020-12-24:
> @HengYongChao Thank you for your support.
> 
> And yes, when Bluetooth connection is established, calling destroy () API can't terminate the connection, and the host still steaming audio to ESP32.
> 
> Would you please give me some direction ?
> It is urgent.
> 
> Thank you.

What example do you use?   If you just want to disconnect the a2dp , call app_bt_stop() will be ok.

Here's example code, which extracted from the DU1906 example:

```
        if (strstr((char *)asr_result->buffer, "打开") && strstr((char *)asr_result->buffer, "蓝牙")) {
            app_bt_start();
            bdsc_engine_open_bt();
            return BDSC_CUSTOM_DESIRE_DEFAULT;
        }
        /* 通过语音控制蓝牙关闭 */
        if (strstr((char *)asr_result->buffer, "关闭") && strstr((char *)asr_result->buffer, "蓝牙")) {
            bdsc_engine_close_bt();
            app_bt_stop();
            return BDSC_CUSTOM_DESIRE_DEFAULT;
        }
```

iloop2020, 2020-12-24:
Thank you for your support.

We are using ADF and the example is player/pipeline_bt_sink/main/play_bt_music_example.c.
Actually, we are using bluetooth_service API, For example "bluetooth_service_start".

it seems "app_bt_stop()" is included in bluetooth service library,
I am not sure if we bypass bluetooth service library, and call "app_bt_stop()" directly, is it allowed ?

Thank you