espressif/esp-adf #546: VOIP algorithm_stream I2Smulti_output timing, AEC not working

Patsjemoe, 2020-12-17:
Hi,
I have defined : audio_element_set_multi_output_ringbuf(i2s_stream_writer, input_rb, 0); which I put into ref_ch  together with the recording I2S signal into the rec_ch of Algorithm_stream. Both inputs are at 48000 going to 12000 in the algo.
No AEC is performed. Echo still there.
Since the path to ref_ch is shorter then rec_ch, I wonder how to align both signals to cancel out. Can I delay the ref_ch to match the input , or how is this done ?
thanks in advance
best regards
Ludo

OctopusHu0105, 2021-01-19:
@Patsjemoe 
please add 
"cfg.out_rb_size =  ****** " before "https://github.com/espressif/esp-adf/blob/390ef2918271572237f7ff4028d0e542b68fb707/components/bluetooth_service/a2dp_stream.c#L237"

note: out_rb_size default 8kb.

bernardoarodrigues, 2022-02-10:
Hi @Patsjemoe, have you succeeded implementing AEC in VOIP? I'm trying to achieve it in a lyrat-mini, but no progress. Already tried with both algorithm types, changed sample rates, channels and everything but nothing worked. 
Thanks in advance

closed 2022-02-17 by espressif-bot