espressif/esp-adf #550: Audio board component only useful for standard

nixmeer, 2020-12-22:
Hello :)

This is a feature request for two features (or where to find them):

I've been using the adf for the past couple of weeks and found lots of helpful stuff in the `/components/audio_board/lyrat_v4_3/`. 

1st: Since I needed to use 16K sample frequency, I had to reimplement `audio_board_init()` and other functions. I could not find a way to pass anything different in there since it uses the `AUDIO_CODEC_DEFAULT_CONFIG()` no matter what. Is there any way to get take the default config, change the parameters to the desired values, and then init the board hardware?

2nd: I took me some time to understand that I need to initialize both the ES8388 and the i2s bus on the ESP32. It would be great to set the wanted settings once (for example only one channel of audio, 16bit and 16k sample frequency) and then call one function that initializes the board and also the i2s_stream and returns both.

Best

Jan

HengYongChao, 2020-12-24:
Hi @nixmeer , if you want configure the i2s, you should reinitializes a member of a structure, as follows:

`    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_READER;
    i2s_cfg.i2s_port = 1;
    i2s_cfg.i2s_config.sample_rate = 16000;
    i2s_cfg.i2s_config.channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT;
    i2s_stream_reader = i2s_stream_init(&i2s_cfg);
`
Here I reconfigure i2s_port = 1 and sample_rate = 16000. And you read the examples, they're very useful.

nixmeer, 2020-12-24:
I know that and did so. But as I said above: 
1st: It would be nice to be able to use `audio_board_init()` with a different config would be nice.
2nd: It would be nice to set the config once, call a single function and get both the board initialized and a fully configured i2s_stream back.

I'm sorry but your answer answered questions I did not ask.

jason-mao, 2022-02-08:
@nixmeer Thank you for your suggestion, we will discuss this.

closed 2022-02-08 by jason-mao