espressif/esp-adf #551: pipeline_bt_source not working on Chip is ESP32-D0WD (revision)

fc92, 2020-12-25:
ESP-IDF : `release v4.2`
ESP-ADF : `commit a3be5b2fcaff02e9303e46a9d8b0f34d0d34f9bb (HEAD, tag: v2.0)`
Trying to stream from example URL : https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3
Hardware: ESP32-D0WD (revision) module on [dev board](https://fr.aliexpress.com/item/4000212139487.html?spm=a2g0s.9042311.0.0.27426c37GqP9JB)

Monitor output:
```
ets Jun  8 2016 00:22:57                                                                                                                
                                                                                                                                        
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)                                                                                 
configsip: 0, SPIWP:0xee                                                                                                                
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00                                                                 
mode:DIO, clock div:2                                                                                                                   
load:0x3fff0030,len:4                                                                                                                   
load:0x3fff0034,len:7044                                                                                                                
load:0x40078000,len:13664                                                                                                               
load:0x40080400,len:4632                                                                                                                
0x40080400: _init at ??:?                                                                                                               
                                                                                                                                        
entry 0x400806f4                                                                                                                        
I (28) boot: ESP-IDF v4.2 2nd stage bootloader                                                                                          
I (28) boot: compile time 23:14:20                                                                                                      
I (28) boot: chip revision: 1                                                                                                           
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0                                                                    
I (38) boot.esp32: SPI Speed      : 40MHz                                                                                               
I (43) boot.esp32: SPI Mode       : DIO                                                                                                 
I (47) boot.esp32: SPI Flash Size : 16MB                                                                                                
I (52) boot: Enabling RNG early entropy source...                                                                                       
I (57) boot: Partition Table:                                                                                                           
I (61) boot: ## Label            Usage          Type ST Offset   Length                                                                 
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000                                                               
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000                                                               
I (83) boot:  2 factory          factory app      00 00 00010000 001e8480                                                               
I (91) boot: End of partition table                                                                                                     
I (95) boot_comm: chip revision: 1, min. application chip revision: 0                                                                   
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x468c4 (288964) map                                               
I (231) esp_image: segment 1: paddr=0x000568ec vaddr=0x3ffbdb60 size=0x04130 ( 16688) load                                              
I (239) esp_image: segment 2: paddr=0x0005aa24 vaddr=0x40080000 size=0x00404 (  1028) load                                              
0x40080000: _WindowOverflow4 at /home/f/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730                                    
                                                                                                                                        
I (240) esp_image: segment 3: paddr=0x0005ae30 vaddr=0x40080404 size=0x051e8 ( 20968) load                                              
I (257) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0x131b04 (1252100) map                                             
0x400d0020: _stext at ??:?                                                                                                              
                                                                                                                                        
I (780) esp_image: segment 5: paddr=0x00191b2c vaddr=0x400855ec size=0x1575c ( 87900) load                                              
0x400855ec: r_assert_param at ??:?                                                                                                      
                                                                                                                                        
I (837) boot: Loaded app from partition at offset 0x10000                                                                               
I (837) boot: Disabling RNG early entropy source...                                                                                     
I (838) psram: This chip is ESP32-D0WD                                                                                                  
I (843) spiram: Found 64MBit SPI RAM device                                                                                             
I (847) spiram: SPI RAM mode: flash 40m sram 40m                                                                                        
I (852) spiram: PSRAM initialized, cache is in low/high (2-core) mode.                                                                  
I (859) cpu_start: Pro cpu up.                                                                                                          
I (863) cpu_start: Application information:                                                                                             
I (868) cpu_start: Project name:     pipeline_bt_source                                                                                 
I (874) cpu_start: App version:      1                                                                                                  
I (878) cpu_start: Compile time:     Dec 25 2020 23:14:14                                                                               
I (884) cpu_start: ELF file SHA256:  6ce07c70d8780fa9...                                                                                
I (890) cpu_start: ESP-IDF:          v4.2                                                                                               
I (895) cpu_start: Starting app cpu, entry point is 0x40081dd4                                                                          
0x40081dd4: call_start_cpu1 at /home/f/esp/esp-idf/components/esp32/cpu_start.c:287                                                     
                                                                                                                                        
I (0) cpu_start: App cpu up.                                                                                                            
I (1786) spiram: SPI SRAM memory test OK
I (1787) heap_init: Initializing. RAM available for dynamic allocation:
I (1787) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1793) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1799) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[... cut ...]
W (9266) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (9266) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, 0
W (9296) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (9296) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, 0
W (9326) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (9326) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, 0
W (9356) BT_APPL: ### UNDERFLOW :: ONLY READ 0 BYTES OUT OF 512 ###
W (9356) BT_APPL: btc_media_aa_prep_sbc_2_send underflow 21, 0
```

fc92, 2020-12-25:
Looks like #115

fc92, 2020-12-30:
A work around has been suggested in this [forum](https://www.esp32.com/viewtopic.php?f=20&t=18545&sid=a44d0fb11d9a1d6eca6e0eecd8580f18) :  
̀̀̀̀```
#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 1, 0))
ESP_ERROR_CHECK(esp_netif_init());
#else
tcpip_adapter_init();
#endif
```
Can you provide a fixed version of `esp_netif_init()` ?

fc92, 2020-12-31:
Same behavior with ESP-IDF v4.3-dev

jason-mao, 2022-08-10:
This example work well on v4.4.2. 

And This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2022-08-10 by jason-mao