espressif/esp-adf #552: How to use SD SPI Host driver to access SD Card ?

fc92, 2020-12-25:
Is it possible to use [SD SPI Host driver ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdspi_host.html) to access a SD Card with ESP-ADF?
Like in this [ESP-IDF example](https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/main/sd_card_example_main.c) using `USE_SPI_MODE`?
I'd like to use a [LOLIN D32 Pro board](https://www.wemos.cc/en/latest/d32/d32_pro.html) with:
```
#define SPI_DMA_CHAN 1

#define PIN_NUM_MISO 19
#define PIN_NUM_MOSI 23
#define PIN_NUM_CLK 18
#define PIN_NUM_CS 0
```

fc92, 2020-12-25:
Looks like #543

HengYongChao, 2020-12-30:
Hi @fc92 
Currently, ADF use SDIO as  sdcard interface, if you need spi interface you may need copy idf spi sdcard implementation in you project. 

sdcard example use two kinds of interfaces spi and sdio : https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card

HengYongChao, 2021-02-23:
Hi @fc92 

SDIO  -->  SPI , YOU only need change the sdcard init mode like this:

```
    // audio_board_sdcard_init(set, SD_MODE_1_LINE);
    audio_board_sdcard_init(set, SD_MODE_SPI);
```

periph sdcard has support 3 modes:
```
typedef enum {
    SD_MODE_SPI    = 0x0,  /*!< sd_card SPI*/
    SD_MODE_1_LINE = 0x1,  /*!< sd_card 1-line SD mode*/
    SD_MODE_4_LINE = 0x2,  /*!< sd_card 4-line SD mode*/
    SD_MODE_MAX,
} periph_sdcard_mode_t;
```

Similar issue #543

closed 2021-09-26 by jason-mao