espressif/esp-adf #553: ESP ADF change/set I2S GPIOs

DEMP1993, 2020-12-29:
Hello,

is it possible to set the GPIO from the I2S stream ?

In the I2S.h from the IDF is the initalisation workflow:

1.  Installing the Driver (_i2s_config_t_)
2. Setting Communication Pins (_i2s_pin_config_t_)
...

In the ADF we configure the I2S Stream [4] over the _i2s_stream_cfg_t_. This includes the _i2s_config_t_ from the I2S.h initalisation workflow, but i can not the _i2s_pin_config_t_ in the _i2s_stream_cfg_t_ from the ADF.

Did someone find out how to change the I2S pins?

HengYongChao, 2020-12-30:
Hi @DEMP1993 , 
In ADF I2s pins are pre-set for support some official development boards, The configuration flow is as follows:
1. Read config by function `get_i2s_pins(I2S_NUM, &i2s_pin_cfg);`, save the pins config in `i2s_pin_cfg` struct.
2.  Set the I2S pins by function `i2s_set_pin(I2S_NUM, &i2s_pin_cfg);`
3. Set mclk output `i2s_mclk_gpio_select(I2S_NUM, GPIO_NUM_0);`

So, if you want change the i2s pins, you may do some steps in the init step of your code like:
1. First initialize the structure  `i2s_pin_config_t i2s_pin_cfg = {....}`
2. Set I2S pins by use function `i2s_set_pin(2S_NUM, &i2s_pin_cfg);`
3. Set mclk output `i2s_mclk_gpio_select(TEST_I2S_NUM, GPIO_NUM_0);`

I hope these answers were helpful to you.

DEMP1993, 2020-12-30:
Hi HengYongChao,

thanks for the answer. It is very helpfull.

closed 2020-12-30 by DEMP1993