espressif/esp-adf #555: A2DP buffer size/delay

shark8594, 2021-01-04:
Hi All!,

I tried the player/pipeline_a2dp_sink_stream example on my ESP-Lyra-T board. I wanted know how to set the A2DP buffer size that controls latency?. For my audio application the BT link is not very good and therefore I want to set the buffer size to be as large a possible so that a few seconds of audio can be buffered.

Thanks!

HengYongChao, 2021-01-19:
Hi, @shark8594 
Do you use a2dp sink role? if yes I think you need modify element out ringbuf size, here:
https://github.com/espressif/esp-adf/blob/master/components/bluetooth_service/a2dp_stream.c#L240
add follow code after line 240.
```
cfg.out_rb_size = 16 * 1024;
```
And if use a2dp source role, please add after line 249.

shark8594, 2021-01-19:
Thanks a lot @HengYongChao, yes I am using a2dp_sink. I managed to change the rb size and can now buffer 11-15s of audio (log below).

 However I see that the RB size is mostly constant while streaming. I am trying to build a system where the number of bytes filled in the ringbuf increases over time. Meaning the BT PHY data rate is 2 to 3 times larger than the encoded bitrate. To accomplish this I want to reduce the encoded bit rate by changing the max bitpool value. Any clue on how to change the max bitpool value? 

[image]

HengYongChao, 2021-01-20:
Hi @shark8594 

Please see here (https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/btc/profile/std/a2dp/bta_av_co.c#L66), a2dp max bitpool default value is 53.
` #define BTA_AV_CO_SBC_MAX_BITPOOL  53 `

shark8594, 2021-01-20:
Thanks @HengYongChao , along with this change I needed to also change: A2D_SBC_IE_MAX_BITPOOL in a2d_sbc.h. I see that the change is taking effect from the HCI logs (below). I see that the rb size is still mostly constant with a lower bitpool meaning the BT PHY data rate was adjusted to match the lower bitrate. Any clue on how to have a BT PHY rate of 328 kbps with low bitpool so that I can buffer more data the longer the source is connected?.

[image]

HengYongChao, 2021-01-22:
Hi @shark8594  I'm not find the api of BT PHY rate of 328 kbps, May be you can calculate the true rate from the bitpool?

closed 2021-04-18 by shark8594