espressif/esp-adf #557: linking failed with dangerous relocation: cannot decode instruction opcode in libesp_processing.a

micky7, 2021-01-09:
Hello,
i have a problem that the linking fails with the following error messages. I have updated to latest github for idf and adf. Now I can not link my application anymore. 

`Linking CXX executable huhu_books.elfm
FAILED: huhu_books.elf 
: && /home/schwm/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++  -mlongcalls   CMakeFiles/huhu_books.elf.dir/project_elf_src_esp32s2.c.obj  -o huhu_books.elf  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32s2/libesp32s2.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/esp-sr/libesp-sr.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/dueros_service/libdueros_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/playlist/libplaylist.a  esp-idf/my_board/libmy_board.a  esp-idf/my_gpio/libmy_gpio.a  esp-idf/pn532/libpn532.a  esp-idf/rgb-led/librgb-led.a  esp-idf/wifi-prov/libwifi-prov.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/schwm/huhu/huhu_books/build/huhu_books.map  -fno-rtti  -fno-lto  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/jsmn/libjsmn.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/esp_actions/libesp_actions.a  -L/home/schwm/Tools/esptools/esp-adf/components/wifi_service/airkiss_config  -lairkiss_aes  esp-idf/dueros_service/libdueros_service.a  -L/home/schwm/Tools/esptools/esp-adf/components/clouds/dueros/lightduer  -lduer-device-v4x  esp-idf/playlist/libplaylist.a  esp-idf/my_gpio/libmy_gpio.a  esp-idf/pn532/libpn532.a  esp-idf/rgb-led/librgb-led.a  esp-idf/display_service/libdisplay_service.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/my_board/libmy_board.a  esp-idf/display_service/libdisplay_service.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/my_board/libmy_board.a  -L/home/schwm/Tools/esptools/esp-adf/components/audio_hal/driver/zl38063/firmware  -lfirmware  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/spiffs/libspiffs.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2/libesp_processing.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_audio/lib/esp32s2/libesp_audio.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2/libesp-opus.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2/libesp-tremor.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32s2/librecorder_engine.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2/libesp_processing.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_audio/lib/esp32s2/libesp_audio.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2/libesp-opus.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2/libesp-tremor.a  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32s2/librecorder_engine.a  esp-idf/esp-sr/libesp-sr.a  -L /home/schwm/Tools/esptools/esp-adf/components/esp-sr/lib  -L /home/schwm/Tools/esptools/esp-adf/components/esp-sr/wake_word_engine  -L /home/schwm/Tools/esptools/esp-adf/components/esp-sr/speech_command_recognition  -L /home/schwm/Tools/esptools/esp-adf/components/esp-sr/acoustic_algorithm  -L /home/schwm/Tools/esptools/esp-adf/components/esp-sr/esp-tts/esp_tts_chinese  -lesp_tts_chinese_esp32s2  -lvoice_set_xiaole_esp32s2  -lvoice_set_template_esp32s2  -Wl,--start-group  -Wl,--end-group  -L /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_audio/lib/esp32s2  -L /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32s2  -L /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_ssdp/lib/esp32s2  -L /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_upnp/lib/esp32s2  -L /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_sip/lib/esp32s2  -L /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32s2  -lesp-aac  -lesp-amr  -lesp-amrwbenc  -lesp-flac  -lesp_ssdp  -lesp-ogg-container  /home/schwm/Tools/esptools/esp-adf/components/esp-adf-libs/esp_sip/lib/esp32s2/libesp_sip.a  -lesp_upnp  -lesp-mp3  -lcodec-utils  -lesp-g711  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/wifi-prov/libwifi-prov.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/console/l [... cut ...]
/home/schwm/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/gitlab-runner/builds/w2jWgEzx/0/adf/esp-adf-libs-source/esp_processing/esp-alc/src/esp_alc.c:303:(.text.alc_volume_setup_process+0x400): dangerous relocation: cannot decode instruction opcode: (.text.alc_volume_setup_process+0x406)
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

houhaiyan, 2021-01-11:
> [External: This email originated outside Espressif]

+MJX

Best regards,
Haiyan Hou 侯海彦​
Software Engineer|Espressif Systems

espressif<https://www.espressif.com/>
#204, Block 2, 690 Bibo Road,
Zhang Jiang High-Tech Park, Shanghai, China

@houhaiyan<https://github.com/houhaiyan> If i stay with the commit 40cc7ba<https://github.com/espressif/esp-adf/commit/40cc7ba9eb911f809676fef511719c365b224f9c> i have no issue with linking.

jason-mao, 2021-01-11:
Hi @micky7,  Would please provide your IDF version and name of example?

micky7, 2021-01-11:
Hi @jason-mao ,

i use the idf version with the commit number 2bfdd036b2dbd07004c8e4f2ffc87c728819b737.

I have used a wifi-provisoning example from idf and added a small audio player example.

```
esp_audio_cfg_t cfg = DEFAULT_ESP_AUDIO_CONFIG();
audio_board_handle_t board_handle = audio_board_init();
cfg.vol_handle = board_handle->audio_hal;
cfg.vol_set = (audio_volume_set)audio_hal_set_volume;
cfg.vol_get = (audio_volume_get)audio_hal_get_volume;
cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
cfg.resample_rate = 44100;

cfg.evt_que = xQueueCreate(3, sizeof(esp_audio_state_t));
player = esp_audio_create(&cfg);

esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);

audio_board_key_init(set);
input_key_service_info_t input_key_info[] = INPUT_KEY_DEFAULT_INFO();
input_key_service_cfg_t input_cfg = INPUT_KEY_SERVICE_DEFAULT_CONFIG();
input_cfg.handle = set;
periph_service_handle_t input_ser = input_key_service_create(&input_cfg);
input_key_service_add_key(input_ser, input_key_info, INPUT_KEY_NUM);
periph_service_set_callback(input_ser, input_key_service_cb, (void *)board_handle);

audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
xTaskCreate(esp_audio_state_task, "player_task", 4096, cfg.evt_que, 1, NULL);

http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
http_cfg.event_handle = _http_stream_event_handle;
http_cfg.type = AUDIO_STREAM_READER;
http_cfg.enable_playlist_parser = false;
audio_element_handle_t http_stream_reader = http_stream_init(&http_cfg);
esp_audio_input_stream_add(player, http_stream_reader);

ESP_LOGI(TAG, "[2.2] Create i2s stream to write data to codec chip");
i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
i2s_cfg.type = AUDIO_STREAM_WRITER;
i2s_cfg.use_alc = true;
i2s_stream_writer = i2s_stream_init(&i2s_cfg);
i2s_alc_volume_set(i2s_stream_writer, -20);
esp_audio_output_stream_add(player, i2s_stream_writer);

mp3_decoder_cfg_t  mp3_dec_cfg  = DEFAULT_MP3_DECODER_CONFIG();
wav_decoder_cfg_t  wav_dec_cfg  = DEFAULT_WAV_DECODER_CONFIG();

esp_audio_codec_lib_add(player, AUDIO_CODEC_TYPE_DECODER, mp3_decoder_init(&mp3_dec_cfg));
esp_audio_codec_lib_add(player, AUDIO_CODEC_TYPE_DECODER, wav_decoder_init(&wav_dec_cfg));

char url[]="https://swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3";
audio_err_t err=esp_audio_play(player, AUDIO_CODEC_TYPE_DECODER, url,0);
ESP_LOGI(TAG, "esp_audio_play err:%x\n", err);
while (1) {}
```

closed 2021-01-18 by espressif-bot