espressif/esp-adf #560: 在进行http上传数据时断开wifi热点，重新连接后无声音播放

Kevincoooool, 2021-01-19:
在进行http上传数据的同时断开wifi热点，此时重新连接后播放器log显示但无声音输出
_http_open
get_local_asr_result() = 1
------------------------------------------
pcName  = OUT_iis pcTaskName = OUT_iis pxNewTCB->pxStack=0x3fff8768 stack_depth = 3584
W (53679) HTTP_STREAM: No more data,errno:0, total_bytes:6552, rlen = 0
W (54141) AUDIO_PIPELINE: Check AEL RUNNING, pl:0x3f8152f8, el:0x3f8168a0, tag:         IN_http, state:15, wanted:12
E (54141) ESP_AUDIO_PLAY: take_play_ctrl_semaphore
E (54147) ESP_AUDIO_PLAY: esp_audio_play_callback audio->status = 1
W (57475) ESP_AUDIO_TASK: Destroy the old pipeline
W (57475) ESP_AUDIO_TASK: The old pipeline destroyed
E (57476) ESP_AUDIO_PLAY: gave_play_ctrl_semaphore
E (57481) ESP_AUDIO_CTRL: media_ctrl_audio_info_get, user_data_2:0, dat:0x0-0, line:1153
E (57489) ESP_AUDIO_PLAY: play tts finished
E (57494) ESP_AUDIO_PLAY: last_pause_audio_type = 0 is_wakeup_play 0
E (57501) ESP_AUDIO_PLAY: play_end_call_back
E (57506) ESP_AUDIO_PLAY: roobo_play_queue msg.type 1
E (57512) ESP_AUDIO_PLAY: esp_audio_play_callback audio->status = 4
W (57520) AUDIO_PIPELINE: There are no listener registered

_http_open
pcName  = OUT_iis pcTaskName = OUT_iis pxNewTCB->pxStack=0x3fff8768 stack_depth = 3584
E (57651) ESP_AUDIO_PLAY: take_play_ctrl_semaphore
E (57651) ESP_AUDIO_PLAY: esp_audio_play_callback audio->status = 1
E (58760) TIME: cur data Tue Jan 19 15:03:35 15:03 RTOS TIMER 20
E (68291) TEST_SPEEEK: HTTP GET Status = 200, content_length = -1
E (68760) TIME: cur data Tue Jan 19 15:03:45 15:03 RTOS TIMER 30

HengYongChao, 2021-01-19:
Hi, @Kevincoooool 
如你描述 ： `在进行http上传数据时断开wifi热点，重新连接后无声音播放 (AUD-2728)`，
我不太明白的是你的程序是在做数据上传，重新连接后无声音播放是指提示音还是你正在上传播放的音乐？ 
1. 如果是 wifi 断开提示音， 那么需要在接收 wifi 断开事件后播放提示音即可。
2. 如果是上传的音乐， 那么断开就是异常时间，需要做异常处理， WiFi 断开后需要重连 socket ，重新获取网络数据进行播放。

如果理解或回复有误，欢迎补充讨论。

Kevincoooool, 2021-01-25:
> Hi, @Kevincoooool
> 如你描述 ： `在进行http上传数据时断开wifi热点，重新连接后无声音播放 (AUD-2728)`，
> 我不太明白的是你的程序是在做数据上传，重新连接后无声音播放是指提示音还是你正在上传播放的音乐？
> 
> 1. 如果是 wifi 断开提示音， 那么需要在接收 wifi 断开事件后播放提示音即可。
> 2. 如果是上传的音乐， 那么断开就是异常时间，需要做异常处理， WiFi 断开后需要重连 socket ，重新获取网络数据进行播放。
> 
> 如果理解或回复有误，欢迎补充讨论。

你好，这个问题我已经做了规避处理，但是遇到了另一个问题，播放在线音乐时出现的异常复位：
W (481354) AUDIO_PIPELINE: There are no listener registered

_http_open
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094ae7  PS      : 0x00060a30  A0      : 0x8014c211  A1      : 0x3f816130
0x40094ae7: xTaskCreatePinnedToCore at D:/esp/msys32/home/97354/esp/esp-adf/esp-idf/components/freertos/tasks.c:3683

A2      : 0x00000000  A3      : 0x3f816fb0  A4      : 0x00000e00  A5      : 0x3f816ea0
A6      : 0x00000006  A7      : 0x00000804  A8      : 0x00000000  A9      : 0x3f816100
A10     : 0x00000000  A11     : 0x00000804  A12     : 0x00000003  A13     : 0x00000008
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000034  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

ELF file SHA256:

Backtrace: 0x40094ae7:0x3f816130 0x4014c20e:0x3f816170 0x4014a2f9:0x3f8161e0 0x401b6395:0x3f816230
0x40094ae7: xTaskCreatePinnedToCore at D:/esp/msys32/home/97354/esp/esp-adf/esp-idf/components/freertos/tasks.c:3683

0x4014c20e: audio_thread_create at D:/esp/msys32/home/97354/esp/esp-adf/components/audio_sal/audio_thread.c:75

0x4014a2f9: audio_element_run at D:/esp/msys32/home/97354/esp/esp-adf/components/audio_pipeline/audio_element.c:1307

0x401b6395: media_ctrl_task at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_audio/src/media_ctrl_task.c:654

Kevincoooool, 2021-01-25:
内存泄漏导致

closed 2021-01-25 by Kevincoooool