espressif/esp-adf #561: ESP  LyraT mini v1.2 output mic to speaker

vrycue, 2021-01-20:
is it possible to output mic directly to the speakers? I tried the pipeline that looks like this:       i2s_reader --->  i2swriter, I can only hear noise even when using headphones.

HengYongChao, 2021-01-20:
Hi, @vrycue 
FYI:  [MIC] --> [CODEC CHIP] --> I2S_READER  --> RINGBUFF --> I2S_WRITER --> [CODEC CHIP] -->[PA] --> [SPEAKER]

vrycue, 2021-01-20:
I have modified the passthru example here is my code`

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "audio_pipeline.h"
#include "i2s_stream.h"
#include "raw_stream.h"
#include "board.h"

static const char *TAG = "PASSTHRU";

void app_main(void)
{
    audio_pipeline_handle_t pipeline;
    audio_element_handle_t i2s_stream_writer, i2s_stream_reader, el_raw_write;

    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    ESP_LOGI(TAG, "[ 1 ] Start codec chip");
    audio_board_handle_t board_handle = audio_board_init();

#ifdef CONFIG_ESP32_S2_KALUGA_1_V1_2_BOARD
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
#else
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
#endif

    audio_hal_set_volume(board_handle-> audio_hal, 100);

    ESP_LOGI(TAG, "[ 2 ] Create audio pipeline for playback");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline = audio_pipeline_init(&pipeline_cfg);

    ESP_LOGI(TAG, "[3.1] Create i2s stream to write data to codec chip");
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_WRITER;
    i2s_stream_writer = i2s_stream_init(&i2s_cfg);

    ESP_LOGI(TAG, "[3.2] Create i2s stream to read data from codec chip");
    i2s_stream_cfg_t i2s_cfg_read = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg_read.type = AUDIO_STREAM_READER;
    i2s_cfg.i2s_port = 1;
    i2s_cfg.multi_out_num = 1;
    i2s_stream_reader = i2s_stream_init(&i2s_cfg_read);

    ESP_LOGI(TAG, "[4.2] Create raw stream of base wav to write data");
    raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_cfg.type = AUDIO_STREAM_WRITER;

    el_raw_write = raw_stream_init(&raw_cfg);

    ESP_LOGI(TAG, "[3.3] Register all elements to audio pipeline");
    audio_pipeline_register(pipeline, i2s_stream_reader, "i2s_read");
    audio_pipeline_register(pipeline, el_raw_write, "raw");
    audio_pipeline_register(pipeline, i2s_stream_writer, "i2s_write");

    ESP_LOGI(TAG, "[3.4] Link it together [codec_chip]-->i2s_stream_reader-->i2s_stream_writer-->[codec_chip]");
    const char *link_tag[3] = {"i2s_read", "raw", "i2s_write"};
    audio_pipeline_link(pipeline, &link_tag[0], 3);

    ringbuf_handle_t rb = audio_element_get_output_ringbuf(el_raw_write);
    audio_element_set_multi_output_ringbuf(i2s_stream_reader, rb, 0);

    ESP_LOGI(TAG, "[ 4 ] Set up  event listener");
    audio_event_iface_cfg_t evt_cfg = AUDIO_EVENT_IFACE_DEFAULT_CFG();
    audio_event_iface_handle_t evt = audio_event_iface_init(&evt_cfg);

[... 71 lines cut ...]
I (404) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (404) LYRAT_MINI_V1_1: I2S0, MCLK output by GPIO0
I (414) PASSTHRU: [3.2] Create i2s stream to read data from codec chip
W (414) I2S: I2S driver already installed
I (424) LYRAT_MINI_V1_1: I2S0, MCLK output by GPIO0
I (424) PASSTHRU: [4.2] Create raw stream of base wav to write data
I (434) PASSTHRU: [3.3] Register all elements to audio pipeline
I (444) PASSTHRU: [3.4] Link it together [codec_chip]-->i2s_stream_reader-->i2s_stream_writer-->[codec_chip]
I (454) AUDIO_PIPELINE: link el->rb, el:0x3ffb8828, tag:i2s_read, rb:0x3ffb8ee8
I (464) AUDIO_PIPELINE: link el->rb, el:0x3ffb8b50, tag:raw, rb:0x3ffbb024
I (464) PASSTHRU: [ 4 ] Set up  event listener
I (474) PASSTHRU: [4.1] Listening event from all elements of pipeline
I (474) PASSTHRU: [ 5 ] Start audio_pipeline
I (484) AUDIO_ELEMENT: [i2s_read-0x3ffb8828] Element task created
I (494) AUDIO_ELEMENT: [raw-0x3ffb8b50] Element task created
I (494) AUDIO_ELEMENT: [i2s_write-0x3ffb8524] Element task created
I (504) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:259200 Bytes

I (514) AUDIO_ELEMENT: [i2s_read] AEL_MSG_CMD_RESUME,state:1
I (514) I2S_STREAM: AUDIO_STREAM_READER,Rate:44100,ch:2
I (544) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (554) AUDIO_ELEMENT: [i2s_write] AEL_MSG_CMD_RESUME,state:1
I (554) I2S_STREAM: AUDIO_STREAM_WRITER
I (564) AUDIO_PIPELINE: Pipeline started
I (564) PASSTHRU: [ 6 ] Listen for all pipeline events

HengYongChao, 2021-01-20:
Hi @vrycue 
I think you should read the example `record_while_play` , It's Similar to your need.
https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/record_while_play

vrycue, 2021-01-20:
Hi @HengYongChao, okay ill check it out. Thank you.

closed 2022-02-08 by jason-mao

bulbitron, 2023-06-24:
Hello, I join the question, to do something that should be simple I am going crazy, LyraTmini microphone input, and output in real time, some of us are newbies, thanks.

HyouinSchoolAcc, 2024-08-19:
Is it just me or have I never seen anyone managed to get lyrat mini's 1.2 microphone to work? I have never seen it work anywhere.

Bhavesh-Moradiya, 2024-08-23:
lyrat mini's 1.2 microphone connected to ADC ES7243 only i guess, is it true?

Bhavesh-Moradiya, 2024-08-23:
I am trying to get raw PCM data and want to stream on UDP socket but raw data buffer always 0