espressif/esp-adf #562: AUDIO_HAL: codec init failed!

shameemsanoodh, 2021-01-23:
This was the error while compiling ADF example library for lyrat v4.3.

C:\esp\esp-idf\esp-adf\examples\get-started\play_mp3>idf.py monitor
Executing action: monitor
Choosing default port b'COM3' (use '-p PORT' option to set a specific serial port)
Running idf_monitor in directory c:\esp\esp-idf\esp-adf\examples\get-started\play_mp3
Executing "C:\esp\tools\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe C:\esp\esp-idf\tools/idf_monitor.py -p COM3 -b 115200 --toolchain-prefix xtensa-esp32-elf- c:\esp\esp-idf\esp-adf\examples\get-started\play_mp3\build\play_mp3.elf -m 'C:\esp\tools\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe' 'C:\esp\esp-idf\tools\idf.py'"...
--- idf_monitor on COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6872
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (30) boot: ESP-IDF v4.1-dirty 2nd stage bootloader
I (30) boot: compile time 18:56:38
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 8MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f5e4 (128484) map
I (162) esp_image: segment 1: paddr=0x0002f60c vaddr=0x3ffb0000 size=0x00a0c (  2572) load
I (164) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x24418 (148504) map
0x400d0020: _stext at ??:?

I (225) esp_image: segment 3: paddr=0x00054440 vaddr=0x3ffb0a0c size=0x017c0 (  6080) load
I (228) esp_image: segment 4: paddr=0x00055c08 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (233) esp_image: segment 5: paddr=0x00056014 vaddr=0x40080404 size=0x0ba44 ( 47684) load
I (268) boot: Loaded app from partition at offset 0x10000
I (268) boot: Disabling RNG early entropy source...
I (269) cpu_start: Pro cpu up.
I (272) cpu_start: Application information:
I (277) cpu_start: Project name:     play_mp3
I (282) cpu_start: App version:      v2.2-33-g7fca3dc-dirty
I (289) cpu_start: Compile time:     Jan 23 2021 18:55:51
I (295) cpu_start: ELF file SHA256:  6625a2c67053ea30...
I (301) cpu_start: ESP-IDF:          v4.1-dirty
I (306) cpu_start: Starting app cpu, entry point is 0x40081124
0x40081124: call_start_cpu1 at C:/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (316) heap_init: Initializing. RAM available for dynamic allocation:
I (323) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (329) heap_init: At 3FFB2A10 len 0002D5F0 (181 KiB): DRAM
I (335) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (342) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (348) heap_init: At 4008BE48 len 000141B8 (80 KiB): IRAM
I (354) cpu_start: Pro cpu start user code
I (373) spi_flash: detected chip: gd
I (373) spi_flash: flash io: dio
I (373) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (381) PLAY_MP3_FLASH: [ 1 ] Start audio codec chip
E (391) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (401) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (411) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (431) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (441) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (451) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (471) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (481) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (491) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (511) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (521) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (531) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (551) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (561) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (571) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (591) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (601) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (611) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (621) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (641) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (651) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (661) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (681) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (691) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (701) I2C_BUS: C:/esp/esp-idf/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (721) AUDIO_HAL: codec init failed!
E (721) AUDIO_BOARD: C:/esp/esp-idf/esp-adf/components/audio_board/lyrat_v4_3/board.c:55 (audio_board_codec_init): Got NULL Pointer
E (731) AUDIO_HAL: audio_hal handle is null
I (741) PLAY_MP3_FLASH: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline event
I (751) PLAY_MP3_FLASH: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback
I (761) PLAY_MP3_FLASH: [2.2] Create i2s stream to write data to codec chip
I (791) PLAY_MP3_FLASH: [2.3] Register all elements to audio pipeline
I (791) PLAY_MP3_FLASH: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip]
I (801) PLAY_MP3_FLASH: [ 3 ] Set up  event listener
I (801) PLAY_MP3_FLASH: [3.1] Listening event from all elements of pipeline
I (811) PLAY_MP3_FLASH: [ 4 ] Start audio_pipeline
I (831) PLAY_MP3_FLASH: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2

douardda, 2021-01-24:
Are you using a Lyrat 4.3 board? 
I'm asking because I had similar errors because I am using a bare esp32-s2-soola board with a pcm5142 dac, and I had to write a custom board definition with a simple driver for this dac to make the example work. (now I'm fighting with the wifi configuration using ADF's "wifi peripheral", but that's unrelated).

HengYongChao, 2021-01-25:
Hi @shameemsanoodh ,
What's adf version do you using? 

I compile adf\examples\get-started\play_mp3 example, I didn't repeat the problem you reported.

adf: (66eb22d3bd9181761e6b338d9434c5c9387c0432),
idf v4.1: (9778b163b174b1e4702cf354e9a26d98c511fd6c), 
board is Lyrat v4.3.

Does this happen every time when you try, or occasionally?

shameemsanoodh, 2021-01-25:
hi @HengYongChao 
I am using a custom wrover Dev board with flash size of 8MB and es8388 ic for my project without lyrat v4.3 . 
I have used the schematic of lyrat and es8388 for reference. 

can u suggest me if any exceptional configuration required for my custom board?

when I try to upload the example code of adf this is the error it shows.

HengYongChao, 2021-01-25:
> hi @HengYongChao
> I am using a custom wrover Dev board with flash size of 8MB and es8388 ic for my project without lyrat v4.3 .
> I have used the schematic of lyrat and es8388 for reference.
>
> can u [...]

Well, I think you need check  es8388 i2c port pins, at the init step es8388 registers will be setting. and some register of es8388 also need i2s clk to sync settings.

If your project schematic is consistent with Lyrat V4.3, you may need to check the hardware connectivity of your board.

shameemsanoodh, 2021-01-25:
I use the project schematic of lyrat v4.3 all the hardware connectivity is as per the board configuration.

https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/board-esp32-lyrat-v4.3.html
These are i2c and i2s pins I followed for my project.
Is there any specific configuration in the example/ play_mp3 for using a custom board?

HengYongChao, 2021-01-25:
If your board has the same connection as Lyrat V4.3, it should not be a problem, so do you selects the Lyrat V4.3 development board HAL in menuconfig already?

shameemsanoodh, 2021-01-25:
yes selected.
any other configuration required in menuconfig?

shameemsanoodh, 2021-01-25:
[image]

HengYongChao, 2021-01-25:
In general, if your board is the same as the official one, it should be working fine that assuming your board hardware is working properly.

You may need to use an oscilloscope or signal analyzer to determine the communication, and that doesn't rule out the possibility that the Codec chip isn't working properly.

shameemsanoodh, 2021-01-25:
[image]

This was the error....

shameemsanoodh, 2021-01-25:
[image]

it shows memory exhausted

douardda, 2021-01-25:
@shameemsanoodh it's pretty hard to tell what your problem really is there... 

Your original post clearly indicates a problem initializing the es8388 dac, so either a wrong i2c pin configuration or a hw problem. Then you put screenshots of other things that looks completely unrelated... In your screenshot "This was the error" I see no real error: the play_mp3 example does create the audio pipeline and plays the mp3 file then stops the pipeline. The errors you see in this screenshot are only related to the way the pipeline is shut down and have no impact on the overall result of this example.
Then you put a screenshot of another problem ("memory exhausted" above) without giving any clue on what you changed in your project to get this error. 

Please give better context if you want helpful comments... Maybe show the schematics or your board and publish your modified version of the play_mp3 example somewhere, for example. 

As far as I understand your problem, you must first and foremost debug your i2c communication problem, as suggested by  @HengYongChao (randomly commenting lines in the source code is not a efficient method to fix your problem.)

shameemsanoodh, 2021-01-27:
Hi @douardda @HengYongChao
Hardware configuration as per the schematic provided by espressif and tried to upload the ADF example code play_mp3.
[image]
it started execution and show an exception unhandled error.

HengYongChao, 2021-01-28:
> Hi @douardda @HengYongChao
> Hardware configuration as per the schematic provided by espressif and tried to upload the ADF example code play_mp3.
> [image]
> it started execution and show an exception unhandled error.

Hi @douardda ,

This error is a `LoadProhibited` error, which usually means that this type of CPU exception occurs when the application tries to read or write to an invalid memory location.

The register `EXCVADDR` value is `0x00000003`. It shows that this may be a null pointer of a structure, and you are referencing its members. So its address is 0x00000003. You need to check the structure pointer near the crash code.

cattleliu, 2021-08-09:
I also encountered the same problem, I made a simplification on the basis of ESP32-LyraT4.3, I used the same module ESP32, but flash is 4Mb, the same error.I observed with the oscilloscope that the LEVEL of CDATA pin of the ESP module did not change, and that of the other pin CCLK changed. I confirmed that there was no error in the electrical connection.

HengYongChao, 2021-08-10:
> I also encountered the same problem, I made a simplification on the basis of ESP32-LyraT4.3, I used the same module ESP32, but flash is 4Mb, the same error.I observed with the oscilloscope that the [...]

Hi @cattleliu 

If you use a self-made board, make sure you have a good connection and initialized the codec. If it prompts that the initialization failed, please find out at which step the code failed. This is not difficult.

If you need our help, it is best to explain what you have tried yourself, where you are stuck, and provide related error logs and codes.

cattleliu, 2021-08-16:
> Hi @cattleliu
>
> If you use a self-made board, make sure you have a good connection and initialized the codec. If it prompts that the initialization failed, please find out at which step the code [...]

Thank you for your reply. I have found the problem. It is a hardware problem，The pull-up resistor of the data line was short-circuit, so there was no way to pull the electricity flat. I found this problem during a separate test with GPIO. Now I have repaired the welding problem and initialized successfully

HengYongChao, 2021-08-16:
> Thank you for your reply. I have found the problem. It is a hardware problem，The pull-up resistor of the data line was short-circuit, so there was no way to pull the electricity flat. I found this [...]

That sounds good! Codec initialization failure is usually a hardware problem.

xiaohuihere, 2023-02-23:
> 
Hi, this is the log of my error report, I have combined the keystroke with the function to play mp3, when I press a key it will automatically read the internal mp3 file for playback. I have tried to burn my code in esp32 lyrat v4.3 and the function works fine.
```
PS C:\Users\22380\Documents\DebugCase\KeChuang\test\pipeline_play_sdcard_music> set IDF_PATH=C:\Users\22380\esp\esp-idf
PS C:\Users\22380\Documents\DebugCase\KeChuang\test\pipeline_play_sdcard_music> C:\Users\22380\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\22380\esp\esp-idf\tools\idf_monitor.py -p 
COM3 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 c:\Users\22380\Documents\DebugCase\KeChuang\test\pipeline_play_sdcard_music\build\play_sdcard_music.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- idf_monitor on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14876
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.3-316-ge86181704a-dirty 2nd stage bootloader
I (29) boot: compile time 14:44:38
I (30) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=17384h ( 95108) map
I (148) esp_image: segment 1: paddr=000273ac vaddr=3ffb0000 size=02660h (  9824) load
I (152) esp_image: segment 2: paddr=00029a14 vaddr=40080000 size=06604h ( 26116) load
I (164) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=3ac40h (240704) map
I (252) esp_image: segment 4: paddr=0006ac68 vaddr=40086604 size=09d08h ( 40200) load
I (276) boot: Loaded app from partition at offset 0x10000
I (276) boot: Disabling RNG early entropy source...
I (288) cpu_start: Pro cpu up.
I (288) cpu_start: Starting app cpu, entry point is 0x40081424
0x40081424: call_start_cpu1 at C:/Users/22380/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (302) cpu_start: Pro cpu start user code
I (302) cpu_start: cpu freq: 160000000
I (302) cpu_start: Application information:
I (307) cpu_start: Project name:     play_sdcard_music
I (312) cpu_start: App version:      1
I (317) cpu_start: Compile time:     Feb 23 2023 14:44:19
I (323) cpu_start: ELF file SHA256:  445bed0fde4c6967...
I (329) cpu_start: ESP-IDF:          v4.4.3-316-ge86181704a-dirty
I (336) heap_init: Initializing. RAM available for dynamic allocation:
I (343) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (349) heap_init: At 3FFB31B8 len 0002CE48 (179 KiB): DRAM
I (355) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (362) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (368) heap_init: At 4009030C len 0000FCF4 (63 KiB): IRAM
I (376) spi_flash: detected chip: generic
I (379) spi_flash: flash io: dio
I (384) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (394) CHECK_BOARD_BUTTONS: [ 1 ] Initialize peripherals
I (404) CHECK_BOARD_BUTTONS: [ 2 ] Initialize Button peripheral with board init
[... 80 lines cut ...]
I (241364) PLAY_SDCARD_MUSIC: [4.2] Listening event from peripherals
I (241364) PLAY_SDCARD_MUSIC: [ 5 ] Start audio_pipeline
I (241374) PLAY_SDCARD_MUSIC: [ 6 ] Listen for all pipeline events
I (241534) PLAY_SDCARD_MUSIC: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, 
ch=2
W (241564) AUDIO_ELEMENT: [i2s-0x3ffc3470] RESUME timeout
W (475284) FATFS_STREAM: No more data, ret:0
W (475604) PLAY_SDCARD_MUSIC: [ * ] Stop event received
I (475604) PLAY_SDCARD_MUSIC: [ 7 ] Stop audio_pipeline
E (475604) AUDIO_ELEMENT: [file] Element already stopped
E (475614) AUDIO_ELEMENT: [dec] Element already stopped
E (475624) AUDIO_ELEMENT: [i2s] Element already stopped
W (475624) AUDIO_PIPELINE: There are no listener registered
W (475634) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (475644) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (475654) AUDIO_ELEMENT: [dec] Element has not create when AUDIO_ELEMENT_TERMINATE
```
So I drew the PCB according to lyrat's schematic and made sure that the board selected for menuconfig is also esp32 lyrat v4.3, however he shows ```I2C Bus WriteReg Error```, ```codec init failed!``` and``` audio_hal handle is null0```.

This is my schematic, I've confirmed it many times but it's basically the same as the original schematic, so if this really isn't an error, I'd like to ask if there are any other settings or actions that need to be done to initialise it?

[SCH_Main_Control_Board_2023-02-23.pdf](https://github.com/espressif/esp-adf/files/10812671/SCH_Main_Control_Board_2023-02-23.pdf)
[SCH_SD card module_2023-02-23.pdf](https://github.com/espressif/esp-adf/files/10812673/SCH_SD.card.module_2023-02-23.pdf)
[SCH_Audio Processing Module_2023-02-23.pdf](https://github.com/espressif/esp-adf/files/10812674/SCH_Audio.Processing.Module_2023-02-23.pdf)
[Connection plate_2023-02-23.pdf](https://github.com/espressif/esp-adf/files/10812680/Connection.plate_2023-02-23.pdf)

closed 2023-09-05 by jason-mao