espressif/esp-adf #563: Update to IDF 4.2

douardda, 2021-01-24:
What's the status about using IDF 4.2 (instead of the current 3.3.2+)?
Also, it's unclear to me how the version of IDF impact the config and compilation of ADF projects. 
The "getting-started' guide of ADF says to install EDF first (thus, the current version, aka 4.2), then to downloas ADF which has IDF 3.3.2 (as a git submodule)

So with this config, one have the idf.py tool from IDF 4.2 but the IDF 3.2.2 remains in the esp-adf directory. So which part of IDF 3.2.2 is used? and which part of IDF 4.2 is used? 

This is pretty confusing...

HengYongChao, 2021-01-25:
Hi @douardda,

ADF can build on IDF version (v3.3, v4.0,v4.1,v4.2) , It's depend on $IDF_PATH and toolchain settings.

This means that when you set the external IDF path and the compiler tool, the ADF built-in IDF(v3.3) will not be referenced by the compiler. 

This is also mentioned in the ADF documentation: https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#step-1-set-up-esp-idf

douardda, 2021-01-25:
Thanks for the explanation. However I disagree this is mentioned in the doc: the fact there is old version of IDF "embedded" with ADF but unused under "normal" usage is not mentioned and is confusing (at least to me). 

Why not just remove this esp-idf subrepo from the esp-adf one? 

Anyway, thanks again.

douardda, 2021-01-25:
I guess this issue can be closed.

closed 2021-01-25 by douardda