espressif/esp-adf #564: The wifi peripheral seems to be broken

douardda, 2021-01-24:
Trying to make the pipeline_http_mp3 example work on a esp32-s2-soola board (WROVER-I) it blocks with:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x4
load:0x3ffe6104,len:0x1894
load:0x4004c000,len:0x1688
load:0x40050000,len:0x20fc
entry 0x4004c324
I (46) boot: ESP-IDF v4.2 2nd stage bootloader
I (46) boot: compile time 18:48:22
I (47) boot: chip revision: 0
I (48) boot.esp32s2: SPI Speed      : 80MHz
I (53) boot.esp32s2: SPI Mode       : DIO
I (58) boot.esp32s2: SPI Flash Size : 4MB
I (62) boot: Enabling RNG early entropy source...
I (68) boot: Partition Table:
I (71) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (86) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (93) boot:  2 factory          factory app      00 00 00010000 00300000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x25934 (153908) map
I (146) esp_image: segment 1: paddr=0x0003595c vaddr=0x3ffc9780 size=0x03334 ( 13108) load
I (150) esp_image: segment 2: paddr=0x00038c98 vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at /home/ddouard/src/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (153) esp_image: segment 3: paddr=0x000390a4 vaddr=0x40024404 size=0x06f74 ( 28532) load
I (170) esp_image: segment 4: paddr=0x00040020 vaddr=0x40080020 size=0x9e7d4 (649172) map
0x40080020: _stext at ??:?

I (306) esp_image: segment 5: paddr=0x000de7fc vaddr=0x4002b378 size=0x0e404 ( 58372) load
0x4002b378: xRingbufferSendFromISR at /home/ddouard/src/esp/esp-idf/components/esp_ringbuf/ringbuf.c:1087

I (334) boot: Loaded app from partition at offset 0x10000
I (334) boot: Disabling RNG early entropy source...
I (334) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (342) cpu_start: Pro cpu up.
I (345) cpu_start: Application information:
I (350) cpu_start: Project name:     play_http_mp3
I (356) cpu_start: App version:      1
I (360) cpu_start: Compile time:     Jan 24 2021 18:48:17
I (366) cpu_start: ELF file SHA256:  931d8e6459ebc54a...
I (372) cpu_start: ESP-IDF:          v4.2
I (377) cpu_start: Single core mode
I (381) heap_init: Initializing. RAM available for dynamic allocation:
I (388) heap_init: At 3FFD1050 len 0002AFB0 (171 KiB): DRAM
I (395) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (401) cpu_start: Pro cpu start user code
I (464) spi_flash: detected chip: generic
I (465) spi_flash: flash io: dio
I (465) cpu_start: Starting scheduler on PRO CPU.
I (477) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip
I (657) DRV8311: ES8311 in Slave mode
I (1047) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1207) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (1437) HTTP_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (1437) HTTP_MP3_EXAMPLE: [2.1] Create http stream to read data
I (1447) HTTP_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (1457) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1467) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1477) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (1487) HTTP_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (1487) MP3_DECODER: MP3 init
I (1497) HTTP_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (1507) HTTP_MP3_EXAMPLE: [2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (1517) AUDIO_PIPELINE: link el->rb, el:0x3ffd6578, tag:http, rb:0x3ffd8fc0
I (1517) AUDIO_PIPELINE: link el->rb, el:0x3ffd8c48, tag:mp3, rb:0x3ffde0fc
I (1527) HTTP_MP3_EXAMPLE: [2.6] Set up  uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)
I (1537) HTTP_MP3_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
I (1557) wifi:wifi driver task: 3ffe3268, prio:23, stack:6656, core=0
I (1557) system_api: Base MAC address is not set
I (1557) system_api: read default base MAC address from EFUSE
I (1567) wifi:wifi firmware version: 1865b55
I (1567) wifi:wifi certification version: v7.0
I (1577) wifi:config NVS flash: enabled
I (1577) wifi:config nano formating: disabled
I (1577) wifi:Init data frame dynamic rx buffer num: 32
I (1587) wifi:Init management frame dynamic rx buffer num: 32
I (1587) wifi:Init management short buffer num: 32
I (1597) wifi:Init dynamic tx buffer num: 32
I (1597) wifi:Init static rx buffer size: 1600
I (1607) wifi:Init static rx buffer num: 10
I (1607) wifi:Init dynamic rx buffer num: 32
I (1607) wifi_init: rx ba win: 6
I (1617) wifi_init: tcpip mbox: 32
I (1617) wifi_init: udp mbox: 6
I (1627) wifi_init: tcp mbox: 6
I (1627) wifi_init: tcp tx win: 5744
I (1627) wifi_init: tcp rx win: 5744
I (1637) wifi_init: tcp mss: 1440
I (1637) wifi_init: WiFi IRAM OP enabled
I (1647) wifi_init: WiFi RX IRAM OP enabled
I (1787) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2
I (1787) wifi:enable tsf
I (1787) wifi:mode : sta (xx:xx:xx:xx:xx:xx)
I (3837) wifi:new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
I (4537) wifi:state: init -> auth (b0)
I (4547) wifi:state: auth -> assoc (0)
I (4557) wifi:state: assoc -> run (10)
I (4567) wifi:connected with xxxx, aid = 2, channel 13, BW20, bssid = xx:xx:xx:xx:xx:xx
I (4567) wifi:security: WPA2-PSK, phy: bgn, rssi: -53
I (4567) wifi:pm start, type: 1
W (4567) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (4617) wifi:AP's beacon interval = 102400 us, DTIM period = 2
```
So the WPA2 handcheck seems to be ok, but then the call to `periph_wifi_wait_for_connected()` blocks because it never reaches  the expected connected state (apparently), whereas a WIFI even id 4 (which should be the enum value WIFI_EVENT_STA_CONNECTED from `esp-idf/components/esp_wifi/include/esp_wifi_types.h`)

douardda, 2021-01-24:
So it looks to me that for some reason the dhcp config is not executed. If I read esp_wifi.h, this is normally done by calling `tcpip_adapter_start()` which is normally done by default wifi event task on the SYSTEM_EVENT_STA_CONNECTED, the one that generates the "PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4"

douardda, 2021-01-24:
Oh this has been reported on the forum 
  https://esp32.com/viewtopic.php?f=20&t=18545&sid=60d9c7ab953680531ef804fd51523e99

douardda, 2021-01-25:
Other note: in the cli example there is:

```
#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 1, 0))
    ESP_ERROR_CHECK(esp_netif_init());
#else
    tcpip_adapter_init();
#endif
```
so with my version 4.2 of IDF, the `esp_netif_init()` is called. However using this the wifi does not connect (same as described above), and I need to explicitly call `tcpip_adapter_init();` to make it work.

So there really is something funky going on there with the new esp_netif API.

jason-mao, 2021-01-27:
@douardda Thanks for your report.

closed 2021-01-27 by espressif-bot

douardda, 2021-01-27:
thanks @jason-mao

mparker-landt, 2021-03-01:
Can anyone help me as I am still getting this error when I try to use the "Play mp3 file from HTTP" example.
I added the tcpip_adapter_init line as suggested but my program still times out on periph_wifi_wait_for_connected. The code I changed as shown:

----- snip ------
esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
periph_wifi_cfg_t wifi_cfg = {
.ssid = CONFIG_WIFI_SSID,
.password = CONFIG_WIFI_PASSWORD,
};
tcpip_adapter_init(); // <----- ADD THIS LINE
esp_periph_handle_t wifi_handle = periph_wifi_init(&wifi_cfg);
esp_periph_start(set, wifi_handle);
periph_wifi_wait_for_connected(wifi_handle, portMAX_DELAY);
----- snip ------