espressif/esp-adf #565: Remove the esp-idf submodule from the esp-adf repository

douardda, 2021-01-25:
it looks to me that this is not used any more (in favor of an externally installed IDF environment, as described in the getting-started manual.)

So keeping this subrepo as part of the esp-adf is very confusing on which version if IDF and idf libraries are actually used when developing an adf project.

(see #563)

HengYongChao, 2021-01-26:
Hello, @douardda 

I can explain why the built-in IDF version is lagging. 

When a release of ADF is released, the built-in IDF version is relatively new, but with the passage of time, the built-in IDF version will slowly lag behind the latest version of IDF. 

However, this is not a problem, because in the same period, IDF will have several releases of different versions, which are used by customers who have already mass-produced. The mass-produced products tend not to introduce new changes, but only accept bugfix.

Of course, a new IDF version will be introduced as a built-in version in a future adf release.

Also, you can read IDF version update rules and related information, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html

I hope these answers will answer your questions.

douardda, 2021-01-26:
Thanks @HengYongChao 

I understand, but I still think this is unnecessary (and confusing). The fact a release of ADF depends on a specific (tested) version of IDF is expected, but it is a "deployment" problem, not a development one, so there is no need to obfuscate the development git repo with it. 

For example, in most projects that manages debian packages along with the source code, the debian packaging files are not present in the main branch, but in a dedicated "debian" branch; you let the distribution related code (here the debian package scaffolding) isolated.

In the case of ADF, the fact that the development setup embeds a dependency that is ignored (under normal conditions) because a newer version is preferred is confusing. 

So my 2cts is either remove it entirely (my preferred choice) and document how to get and install the proper version of IDF in the README file, or at least add a paragraph in the getting-started guide explaining this ADF subrepo is ignored (by default).

HengYongChao, 2021-01-26:
Hi @douardda 

The current built-in IDF is more of an ADF dependency, so when I clone the entire adf repository, I can immediately compile my demo directly using the built-in IDF, which is not the latest IDF version sometimes of course.

For customers who are already familiar with the espressif compilation process, they are free to choose the version of IDF they want, but the built-in version of IDF provides the last alternate option.

Thank you for your suggestion, we'll think your suggestion seriously.

closed 2022-02-08 by jason-mao