espressif/esp-adf #567: korvo_du1906编译问题

lindabang, 2021-01-26:
问题：esp-adf/examples/korvo_du1906/components/bdsc_engine/lib 下的.a库文件使用了绝对路径（/home/donglianghao/Baidu/baidu_dul1906/dul1906/components/bdsc_engine/bdsc_engine.c），这样在其他人的机器上无法Link 

以下是详细的编译日志：
bonddeMacBook-Pro:korvo_du1906 bond$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/bond/esp/esp-adf/examples/korvo_du1906/build
Executing "ninja all"...
[6/21] Building C object esp-idf/audio_player/CMakeFiles/__idf_audio_player.dir/audio_player_helper_default.c.obj
../components/audio_player/audio_player_helper_default.c: In function 'setup_esp_audio_instance':
../components/audio_player/audio_player_helper_default.c:134:5: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
     i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT();
     ^~~~~~~~~~~~~~~~
In file included from /Users/bond/esp/esp-idf/components/hal/esp32/include/hal/i2s_ll.h:30,
                 from /Users/bond/esp/esp-idf/components/hal/include/hal/i2s_hal.h:28,
                 from /Users/bond/esp/esp-idf/components/driver/include/driver/i2s.h:24,
                 from /Users/bond/esp/esp-adf/components/audio_stream/include/i2s_stream.h:28,
                 from ../components/audio_player/audio_player_helper_default.c:39:
/Users/bond/esp/esp-idf/components/hal/include/hal/i2s_types.h:70:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
[8/21] Building C object esp-idf/audio_player/CMakeFiles/__idf_audio_player.dir/audio_player_pipeline_int_tone.c.obj
../components/audio_player/audio_player_pipeline_int_tone.c: In function 'audio_player_int_tone_init':
../components/audio_player/audio_player_pipeline_int_tone.c:76:5: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
     i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
     ^~~~~~~~~~~~~~~~
In file included from /Users/bond/esp/esp-idf/components/hal/esp32/include/hal/i2s_ll.h:30,
                 from /Users/bond/esp/esp-idf/components/hal/include/hal/i2s_hal.h:28,
                 from /Users/bond/esp/esp-idf/components/driver/include/driver/i2s.h:24,
                 from /Users/bond/esp/esp-adf/components/audio_stream/include/i2s_stream.h:28,
                 from ../components/audio_player/audio_player_pipeline_int_tone.c:36:
/Users/bond/esp/esp-idf/components/hal/include/hal/i2s_types.h:70:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
[9/21] Performing build step for 'bootloader'
ninja: no work to do.
[10/19] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_player_init.c.obj
../main/app_player_init.c: In function 'app_player_init':
../main/app_player_init.c:152:5: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
     i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT();
     ^~~~~~~~~~~~~~~~
In file included from /Users/bond/esp/esp-idf/components/hal/esp32/include/hal/i2s_ll.h:30,
                 from /Users/bond/esp/esp-idf/components/hal/include/hal/i2s_hal.h:28,
                 from /Users/bond/esp/esp-idf/components/driver/include/driver/i2s.h:24,
                 from /Users/bond/esp/esp-adf/components/audio_board/include/board_pins_config.h:29,
                 from /Users/bond/esp/esp-adf/components/audio_board/esp32_korvo_du1906/board.h:30,
                 from ../main/app_player_init.c:25:
/Users/bond/esp/esp-idf/components/hal/include/hal/i2s_types.h:70:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
[18/19] Linking CXX executable korvo_du1906.elf
FAILED: korvo_du1906.elf 
: && /Users/bond/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  CMakeFiles/korvo_du1906.elf.dir/project_elf_src_esp32.c.obj -o korvo_du1906.elf  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/esp-sr/libesp-sr.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/audio_board/libaudio_board.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/dueros_service/libdueros_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/playlist/libplaylist.a  esp-idf/audio_flash_tone/libaudio_flash_tone.a  esp-idf/audio_player/libaudio_player.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/Users/bond/esp/esp-adf/examples/korvo_du1906/build/korvo_du1906.map  -fno-rtti  -fno-lto  esp-idf/unity/libunity.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/jsmn/libjsmn.a  -L /Users/bond/esp/esp-adf/examples/korvo_du1906/components/bds_private/lib  -lbds_private_info  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/audio_player/libaudio_player.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/esp_actions/libesp_actions.a  -L/Users/bond/esp/esp-adf/components/wifi_service/airkiss_config  -lairkiss_aes  esp-idf/dueros_service/libdueros_service.a  esp-idf/display_service/libdisplay_service.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp- [... cut ...]
/Users/bond/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../components/bdsc_engine/lib/libbdsc_engine.a(bdsc_cmd.o): in function `ws_connect_config':
/home/donglianghao/Baidu/baidu_dul1906/dul1906/components/bdsc_engine/bdsc_cmd.c:237: undefined reference to `bds_client_send'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
bonddeMacBook-Pro:korvo_du1906 bond$

HengYongChao, 2021-01-27:
Hi, @lindabang 

`korvo_du1906 ` example docs says,  you should use IDF branch `audio/stack_on_psram_v3.3`

Here: https://github.com/espressif/esp-adf/blob/master/examples/korvo_du1906/README.md
> IDF branch
> For now, we need to select audio IDF branch as follow.
> 
> cd $IDF_PATH
> git checkout master
> git pull
> git checkout audio/stack_on_psram_v3.3
> git submodule update --init --recursive
> The latest commit ID is 0c1859a5a5a5eded5af1b9423c290781e3973915

Please read readme.md first and use correct idf branch, and try again.

jason-mao, 2021-01-27:
Duplicate issue, closed.

closed 2021-01-27 by jason-mao