espressif/esp-adf #568: How to make playback of mp3 from SD card using esp32-s2?

sudheeshsmadhav, 2021-01-27:
Hi all,
Working on mp3 playback from sd card using esp32-s2. wroom is used. I'm able to create/save a text file and open/edit that file. but stuck at mp3 playback. 
requesting experienced suggestions and guidance. 
I will be happy if there is a sample code.

thanks in advance.

HengYongChao, 2021-01-27:
Hi, @sudheeshsmadhav 

Currently adf support sdio interface as fatfs dirver, but s2 has no sdio interface.

So you need to choice spi as fatfs interface, and app code do not need change.

sudheeshsmadhav, 2021-01-30:
@HengYongChao Sir, can you please provide a sample code?

sudheeshsmadhav, 2021-02-06:
Any updates? Highly appreciates your suggestions...

HengYongChao, 2021-02-08:
> BY default, the audio hal takes LyraT as board, when I change to Kaluga-1 it shows some errors. Any updates? Highly appreciates your suggestions...
> 
> esp-idf/main/CMakeFiles/__idf_main.dir/http_play_and_save_to_file.c.obj
> /root/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -DADF_VER="" -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I/root/esp/esp-idf/components/newlib/platform_include -I/root/esp/esp-idf/components/freertos/include -I/root/esp/esp-idf/components/freertos/xtensa/include -I/root/esp/esp-idf/components/heap/include -I/root/esp/esp-idf/components/log/include -I/root/esp/esp-idf/components/lwip/include/apps -I/root/esp/esp-idf/components/lwip/include/apps/sntp -I/root/esp/esp-idf/components/lwip/lwip/src/include -I/root/esp/esp-idf/components/lwip/port/esp32/include -I/root/esp/esp-idf/components/lwip/port/esp32/include/arch -I/root/esp/esp-idf/components/soc/include -I/root/esp/esp-idf/components/soc/src/esp32s2/. -I/root/esp/esp-idf/components/soc/src/esp32s2/include -I/root/esp/esp-idf/components/hal/esp32s2/include -I/root/esp/esp-idf/components/hal/include -I/root/esp/esp-idf/components/esp_rom/include -I/root/esp/esp-idf/components/esp_common/include -I/root/esp/esp-idf/components/esp_system/include -I/root/esp/esp-idf/components/xtensa/include -I/root/esp/esp-idf/components/xtensa/esp32s2/include -I/root/esp/esp-idf/components/esp32s2/include -I/root/esp/esp-idf/components/driver/include -I/root/esp/esp-idf/components/driver/esp32s2/include -I/root/esp/esp-idf/components/esp_ringbuf/include -I/root/esp/esp-idf/components/efuse/include -I/root/esp/esp-idf/components/efuse/esp32s2/include -I/root/esp/esp-idf/components/espcoredump/include -I/root/esp/esp-idf/components/esp_timer/include -I/root/esp/esp-idf/components/esp_ipc/include -I/root/esp/esp-idf/components/esp_pm/include -I/root/esp/esp-idf/components/soc/soc/esp32s2/include -I/root/esp/esp-idf/components/soc/soc/esp32s2/../include -I/root/esp/esp-idf/components/soc/soc/esp32s2/private_include -I/root/esp/esp-idf/components/vfs/include -I/root/esp/esp-idf/components/esp_wifi/include -I/root/esp/esp-idf/components/esp_wifi/esp32s2/include -I/root/esp/esp-idf/components/esp_event/include -I/root/esp/esp-idf/components/esp_netif/include -I/root/esp/esp-idf/components/esp_eth/include -I/root/esp/esp-idf/components/tcpip_adapter/include -I/root/esp/esp-idf/components/app_trace/include -I/root/esp/esp-idf/components/mbedtls/port/include -I/root/esp/esp-idf/components/mbedtls/mbedtls/include -I/root/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/root/esp/esp-idf/components/bootloader_support/include -I/root/esp/esp-idf/components/app_update/include -I/root/esp/esp-idf/components/spi_flash/include -I/root/esp/esp-idf/components/nvs_flash/include -I/root/esp/esp-idf/components/pthread/include -I/root/esp/esp-idf/components/wpa_supplicant/include -I/root/esp/esp-idf/components/wpa_supplicant/port/include -I/root/esp/esp-idf/components/wpa_supplicant/include/esp_supplicant -I/root/esp/esp-idf/components/asio/asio/asio/include -I/root/esp/esp-idf/components/asio/port/include -I/root/esp/esp-idf/components/cbor/port/include -I/root/esp/esp-idf/components/unity/include -I/root/esp/esp-idf/components/unity/unity/src -I/root/esp/esp-idf/components/unity/unity/extras/fixture/src -I/root/esp/esp-idf/components/cmock/CMock/src -I/root/esp/esp-idf/components/coap/port/include -I/root/esp/esp-idf/components/coap/port/include/coap -I/root/esp/esp-idf/components/coap/libcoap/include -I/root/esp/esp-idf/components/coap/libcoap/include/coap2 -I/root/esp/esp-idf/components/console -I/root/esp/esp-idf/components/nghttp/port/include -I/root/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/root/esp/esp-idf/components/esp-tls -I/root/esp/esp-idf/components/esp_adc_cal/include -I/root/esp/esp-idf/components/esp_gdbstub/include -I/root/esp/esp-idf/components/esp_hid/include -I/root/esp/esp-idf/components/tcp_transport/include -I/root/esp/esp-idf/components/esp_http_client/include -I/root/esp/esp-idf/components/esp_http_server/include -I/root/esp/esp-idf/components/esp_https_ota/include -I/root/esp/esp-idf/components/protobuf-c/protobuf-c -I/root/esp/esp-idf/components/protocomm/include/common -I/root/esp/esp-idf/components/protocomm/include/security -I/root/esp/esp-idf/components/protocomm/include/transports -I/root/esp/esp-idf/components/mdns/include -I/root/esp/esp-idf/components/esp_local_ctrl/include -I/root/esp/esp-idf/components/sdmmc/include -I/root/esp/esp-idf/components/esp_serial_slave_link/include -I/root/esp/esp-idf/components/esp_websocket_client/include -I/root/esp/esp-idf/components/expat/expat/expat/lib -I/root/esp/esp-idf/components/expat/port/include -I/root/esp/esp-idf/components/wear_levelling/include -I/root/esp/esp-idf/components/fatfs/diskio -I/root/esp/esp-idf/components/fatfs/vfs -I/root/esp/esp-idf/components/fatfs/src -I/root/esp/esp-idf/components/freemodbus/common/include -I/root/esp/esp-idf/components/idf_test/include -I/root/esp/esp-idf/components/idf_test/include/esp32s2 -I/root/esp/esp-idf/components/jsmn/include -I/root/esp/esp-idf/components/json/cJSON -I/root/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include -I/root/esp/esp-idf/components/libsodium/port_include -I/root/esp/esp-idf/components/mqtt/esp-mqtt/include -I/root/esp/esp-idf/components/openssl/include - [... cut ...]
> /root/esp/esp-idf/components/hal/include/hal/i2s_types.h:71:5: note: declared here
> I2S_COMM_FORMAT_I2S_MSB **attribute**((deprecated)) = 0x01, /_!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`_/
> ^~~~~~~~~~~~~~~~~~~~~~~
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1

HI , @sudheeshsmadhav 

The log provided shows some undeclared errors.

It may be an error such as not including a header file. You need to be careful about some dependency issues when porting.

sudheeshsmadhav, 2021-02-10:
the i2s header files seems to be OK.

sudheeshsmadhav, 2021-02-23:
esp32 works perfectly with internal DAC. While using the same code for Espressif ESP32-S2-Soala-1_V1.2, (MP3 Playback from spiffs) it gives error that the problem with i2s as my previous post. Still the issue not resolved. Any example code released so far? Play back from spiffs or from SD Card? Expecting a solution at the earliest. Thank you.

JVKran, 2021-02-25:
Hi there!

I've just encountered this issue as well, allbeit with another example (the "Play MP3 file from SD Card" to be precise). I ended up just using the sdspi host in the audio_board_sdcard_init function of my custom board. It ended up looking (and working) like this. Just like HengYongChao mentioned.

``` c++
esp_err_t audio_board_sdcard_init(esp_periph_set_handle_t set, periph_sdcard_mode_t mode){
    esp_err_t ret;

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = SDCARD_FORMAT_ON_MOUNT_FAIL,
        .max_files = SDCARD_OPEN_FILE_NUM_MAX,
        .allocation_unit_size = 16 * 1024
    };
    sdmmc_card_t* card;
    const char mount_point[] = SDCARD_MOUNT_POINT;
    ESP_LOGI(TAG, "Initializing SD card");

    gpio_set_pull_mode(SDCARD_MISO_GPIO, GPIO_PULLUP_ONLY);     // TODO: Replace with seperate pull-ups.
    gpio_set_pull_mode(SDCARD_MOSI_GPIO, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(SDCARD_SCLK_GPIO, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(SDCARD_CS_GPIO, GPIO_PULLUP_ONLY);

    ESP_LOGI(TAG, "Using SPI peripheral");

    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    spi_bus_config_t bus_cfg = {
        .mosi_io_num = SDCARD_MOSI_GPIO,
        .miso_io_num = SDCARD_MISO_GPIO,
        .sclk_io_num = SDCARD_SCLK_GPIO,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 4000,
    };
    ret = spi_bus_initialize(host.slot, &bus_cfg, SPI_DMA_CHAN);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize bus.");
        return ret;
    }

    sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
    slot_config.gpio_cs = SDCARD_CS_GPIO;
    slot_config.host_id = host.slot;

    ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                "If you want the card to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return ret;
    }
    return ret;
}
```

I hope this helps you and maybe others in the future as well!

Kind regards,

Jochem

sudheeshsmadhav, 2021-03-20:
Thanks @JVKran ,

It helps me. i2s shows some error while using the internal DAC. i think, i2s function is not portable to esp23s2 from esp32.

closed 2021-11-22 by sudheeshsmadhav