espressif/esp-adf #570: VOIP algorithm stream - Rec channel input delay

Patsjemoe, 2021-01-30:
Hi,
How to add a delay in the Rec channel input of the algorithm stream ?
In this case it would be an I2S stream at 48000.
Regards
Ludo

ahhfzhang, 2021-02-03:
Hi Ludo， 

 We have added an algorithm stream for the VoIP AEC feature (0b715aa5a01c6e999e5c98730e324325f6770c1d), hope you can try it. 
 
But we only support the ESP32-LyraT-Mini board, it is very difficult for software to copy data to control the delay of recording and playback, which will result in poor AEC performance of the ESP32-LyraT board.

Patsjemoe, 2021-02-03:
hi,
thank you for your feedback.
is it possible to tell why this is more difficult for lyrat then lyratmini ? 
understanding may help me to try with lyrat. (do not have mini)
regards
ludo

ahhfzhang, 2021-02-03:
Hi ludo 

ESP32-LyraT-Mini  schematic
[image]
The reference signal and the recording signal are respectively read in from the left channel and the right channel of the same I2S  at the same time, but ESP32-LyraT 4.3 need software to copy the I2S element input buffer as a reference signal, if ringbuf delay more than 16ms, the AEC may be invalid, the actual situation is that this delay will not be as stable as the ESP32-LyraT-Mini .

Patsjemoe, 2021-02-03:
Hi, thank you for your feedback.
Sorry but Iam still trying to understand.... Why don't you have to copy the i2s element input buffer in the lyrat- mini?
Regards
Ludo

Patsjemoe, 2021-02-04:
Hi Michael,
Looking to the changes you made, it looks as you changed to type 1 algo stream with LyraT mini. 
Is this correct? 
I cannot test your solution yet, as I only have lyrat and audio kit boards. Ordered a lyrat mini, but this will take some time to arrive. 

Regards

ahhfzhang, 2021-02-04:
Hi Ludo.

thanks for your support.

`Looking to the changes you made, it looks as you changed to type 1 algo stream with LyraT mini.
Is this correct?`  
 >>  Yes, we currently only support type 1 algo stream in VoIP.

Patsjemoe, 2021-02-05:
Hi Michael,
Espressif documents show es7234 adc as microphone input for lyrat-mini. Cannot find it on the Web. Where can this be found?
Is it correct that you put the microphone input in one i2s channel, and the i2s ref signal in the other channel?
Regards
Ludo

Patsjemoe, 2021-02-06:
hi Michael,
your document specifies mini board v1.1 selection in voip.c
can it be mini board v1.2 ?
regards
ludo

ahhfzhang, 2021-02-07:
Hi Ludo,

https://docs.espressif.com/projects/esp-adf/en/latest/get-started/get-started-esp32-lyrat-mini.html
https://dl.espressif.com/dl/schematics/SCH_ESP32-LYRAT-MINI_V1.2_20190605.pdf

Both v1.1 and v1.2 boards are the same.

Patsjemoe, 2021-02-07:
hi Michael,
I looked in more detail in the schematic, and now think I understand....but had to look in the complete schematic, not only the part with the es7243 you showed earlier. It shows that the output of the codec is going directly into the second analog input of the es7243, which makes that the output i2s1 has both rec and ref in the 2 channels that are fed into algo stream.
please confirm this is right.
Now I have to wait for the mini board to be shipped, so I can test your SW.
thanks for your help
regards
ludo

Patsjemoe, 2021-02-08:
Hi Michael, @jason-mao 
Some time ago, an application esp-adf/examples/advanced_examples/algorithm/main/algorithm_examples.c was made with AEC. I checked the file, and it seems you can select lyra-t  or lyrat-mini.
Can you tell me why it would be more difficult to use lyrat in voip, then in the mp3 example?
I have used this example as base for voip... without succes. 
Regards
Ludo

Patsjemoe, 2021-02-11:
Hi Michael,
I have an audio_element_set_multi_output_ringbuf(i2s_stream_writer, input_rb, 0).
Is it possible to define/ double the size of this buffer, and have the input_rb to start only at for example the middle? This could be a way to create a delay and match it.
Regards
Ludo

Patsjemoe, 2021-02-15:
Hi Michael,
I just received the mini board, but get following error when trying to flash : 

$ make flash monitor -j5
Toolchain path: c:/path/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/path/esp/esp-idf-v3.3.2/requirements.txt are satisfied.
App "voip_app" version: v2.2-dirty

Flashing binaries to serial port com5 (app at offset 0x10000)...
esptool.py v2.8
Serial port com5
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [/c/msys32/path/esp/esp-idf-v3.3.2/components/esptool_py/Makefile.projbuild:71: flash] Fout 2

I pushed boot, then reset push loose, then boot loose....
But it is not programming
any suggestion ?
thanks in advance.
regards
Ludo

Patsjemoe, 2021-02-15:
Hi Michael,
Received the mini board today and started up.
It loads OK ....W (3208) SIP: CHANGE STATE FROM 1, TO 2, :func: sip_register:1632
I then push play, and my smartphone receives the call, I accept, but then esp32  stops with following error : it says **When type1 is choosen, both these algorithms should be used_**..........but  it is  not specified what "both" means ?

I (346428) AUDIO_ELEMENT: [sip_dec] AEL_MSG_CMD_RESUME,state:1
I (346428) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1
I (346438) ADF_BIT_STREAM: The element is 0x3ffce620. The reserve data 2 is 0x0.
I (346448) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (346448) I2S_STREAM: AUDIO_STREAM_WRITER
I (346458) AUDIO_PIPELINE: Pipeline started
I (346458) VOIP_EXAMPLE: SIP player has been created
I (346468) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (346478) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (346488) I2S: APLL: Req RATE: 16000, real rate: 15999.986, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 40959
96.500, SCLK: 511999.562500, diva: 1, divb: 0
I (346498) LYRAT_MINI_V1_1: I2S1, MCLK output by GPIO0

----->>E (346498) ALGORITHM_STREAM: **_**When type1 is choosen, both these algorithms should be used_****

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e98bb  PS      : 0x00060f30  A0      : 0x800eaf62  A1      : 0x3ffce160
0x400e98bb: audio_element_set_tag at C:/msys32/path/esp/esp-adf-v2.2/components/audio_pipeline/
audio_element.c:806

A2      : 0x00000000  A3      : 0x3f4032e0  A4      : 0x00000044  A5      : 0x3ffce194
A6      : 0x00000000  A7      : 0x00000004  A8      : 0x3ffdae84  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x3ffdb1ac  A12     : 0x40082ec8  A13     : 0x3ffce120
0x40082ec8: _calloc_r at C:/msys32/home/ludok/esp/esp-idf-v3.3.2/components/newlib/syscalls.c:51

A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000005c  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff

regards
Ludo

Patsjemoe, 2021-02-16:
> Hi Michael,
> Received the mini board today and started up.
> It loads OK ....W (3208) SIP: CHANGE STATE FROM 1, TO 2, :func: sip_register:1632
> I then push play, and my smartphone receives the call, I [...]

Michael, 
Which idf and adf version did you test with?
Regards
Ludo

Patsjemoe, 2021-02-16:
> Michael,
> Which idf and adf version did you test with?
> Regards
> Ludo

Hi Michael,
I get a weard error : 
It refers to an address wich is not mine : 
/home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_speec (see below). What is happening here ?

failed to calloc real fft cos/sin windows table
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008469f  PS      : 0x00060f30  A0      : 0x80100989  A1      : 0x3ffe5d70
0x4008469f: fft_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_speec
h_enhancement/fft.c:762
 (inlined by) fftr_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_sp
eech_enhancement/fft.c:1050

A2      : 0x00000000  A3      : 0x3ffecfb0  A4      : 0x00000400  A5      : 0x3ffeefb8
A6      : 0x7f800000  A7      : 0x00000200  A8      : 0x3d000000  A9      : 0x00000017
A10     : 0x01000000  A11     : 0x3ffdd8ac  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00060520  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x40100958  LEND    : 0x4010096b  LCOUNT  : 0x00000000
0x40100958: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_ec
ho_cancellation/esp_aec3.c:110 (discriminator 3)

0x4010096b: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_ec
ho_cancellation/esp_aec3.c:108 (discriminator 3)

ahhfzhang, 2021-02-20:
Hi Ludo,

> I then push play, and my smartphone receives the call, I accept, but then esp32 stops with following error : it says When type1 is choosen, both these algorithms should be used_..........but it is not specified what "both" means ?

https://github.com/espressif/esp-adf/blob/28594759fc3873bb15a233fc73d4ad34bc0fd4aa/components/audio_stream/algorithm_stream.c#L610-L613

`you can see here, but did you modify the **algo_mask**? `

> Which idf and adf version did you test with?
    
commit : 
```
ADF: 88ce2d50
IDF: 722043f7
```

Patsjemoe, 2021-02-20:
Hi Michael,
Thanks.
I found this line, but it shows 3 items, not 2., AEC, AGC and NS. To which
2 does it refer? Or should it say 3?

Note :
I manually copied the 4 changed files, but can this be done automatically
with git?
Regards
Ludo

ahhfzhang, 2021-02-20:
Hi Ludo 

About `algo_mask` you can use the default values. 
https://github.com/espressif/esp-adf/blob/28594759fc3873bb15a233fc73d4ad34bc0fd4aa/components/audio_stream/include/algorithm_stream.h#L135

The ADF commit I use is closer to the master branch and the IDF commit was release/v3.3.2.

Patsjemoe, 2021-02-20:
Hi Michael,
How can I install that version?
(I just copied files into v2. 2)
Regards
Ludo

ahhfzhang, 2021-02-20:
```
git checkout master 
git pull 
git checkout 88ce2d5
git submodule update --init --recursive
```

Patsjemoe, 2021-02-20:
Hi Michael,
Did an update an ran idf 3.3.2, ADF 2.2.
Get following error : any suggestion ?
$ make flash monitor -j5
Toolchain path: c:/users/ludok/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
CC build/audio_stream/algorithm_stream.o
Python requirements from C:/msys32/home/ludok/esp/esp-idf-v3.3.2/requirements.txt are satisfied.
CC build/audio_stream/spiffs_stream.o
App "voip_app" version: v2.2-dirty
CC build/bootloader_support/src/bootloader_random.o
CC build/bt/bt.o
C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c: In function '_algo_open':
C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:169:42: error: implicit declaration of function 'aec_pro_create' [-Werror=implicit-function-declaration]
         _success &= ((algo->aec_handle = aec_pro_create(AEC_FRAME_LENGTH_MS, ALGORITHM_STREAM_DEFAULT_CHANNEL, ALGORITHM_STREAM_DEFAULT_AEC_MODE)) != NULL);
                                          ^
C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:169:40: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
         _success &= ((algo->aec_handle = aec_pro_create(AEC_FRAME_LENGTH_MS, ALGORITHM_STREAM_DEFAULT_CHANNEL, ALGORITHM_STREAM_DEFAULT_AEC_MODE)) != NULL);
                                        ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/c/msys32/home/ludok/esp/esp-idf-v3.3.2/make/component_wrapper.mk:290: algorithm_stream.o] Fout 1
make[1]: *** Wachten op onvoltooide taken...

thanks in advance for your help.
Regards
Ludo

ahhfzhang, 2021-02-22:
Hi, ludo

You need to update the `esp-sr` submodule to the latest version.

But I still think you should update the ADF to the master branch.

Patsjemoe, 2021-02-22:
Hi Michael,
I updated the changed files in esp-sr (manually, because instructions did
not work...see below).  Can you advise how to be sure I have a correct
update ?

$ git checkout master
error: Your local changes to the following files would be overwritten by
checkout:
        components/audio_stream/algorithm_stream.c
        components/audio_stream/include/algorithm_stream.h
        examples/advanced_examples/voip/main/voip_app.c
Please commit your changes or stash them before you switch branches.
Aborting <<<---------

$ git pull
You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

$ git checkout 88ce2d5
error: Your local changes to the following files would be overwritten by
checkout:
        components/audio_stream/algorithm_stream.c
        components/audio_stream/include/algorithm_stream.h
        examples/advanced_examples/voip/main/voip_app.c
Please commit your changes or stash them before you switch branches.
Aborting

ahhfzhang, 2021-02-22:
You can force updates to the master branch with `git checkout master -f`, but your local changes would be overwritten.

Patsjemoe, 2021-02-22:
Hi Michael,
Updates worked , but still get the error that it refers to your PC Path.
****************************************************************************************************************************
failed to calloc real fft cos/sin windows table
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008469f  PS      : 0x00060d30  A0      : 0x801003fd  A1      : 0x3ffe5d70
0x4008469f: fft_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_spee
ch_enhancement/fft.c:762
 (inlined by) fftr_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_s
peech_enhancement/fft.c:1050

A2      : 0x00000000  A3      : 0x3ffecf44  A4      : 0x00000400  A5      : 0x3ffeef4c
A6      : 0x7f800000  A7      : 0x00000200  A8      : 0x3d000000  A9      : 0x00000017
A10     : 0x01000000  A11     : 0x3ffdd8a8  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00060320  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x401003cc  LEND    : 0x401003df  LCOUNT  : 0x00000000
0x401003cc: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_e
cho_cancellation/esp_aec3.c:110 (discriminator 3)

0x401003df: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_e
cho_cancellation/esp_aec3.c:108 (discriminator 3)

ELF file SHA256: 6f7eae724aeb705c1a78c23fdf565667508b35b6c8490f30133bff32bfe164db

Backtrace: 0x4008469f:0x3ffe5d70 0x401003fa:0x3ffe5da0 0x400fef64:0x3ffe5df0 0x400ebfeb:0x3ffe5e10 0x400ec245:0x3ffe5e30 0x400ec635
:0x3ffe6260 0x400ea1f7:0x3ffe6280 0x400ea370:0x3ffe62b0 0x40091d5e:0x3ffe62e0
0x4008469f: fft_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_spee
ch_enhancement/fft.c:762
 (inlined by) fftr_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_s
peech_enhancement/fft.c:1050

0x401003fa: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_e
cho_cancellation/esp_aec3.c:112 (discriminator 2)

0x400fef64: aec_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/esp_aec.c:71

0x400ebfeb: algorithm_process at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:574

0x400ec245: algorithm_data_process_for_type1 at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:57
4

0x400ec635: _algo_process at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:574

0x400ea1f7: audio_element_process_running at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_pipeline/audio_element.c:826

0x400ea370: audio_element_task at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_pipeline/audio_element.c:826

0x40091d5e: vPortTaskWrapper at C:/msys32/home/ludok/esp/esp-idf-v3.3.2/components/freertos/port.c:403

Rebooting...

*****************************************************************************************************************************************
update information : 
$ git checkout master -f
Previous HEAD position was d32cf8a Merge branch 'bugfix/crash_from_get_info' int                                                   o 'master'
Switched to branch 'master'
Your branch is up to date with 'origin/master'.

ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
$ git pull
Already up to date.

ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
$ git checkout 88ce2d5
M       components/esp-adf-libs
M       components/esp-sr
Note: checking out '88ce2d5'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 88ce2d5 fix codec es8311 and es7243 i2c bus write failed issue

ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
$ git submodule update --init --recursive
Submodule path 'components/esp-adf-libs': checked out '59f531061f492eeb17894c100                                                   ac044afc554a4ab'
error: Your local changes to the following files would be overwritten by checkou                                                   t:
        acoustic_algorithm/include/esp_aec.h
        acoustic_algorithm/libesp_audio_processor.a
Please commit your changes or stash them before you switch branches.
Aborting
Unable to checkout 'c5896943ea278195968c93c8b3466c720e641ebc' in submodule path 'components/esp-sr'

ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
$ git submodule update --init --recursive -f
Submodule path 'components/esp-adf-libs': checked out '59f531061f492eeb17894c100ac044afc554a4ab'
Submodule path 'components/esp-sr': checked out 'c5896943ea278195968c93c8b3466c720e641ebc'
Submodule path 'esp-idf': checked out '722043f734fa556d66d57473ac266fb1d0ec5ad2'
...and a lot more in esp-idf.

Patsjemoe, 2021-02-23:
Michael,
In which file do you define where to look for the files it is looking for?

home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_spee

Patsjemoe, 2021-02-25:
Michael,
I thought this would be easy, as you specify a path to files on your pc...., which of course cannot be found on my pc..,. So it reboots. 
In which file can I change this?
Regards
Ludo

ahhfzhang, 2021-02-25:
Hi Ludo,

These files are not on my computer, but I consulted its owner, he said that maybe some default AEC parameters have been modified, and I can't reproduce this crash.

Patsjemoe, 2021-02-25:
Hi Michael,
Maybe try on someone else's computer?
Are there any files that I should verify?
Regards
Ludo

Patsjemoe, 2021-02-25:
Hi MIchael,
Just to be sure it is clear : it is not at compile time, but when the program is fully loaded and I push the play button..
Regards
Ludo

Patsjemoe, 2021-02-25:
Michael,
I found the path in libesp_audio_processor.a by hexedit the file : 
�����/home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor�/home/zhaoyang/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/machine�/home/zhaoyang/.espressif/tools/xtensa-esp32-elf/esp

Patsjemoe, 2021-02-25:
Michael,
when looking in the file, it appears to only to refer to esp-2020r3-8.4.0
I am using 1.22.0-80-g6c4433a5-5.2.0
could this be the issue ?
does esp-2020r3-8.4.0 work with msys32, make ?

ahhfzhang, 2021-02-25:
Hi ludo , 

No, ADF 2.2 don't support this feature, and I also use `gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) `

Patsjemoe, 2021-02-25:
Which version('s) do I have to use?

Patsjemoe, 2021-02-26:
Michael,
Please confirm which IDF and Adf version I must use.

Patsjemoe, 2021-02-26:
> libesp_audio_processor.a

Michael,
Can you confirm what is happening in the libesp_audio_processor.a file ? There is clearly a link with the problem ...

ahhfzhang, 2021-03-01:
Hi Ludo,

[https://github.com/espressif/esp-sr]() 
You can ask questions about the libesp_audio_processor library issue here.

IDF and Adf version is here or you can use ADF master branch.
https://github.com/espressif/esp-adf/issues/570#issuecomment-782552031

Patsjemoe, 2021-03-01:
> #570 (comment)

By version, do you mean?

ADF: 88ce2d50
IDF: 722043f7

I installed these, but still get the error.

I did following : can you confirm if this is then the correct version, or

If I still need to do something more ?

 in ADF v2.2 :
git checkout master
git pull
git checkout 88ce2d5
<https://github.com/espressif/esp-adf/commit/88ce2d50beb9611be5b5d595747f728ff88ff210>
git submodule update --init --recursive

and  in IDF v3.3.2 :
git checkout master
git pull
git checkout 722042f7
git submodule update --init --recursive

Regards

Ludo

jason-mao, 2021-03-01:
Hi Ludo，

> By version, do you mean?
> 
> ADF: 88ce2d5
> IDF: 722043f7
> 
> I installed these, but still get the error.

Is follow crash happened after your build firmware? Did you enable the PSRAM configurations?

> Hi Michael,
> Updates worked , but still get the error that it refers to your PC Path.
> 
> failed to calloc real fft cos/sin windows table
> Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
> Core 1 register dump:
> PC : 0x4008469f PS : 0x00060d30 A0 : 0x801003fd A1 : 0x3ffe5d70
> 0x4008469f: fft_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_spee
> ch_enhancement/fft.c:762
> (inlined by) fftr_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_s
> peech_enhancement/fft.c:1050
> 
> A2 : 0x00000000 A3 : 0x3ffecf44 A4 : 0x00000400 A5 : 0x3ffeef4c
> A6 : 0x7f800000 A7 : 0x00000200 A8 : 0x3d000000 A9 : 0x00000017
> A10 : 0x01000000 A11 : 0x3ffdd8a8 A12 : 0x00000000 A13 : 0x00000001
> A14 : 0x00060320 A15 : 0x00000001 SAR : 0x0000001d EXCCAUSE: 0x0000001c
> EXCVADDR: 0x0000000c LBEG : 0x401003cc LEND : 0x401003df LCOUNT : 0x00000000
> 0x401003cc: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_e
> cho_cancellation/esp_aec3.c:110 (discriminator 3)
> 
> 0x401003df: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_e
> cho_cancellation/esp_aec3.c:108 (discriminator 3)
> 
> ELF file SHA256: 6f7eae724aeb705c1a78c23fdf565667508b35b6c8490f30133bff32bfe164db
> 
> Backtrace: 0x4008469f:0x3ffe5d70 0x401003fa:0x3ffe5da0 0x400fef64:0x3ffe5df0 0x400ebfeb:0x3ffe5e10 0x400ec245:0x3ffe5e30 0x400ec635
> :0x3ffe6260 0x400ea1f7:0x3ffe6280 0x400ea370:0x3ffe62b0 0x40091d5e:0x3ffe62e0
> 0x4008469f: fft_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_spee
> ch_enhancement/fft.c:762
> (inlined by) fftr_esp_proc at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/mic_array_s
> peech_enhancement/fft.c:1050
> 
> 0x401003fa: esp_aec3_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/acoustic_e
> cho_cancellation/esp_aec3.c:112 (discriminator 2)
> 
> 0x400fef64: aec_process at /home/zhaoyang/esp/tmp1/esp_sr_lib/home/madcow/esp_sr_lib/components/esp_audio_processor/esp_aec.c:71
> 
> 0x400ebfeb: algorithm_process at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:574
> 
> 0x400ec245: algorithm_data_process_for_type1 at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:57
> 4
> 
> 0x400ec635: _algo_process at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_stream/algorithm_stream.c:574
> 
> 0x400ea1f7: audio_element_process_running at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_pipeline/audio_element.c:826
> 
> 0x400ea370: audio_element_task at C:/msys32/home/ludok/esp/esp-adf-v2.2/components/audio_pipeline/audio_element.c:826
> 
> 0x40091d5e: vPortTaskWrapper at C:/msys32/home/ludok/esp/esp-idf-v3.3.2/components/freertos/port.c:403
> 
> Rebooting...
> 
> update information :
> $ git checkout master -f
> Previous HEAD position was [d32cf8a](https://github.com/espressif/esp-adf/commit/d32cf8a4eeeadb99cb11d48a064a5020bf2b899a) Merge branch 'bugfix/crash_from_get_info' int o 'master'
> Switched to branch 'master'
> Your branch is up to date with 'origin/master'.
> 
> ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
> $ git pull
> Already up to date.
> 
> ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
> $ git checkout [88ce2d5](https://github.com/espressif/esp-adf/commit/88ce2d50beb9611be5b5d595747f728ff88ff210)
> M components/esp-adf-libs
> M components/esp-sr
> Note: checking out '88ce2d5'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
> git checkout -b
> 
> HEAD is now at [88ce2d5](https://github.com/espressif/esp-adf/commit/88ce2d50beb9611be5b5d595747f728ff88ff210) fix codec es8311 and es7243 i2c bus write failed issue
> 
> ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
> $ git submodule update --init --recursive
> Submodule path 'components/esp-adf-libs': checked out '59f531061f492eeb17894c100 ac044afc554a4ab'
> error: Your local changes to the following files would be overwritten by checkou t:
> acoustic_algorithm/include/esp_aec.h
> acoustic_algorithm/libesp_audio_processor.a
> Please commit your changes or stash them before you switch branches.
> Aborting
> Unable to checkout 'c5896943ea278195968c93c8b3466c720e641ebc' in submodule path 'components/esp-sr'
> 
> ludok@DESKTOP-UHANLJM MINGW32 ~/esp/esp-adf-v2.2
> $ git submodule update --init --recursive -f
> Submodule path 'components/esp-adf-libs': checked out '59f531061f492eeb17894c100ac044afc554a4ab'
> Submodule path 'components/esp-sr': checked out 'c5896943ea278195968c93c8b3466c720e641ebc'
> Submodule path 'esp-idf': checked out '722043f734fa556d66d57473ac266fb1d0ec5ad2'
> ...and a lot more in esp-idf.

Patsjemoe, 2021-03-01:
Hi Michael,
1) As I wrote before : the build is OK, the program loads fine. Then I push the play button, it calls
   I (47099) VOIP_EXAMPLE:  SIP recorder has been created
   I (47109) SIP_RTP: receive task, local addr=192.168.0.165:31546
   :1200609) SIP_RTP: send task, remote addr=192.168.0.203
    and then the error happens.

   failed to calloc real fft cos/sin windows table......

2) I  did not enable the PSRAM,  where do you enable this ?

Patsjemoe, 2021-03-01:
Hi Michael,
enabled PSRAM and it solved the issue :-)).
Now I will test it.
Thanks al lot already for your help.
regards
Ludo

ahmadfayed7, 2021-03-02:
hi patsjemoe 
I would like if you give me some hint on how to connect my esp32 lyrat to freepbx (VOIP) and connect my android phone with that too
please contact me( ahmadfayed7@gmail.com)

Patsjemoe, 2021-03-02:
Hi,
I am not using freepbx, but Asterisk.
Regards

Patsjemoe, 2021-03-08:
Hi Michael,
I tested VOIP with miniboard, but I still get an echo

Patsjemoe, 2021-03-30:
Hi Michael,
Adding a buffer is not working, and no solution given. 
Therefore, need to try again with this suggested solution whee I also have echo. 
Can you confirm that in your test you had no echo at all?
Regards
Ludo

ahhfzhang, 2021-03-30:
Hi Ludo,

We have a new solution for this.
[libesp_audio_processor.zip](https://github.com/espressif/esp-adf/files/6227495/libesp_audio_processor.zip)
Can you try to replace this lib to `esp-adf-internal\components\esp-sr\acoustic_algorithm` and then try again.

PS. The volume better not exceed 60, otherwise there will be a popping sound during the call, at this time, the echo will also be heard.

Patsjemoe, 2021-04-05:
Hi Michael,
LyraT mini seems to work. 
Should this also work with LyraT? 
Regards

closed 2021-04-07 by Patsjemoe

expresspotato, 2021-04-11:
So I've spent the afternoon trying to get the AEC example working with the LyraT board with mixed results...

1) The Algorithm Stream does work but only in Type 1 mode. I couldn't figure out how to separate I2S channels 1 and 2 before piping them to the algorithm, so maybe someone can help with that.

2) With Type 1, you cannot disable AGC or NS so the customer voice is much softer and actually less clear than with it.

3) Noise Suppression (NS) doesn't make a ton of sense to me with only one input source, shouldn't it have two to cancel out the noise received by both microphones and try to preserve the audio that's only being received by the first?

For anyone that has more experience on this the best approach would probably be to separate the i2c channels and then set them with what I presume is:

`ringbuf_handle_t input_rb_algo = algo_stream_get_multi_input_rb(element_algo);
audio_element_set_multi_output_ringbuf(filter_ch_1, input_rb_algo, 0);
audio_element_set_multi_output_ringbuf(filter_ch_2, input_rb_algo, 1);`

This way we could customize based on Type 2.

Patsjemoe, 2021-04-11:
@expresspotato : is AEC working in type 1 mode?
Regards
Ludo

expresspotato, 2021-04-11:
Yes it is, but again see my caveats above. I don't know enough about programming with the audio pipeline to separate the i2s stream into 2 channels before setting them into a multi buffer for the algorithm to read... Or use the original explanation of i2c going into the Algorithm and somehow splitting the second stream that way. 

Since Type 1 reads the whole i2c stream anyways as part of the pipeline, it's annoying the Algorithm interface is designed to only read the 1 channel and we have to spit it to get Type 2. Record channel and Reference channel need to be the same in Type 1 (even though they shouldn't need to be on boards with 2 microphones). 

Here's my attempt, you can test AEC by calling your extension, accepting the call, go into a different room then speak. You shouldn't hear yourself. 

This is with the modified library provided by @ahhfzhang, not sure if it's required or not (didn't test without it). Also note enc_mode is set to 1 on G711 encoder and decoder, the example didn't check for uLaw or aLaw with it, resulting in distored audio. If you are using aLaw, change it accordingly. 

Still hoping there is someone on here so we can improve the Algorithm, it's quite rough in many regards.

`/* VoIP Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "esp_peripherals.h"
#include "periph_wifi.h"
#include "board.h"
#include "input_key_service.h"

#include "audio_mem.h"
#include "raw_stream.h"
#include "filter_resample.h"
#include "esp_sip.h"
#include "g711_decoder.h"
#include "g711_encoder.h"
#include "algorithm_stream.h"

#include "es8388.h"
#include "audio_alc.h"
#include "esp_pm.h"

#include "esp_sntp.h"

#if __has_include("esp_idf_version.h")
#include "esp_idf_version.h"
#else
#define ESP_IDF_VERSION_VAL(major, minor, patch) 1
#endif

#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 1, 0))
#include "esp_netif.h"
#else
#include "tcpip_adapter.h"
#endif

static const char *TAG = "VOIP_EXAMPLE";

#define I2S_SAMPLE_RATE     16000
#define I2S_CHANNELS        2
#define I2S_BITS            16

#define CODEC_SAMPLE_RATE    8000
[... 346 lines cut ...]

    ESP_LOGI(TAG, "[ 3 ] Create and start input key service");
    input_key_service_info_t input_key_info[] = INPUT_KEY_DEFAULT_INFO();
    input_key_service_cfg_t input_cfg = INPUT_KEY_SERVICE_DEFAULT_CONFIG();
    input_cfg.handle = set;
    periph_service_handle_t input_ser = input_key_service_create(&input_cfg);
    input_key_service_add_key(input_ser, input_key_info, INPUT_KEY_NUM);
    periph_service_set_callback(input_ser, input_key_service_cb, (void *)board_handle);

    ESP_LOGI(TAG, "[ 4 ] Create SIP Service");
    sip_config_t sip_cfg = {
        .uri = CONFIG_SIP_URI,
        .event_handler = _sip_event_handler,
        .send_options = true,
#ifdef CONFIG_SIP_CODEC_G711A
        .acodec_type = SIP_ACODEC_G711A,
#else
        .acodec_type = SIP_ACODEC_G711U,
#endif
    };
    sip = esp_sip_init(&sip_cfg);
    esp_sip_start(sip);
}

`

HengYongChao, 2021-04-12:
Hi @expresspotato 
If you use the latest code and find that the problem still exists, please reopen this issue.
If your have a different issue, please open a new issue to facilitate the discussion of issues by category.