espressif/esp-adf #571: Cannot record voice in ESP32-LyraT-mini

githubxup, 2021-02-01:
Dear All,

I cannot hear any voice in the recorded file "rec.amr" using ESP32-LyraT-mini after running pipeline_amr_sdcard example, but I successfully tried the same example in the ESP32-LyraT. I am sure that I have run idf.py menuconfig to change audio hal. Is there anything else I did do it correctly so I cannot record voice from ESP32-LyraT-mini?

James

HengYongChao, 2021-02-01:
Hi @githubxup 

Thanks for report this issue!

I have tried the following version(ADF:66eb22d3b, IDF: 722043f73), the problem does exist and will fix it.

githubxup, 2021-02-03:
After updating the source code, I got one compile error as following messages. I found that 'aec_pro_create' function is not exist in the esp_aec.h. It should be 'aec_create'.
I still cannot get any voice. Should I re-git all esp-adf?
Is there any differences between esp-adf built-in esp-idf and original esp-idf?

/home/glate/test/esp/esp-adf/components/audio_stream/algorithm_stream.c: In function '_algo_open':
/home/glate/test/esp/esp-adf/components/audio_stream/algorithm_stream.c:169:42: error: implicit declaration of function 'aec_pro_create'; did you mean 'aec_create'? [-Werror=implicit-function-declaration]
         _success &= ((algo->aec_handle = aec_pro_create(AEC_FRAME_LENGTH_MS, ALGORITHM_STREAM_DEFAULT_CHANNEL, ALGORITHM_STREAM_DEFAULT_AEC_MODE)) != NULL);
                                          ^~~~~~~~~~~~~~
                                          aec_create
/home/glate/test/esp/esp-adf/components/audio_stream/algorithm_stream.c:169:40: warning: assignment to 'void *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
         _success &= ((algo->aec_handle = aec_pro_create(AEC_FRAME_LENGTH_MS, ALGORITHM_STREAM_DEFAULT_CHANNEL, ALGORITHM_STREAM_DEFAULT_AEC_MODE)) != NULL);

HengYongChao, 2021-02-04:
Hi @githubxup 

I tried the idf version `idf release/v4.1 :9778b163b1`,  `idf release/v4.2 :8f20b9dc8a` and they all work normally.

What the IDF version do you use? please update the git commit info.

githubxup, 2021-02-05:
I use esp-idf version v4.3-dev-2586-g526f68239 and esp-adf version v2.2-58-g0b715aa.
I will try idf release/v4.2 version later and let you know.
Thanks a lot.

githubxup, 2021-02-10:
I still cannot hear any voice from recorded amr file after running pipeline_amr_sdcard example with esp-idf release/v4.2-238-g8cd16b60f and esp-adf v2.2-62-gb077d14. Is there anything wrong here? Maybe I got a problem board.

HengYongChao, 2021-02-18:
> I still cannot hear any voice from recorded amr file after running pipeline_amr_sdcard example with esp-idf release/v4.2-238-g8cd16b60f and esp-adf v2.2-62-gb077d14. Is there anything wrong here? Maybe I got a problem board.

Please try this commit https://github.com/espressif/esp-adf/issues/571#issuecomment-772966994

githubxup, 2021-02-18:
Please help how to clone the specified idf version release/v4.1 :9778b163b1, idf release/v4.2 :8f20b9dc8a.

HengYongChao, 2021-02-18:
> Please help how to clone the specified idf version release/v4.1 :9778b163b1, idf release/v4.2 :8f20b9dc8a.

In the idf folder, use `git pull` and  `git checkout 8f20b9dc8a` commands.

closed 2021-09-26 by jason-mao