espressif/esp-adf #575: board.h not found by compiler

orbitcoms, 2021-02-22:
I am trying to compile a project in vscode for ESP32 that uses adf framework. 
The custom board has been selected in menuconfig but when I compile I get error "no such file or directory #include board.h".

If I change the board to one of the models in the list in menuconfig and recompile, it compiles and finds "board.h" ok.

I have tried various versions of idf and reinstalled adf several times but still cannot get this resolved.

The board files are located in the "components" folder as below.

[image]

HengYongChao, 2021-02-23:
Hi, @orbitcoms 

Are you using your own project or adf example? Is the compilation method cmake or gnu make?

Please provide some more information, Thanks!

orbitcoms, 2021-02-23:
Thanks for your reply.
This is my own project, but I get the same issue when I compile example files in the adf/examples folder (like Play_mp3).
The code is being written in VScode and using cmake compiler I believe.
The version of idf is 4.1 and I installed latest version of adf. Python is version 3.8

HengYongChao, 2021-02-23:
> Thanks for your reply.
> This is my own project, but I get the same issue when I compile example files in the adf/examples folder (like Play_mp3).
> The code is being written in VScode and using cmake compiler I believe.
> The version of idf is 4.1 and I installed latest version of adf. Python is version 3.8
> 
> Is there any screenshot or log file etc I could provide that helps identify what might be the problem?

Well, I think you need modify you  cmakelist.txt, like: [Cmakelist.txt](https://github.com/espressif/esp-adf/blob/master/examples/get-started/play_mp3/components/my_board/CMakeLists.txt)
```
list(APPEND COMPONENT_ADD_INCLUDEDIRS ./my_board_v1_0 ./my_codec_driver)
set(COMPONENT_SRCS
./my_board_v1_0/board.c
./my_board_v1_0/board_pins_config.c
./my_codec_driver/new_codec.c
)
```
More detail in [Build System](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html).

orbitcoms, 2021-02-23:
Below is what my cmakelist looks like and also dos view of the idf components installed

[image]

[image]

HengYongChao, 2021-02-23:
Post the build log please.

orbitcoms, 2021-02-23:
Thanks,

Below is my build output.

C:\temp\orbit-david-audio-mp3>idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\temp\orbit-david-audio-mp3\build
Executing "ninja all"...
[4/188] cmd.exe /C "cd /D C:\temp\orbit-david-audio-mp3\build\esp-idf\partition_table && C:\esp\tools\.espressif\tool...s\cmake\3.13.4\bin\cmake.exe -E echo *******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
storage,data,spiffs,0x110000,1344K,
*******************************************************************************
[5/188] Performing build step for 'bootloader'
ninja: no work to do.
[6/186] Building C object esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/es7148/es7148.c.obj
FAILED: esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/es7148/es7148.c.obj
ccache C:\esp\tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DADF_VER=\"d69a72d-dirty\" -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IC:/esp/esp-adf/components/audio_hal/include -IC:/esp/esp-adf/components/audio_hal/driver/es8388 -IC:/esp/esp-adf/components/audio_hal/driver/es8374 -IC:/esp/esp-adf/components/audio_hal/driver/es8311 -IC:/esp/esp-adf/components/audio_hal/driver/es7243 -IC:/esp/esp-adf/components/audio_hal/driver/es7148 -IC:/esp/esp-adf/components/audio_hal/driver/tas5805m -IC:/esp/esp-adf/components/audio_hal/driver/zl38063 -IC:/esp/esp-adf/components/audio_hal/driver/zl38063/api_lib -IC:/esp/esp-adf/components/audio_hal/driver/zl38063/example_apps -IC:/esp/esp-adf/components/audio_hal/driver/zl38063/firmware -IC:/esp/esp-adf/components/audio_hal/driver/include -IC:/esp/esp-idf/components/newlib/platform_include -IC:/esp/esp-idf/components/freertos/include -IC:/esp/esp-idf/components/heap/include -IC:/esp/esp-idf/components/log/include -IC:/esp/esp-idf/components/lwip/include/apps -IC:/esp/esp-idf/components/lwip/include/apps/sntp -IC:/esp/esp-idf/components/lwip/lwip/src/include -IC:/esp/esp-idf/components/lwip/port/esp32/include -IC:/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/esp/esp-idf/components/lwip/port/esp32/tcp_isn -IC:/esp/esp-idf/components/soc/esp32/include -IC:/esp/esp-idf/components/soc/include -IC:/esp/esp-idf/components/esp_rom/include -IC:/esp/esp-idf/components/esp_common/include -IC:/esp/esp-idf/components/xtensa/include -IC:/esp/esp-idf/components/xtensa/esp32/include -IC:/esp/esp-idf/components/esp32/include -IC:/esp/esp-idf/components/driver/include -IC:/esp/esp-idf/components/driver/esp32/include -IC:/esp/esp-idf/components/esp_ringbuf/include -IC:/esp/esp-idf/components/efuse/include -IC:/esp/esp-idf/components/efuse/esp32/include -IC:/esp/esp-idf/components/vfs/include -IC:/esp/esp-idf/components/esp_wifi/include -IC:/esp/esp-idf/components/esp_wifi/esp32/include -IC:/esp/esp-idf/components/esp_event/include -IC:/esp/esp-idf/components/esp_netif/include -IC:/esp/esp-idf/components/esp_eth/include -IC:/esp/esp-idf/components/tcpip_adapter/include -IC:/esp/esp-idf/components/app_trace/include -IC:/esp/esp-adf/components/audio_sal/include -IC:/esp/esp-adf/components/audio_board/include -IC:/esp/esp-idf/components/mbedtls/port/include -IC:/esp/esp-idf/components/mbedtls/mbedtls/include -IC:/esp/esp-adf/components/esp_peripherals/include -IC:/esp/esp-adf/components/esp_peripherals/lib/adc_button -IC:/esp/esp-adf/components/esp_peripherals/lib/gpio_isr -IC:/esp/esp-adf/components/esp_peripherals/lib/sdcard -IC:/esp/esp-adf/components/esp_peripherals/lib/touch -IC:/esp/esp-adf/components/esp_peripherals/lib/button -IC:/esp/esp-adf/components/esp_peripherals/lib/blufi -IC:/esp/esp-adf/components/esp_peripherals/lib/IS31FL3216 -IC:/esp/esp-adf/components/esp_peripherals/lib/aw2013 -IC:/esp/esp-adf/components/esp_peripherals/driver/i2c_bus -IC:/esp/esp-idf/components/esp_adc_cal/include -IC:/esp/esp-idf/components/fatfs/diskio -IC:/esp/esp-idf/components/fatfs/vfs -IC:/esp/esp-idf/components/fatfs/src -IC:/esp/esp-idf/components/wear_levelling/include -IC:/esp/esp-idf/components/spi_flash/include -IC:/esp/esp-idf/components/sdmmc/include -IC:/esp/esp-idf/components/console -IC:/esp/esp-adf/components/audio_pipeline/include -IC:/esp/esp-adf/components/esp-adf-libs/esp_audio/include -IC:/esp/esp-adf/components/esp-adf-libs/esp_codec/include/codec -IC:/esp/esp-adf/components/esp-adf-libs/esp_codec/include/processing -IC:/esp/esp-adf/components/esp-adf-libs/esp_upnp/include -IC:/esp/esp-adf/components/esp-adf-libs/esp_ssdp/include -IC:/esp/esp-adf/components/esp-adf-libs/recorder_engine/include -IC:/esp/esp-adf/components/esp-adf-libs/audio_misc/include -IC:/esp/esp-adf/components/esp-adf-libs/esp_sip/include -IC:/esp/esp-adf/components/esp-sr/lib/include -IC:/esp/esp-adf/components/esp-sr/wake_word_engine/include -IC:/esp/esp-adf/components/esp-sr/speech_command_recognition/include -IC:/esp/esp-adf/components/esp-sr/acoustic_algorithm/include -IC:/esp/esp-adf/components/esp-sr/esp-tts/esp_tts_chinese/include -IC:/esp/esp-idf/components/nghttp/port/include -IC:/esp/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/esp/esp-idf/components/tcp_transport/include -IC:/esp/esp-idf/components/esp-tl [... cut ...] \CMakeFiles\__idf_audio_hal.dir\driver\tas5805m\tas5805m.c.obj.d -o esp-idf/audio_hal/CMakeFiles/__idf_audio_hal.dir/driver/tas5805m/tas5805m.c.obj   -c C:/esp/esp-adf/components/audio_hal/driver/tas5805m/tas5805m.c
C:/esp/esp-adf/components/audio_hal/driver/tas5805m/tas5805m.c:26:10: fatal error: board.h: No such file or directory
 #include "board.h"
          ^~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\temp\orbit-david-audio-mp3>

orbitcoms, 2021-02-23:
Any ideas at all?
I tried on a different PC at home and get same problem.
Is there any known issue with Anti-Virus maybe removing important files ? I run Norton 360 Lifelock with real time tuning.
Apart from that as a guess, I am all out of ideas.

HengYongChao, 2021-02-23:
hi @orbitcoms 

I think this is a project construction problem, if your project does not find the header file, then you should add header file path to the `cmakelist.txt` file, like: 
```
idf_component_register(INCLUDE_DIRS "include")
```
That would be work fine,  try create an` include `folder and put the header files in it, then try compile the codes.

orbitcoms, 2021-02-23:
Have I got the include in the correct cmakelists.txt (there are 3 in the project

[image]

HengYongChao, 2021-02-23:
Would you like to upload the source code of the minimal system? I can help you why board.h wasn't included.

orbitcoms, 2021-02-23:
Interestingly The same issue occours when I try to run the example application using a custom board. All I did was
* navigate to the `....\esp-adf\examples\get-started\play_mp3`
* `idf.py menuconfig`
* set AUDIO HAL -> Audio Board to "Custom Audio Board"
Compiled and got the same result
Do you think there may be some environmental difference?

HengYongChao, 2021-02-24:
Hi @orbitcoms 
** adf & idf environment：**
>    1. adf(28594759fc)
>    2. idf(9778b163b1)

And menuconfig:
> AUDIO HAL -> Audio Board to "Custom Audio Board"
> My Audio Board -> My audio board (My board v1.0) -> "My board v1.0"

Will got a error:
> examples/get-started/play_mp3/components/my_board/my_board_v1_0/board.h:88:64: error: unknown type name 'periph_sdcard_mode_t'; did you mean 'i2s_dac_mode_t'?

After add include `#include "periph_sdcard.h"`, it build successed.

orbitcoms, 2021-02-24:
I am lost to know what is happening.
I created a complete new environment Oracle virtual box and ran a windows development OS and got same problem with "board.h" not found. A colleague tried in a virtual environment same as mine and gets a different error again.

The new environment did not have Norton Anti-virus, so I don't think AV is affecting the compiling process. I wonder why you can compile the sample Paly_Mp3 and not get same error as my machine???

[image]

HengYongChao, 2021-02-24:
I'm use Ubuntu 18.04.5 lts environment, not windows.

closed 2021-02-24 by espressif-bot

spetroce, 2023-03-08:
> I had the same problem, i managed to solve it by removing the "IF (IDF_VER MATCHES "v4.")" from CMakeLists.txt since i know that i'm using a 4.1 IDF version
> 
> [image]

We had the same issue with build from an IDF Docker image. The /opt/esp/idf/tools/cmake/build.cmake file creates the definition and cmake variable IDF_VER, which is used in a handful of places. The Espressif Docker image (at least for v4.4.4) does not provide the version.txt file needed to set the IDF_VER variable. It can alternatively get the version from 'git describe', but the Docker images do not ship with a .git folder so this is not possible either. The result is that the string HEAD-HASH-NOTFOUND is assigned to IDF_VER and this breaks compilation. So we manually created the version.txt file. 

This is also described [here](https://community.platformio.org/t/reporting-esp-idf-version/14590).