espressif/esp-adf #576: unable to use local ADF components  (replacing ADF components)

derflaryam, 2021-02-23:
I have copied two ADF components audio_board and audio_hal to the local components directory and added the  EXTRA_COMPONENT_DIRS pointing to the project components directory. 
According to the documents, if there is a component with the same name in the IDF/ADF and in the project components directory, the component should be taken from the project components directory. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#multiple-components-with-the-same-name
BUT – it still takes the components from the IDF/ADF installation. 
I have provided idf.py reconfigure output, compile_commands.json and the two CMakeLists.txt. 

Please describe what I need to change, that I can work with the local ADF components.
Thank you  
[adf_component_replace.zip](https://github.com/espressif/esp-adf/files/6027476/adf_component_replace.zip)

HengYongChao, 2021-02-24:
Hi @derflaryam 

> If a component is overridden in an existing project by moving it to a new location, the project will not automatically see the new component path. Run idf.py reconfigure (or delete the project build folder) and then build again.

Use command ` idf.py reconfigure`, and try again?

derflaryam, 2021-02-24:
This is exact what I did ! ! ! 

I was even mentioning it ....    I have provided idf.py reconfigure output in the original request . 

I provided the output from the idf.py reconfigure as well as the  compile_commands.json and the two CMakeLists.txt which are generated after the idf.py reconfigure. 

I also tried deleting the build folder - same result - it does not take the local components

HengYongChao, 2021-02-25:
Hi @derflaryam 

I've test the case. For now ADF can not use same name components to replace adf components, because adf use `EXTRA_COMPONENT_DIRS` point to `$ENV{ADF_PATH}/components`  https://github.com/espressif/esp-adf/blob/master/CMakeLists.txt#L9
```
set(EXTRA_COMPONENT_DIRS $ENV{ADF_PATH}/components)
```

derflaryam, 2021-02-25:
So what is the solution ? 

at least you should mention in the documents that this in not possible with the IDF extensions like ADF or MDF. 

very very sad that someone made a quick and dirty implementation - not caring of users.

But the best would be to fix this item in IDF and the addditional modules ADF and MDF

HengYongChao, 2021-02-26:
Sorry, a  soon release will solve this problem.
Currently you may use macros(#if define) to control the compilation process, here's refer:  [examples/get-started/play_mp3/components/my_board](https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3/components/my_board).

aran-arunakiri, 2021-08-10:
is there a solution for this ?

jason-mao, 2022-02-08:
The IDF v4.4 has been fix this compile issue.

closed 2022-02-08 by jason-mao