espressif/esp-adf #577: Sending amr encoder output data with esp now

utkutpcgl, 2021-02-24:
I receive an error while I am sending the encoded output of amr encoder with espnow to another board.
I use:
- Lyrat and Lyrat MSC boards. I face the error on both. (Whoever is sending, receives the error after some time).
- idf(4.3)-adf(latest).
This is the error I face:
[image]

The error appears sometimes once an hour and sometimes after 5 minutes, it is inconsistent, but mostly it takes some time for the error to occur.
I use a callback function (writeToEspnowSendBuffer) to check whether data is ready at the encoder:
`audio_element_set_write_cb(amrEncoder, writeToEspnowSendBuffer , NULL);`

Then inside the callback (method) writeToEspnowSendBuffer() I call esp_now_send() to send the available data:
   ```
if (esp_now_send(espnowUnicastMacAddr, sendParam->buffer, sendParam->len) != ESP_OK)
    { 
        ESP_LOGI(TAG_SEND_RECIEVE,"esp_now_send caused an error!!!");
        esp_restart();
    }
```
**Some guesses about the error:**
I have read other posts about this issue and I think it is because I call esp_now_send() method too frequently without delay. I do not think that I am out of internal memory since the error shows there is available heap memory. Maybe the output data of the encoder is sometimes available with a delay that is too small for calling esp_now_send() again. I do not want to use the SPI RAM, but lack of memory might also be a cause.
Any advice to overcome this issue? 
Thanks in advance.
Utku

HengYongChao, 2021-03-02:
Hi @utkutpcgl 
> wifi:alloc eb len=64 type=3 fail, heap:145348

This is wifi dirver alloc failed,  type 3 means the WiFi management buffer. It indicates that the WiFi management buffer allocation fails.  

You may periodically print out the available memory of the entire system to determine the real available memory.

utkutpcgl, 2021-03-02:
Thanks @HengYongChao ,

I will share the results later.

utkutpcgl, 2021-03-06:
The remaining heap size is the same almost every time: 
63164 in lyrat

jason-mao, 2021-09-26:
@utkutpcgl How is your issue?

closed 2022-02-08 by jason-mao