espressif/esp-adf #582: freepbx connection

ahmadfayed7, 2021-03-02:
I can not connect my esp32 lyrat to freepbx and setup the VOIP Application 

please Advice

HengYongChao, 2021-03-03:
Hi @ahmadfayed7 

Please Provide more detailed information, like ID&ADF version commit info, and how to opearte it, Thanks.

ahmadfayed7, 2021-03-03:
Hi HengYongChao,

Thank you for fast response.

I am trying to run VIOP example  (https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/voip) on  my ESP32-LyraT board but I have problems with setup Freepbx server to connect with and also I can not find (SIP_URI) from Freepbx to configure the application.

could you please provide steps for setup freepbx server for this example and ESP32-LyraT board ?

thanks in advance

ahhfzhang, 2021-03-03:
Hi @ahmadfayed7

You need to setup FreePBX by yourself.
`https://www.freepbx.org/get-started/ `
`https://www.bujarra.com/instalacion-de-freepbx-configuracion-basica/?lang=en`
 --> You can refer here to build, then use microsip to test your server.

About **`SIP_URI`** is similar to the following format `udp://100:100@192.168.1.123:5060`

closed 2021-09-26 by jason-mao