espressif/esp-adf #583: _WindowOVerflow4

Patsjemoe, 2021-03-05:
Hi,
When loading my program, I get a " _Windowoverflow4"  error : any idea where to look for/how to solve this  ?
I happened when I added power save /modem sleep to the program.
thanks in advance 
Regards
Ludo
******************************************************************************************
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6824
load:0x40078000,len:12072
load:0x40080400,len:6708
entry 0x40080778
I (71) boot: Chip Revision: 1
I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.2-107-g722043f73 2nd stage bootloader
I (39) boot: compile time 10:21:45
I (48) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00200000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x28480 (164992) map
I (168) esp_image: segment 1: paddr=0x000384a8 vaddr=0x3ffb0000 size=0x042e8 ( 17128) load
I (175) esp_image: segment 2: paddr=0x0003c798 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/ludok/esp/esp-idf-v3.3.2/components/freertos/xtensa_vectors.S:1779

I (176) esp_image: segment 3: paddr=0x0003cba0 vaddr=0x40080400 size=0x03470 ( 13424) load
I (190) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x9dccc (646348) map
0x400d0018: _flash_cache_start at ??:?

I (420) esp_image: segment 5: paddr=0x000ddcec vaddr=0x40083870 size=0x12d80 ( 77184) load
0x40083870: spi_flash_erase_range at C:/msys32/home/ludok/esp/esp-idf-v3.3.2/components/spi_flash/flash_ops.c:243

I (465) boot: Loaded app from partition at offset 0x10000
I (466) boot: Disabling RNG early entropy source...
I (466) cpu_start: Pro cpu up.
I (470) cpu_start: Application information:
I (474) cpu_start: Project name:     voip_app
I (479) cpu_start: App version:      v2.1-dirty
I (485) cpu_start: Compile time:     Mar  5 2021 10:52:17
I (491) cpu_start: ELF file SHA256:  46817c3deba0d1b9...
I (497) cpu_start: ESP-IDF:          v3.3.2-107-g722043f73
I (503) cpu_start: Starting app cpu, entry point is 0x400812e4
0x400812e4: call_start_cpu1 at C:/msys32/home/ludok/esp/esp-idf-v3.3.2/components/esp32/cpu_start.c:268

HengYongChao, 2021-03-06:
Hi @Patsjemoe 

`WindowOverflow `and similar lines are not an error: IDF monitor decodes addresses found in serial output, converting them to function names. See https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses.With regards to the error coming from core dump — it is benign, no real issue.

Patsjemoe, 2021-03-06:
@HengYongChao : thank you, good to hear it is not an issue.... Although I fail to understand why it suddenly happens and why it is usefull....

closed 2021-03-06 by Patsjemoe