espressif/esp-adf #585: how to put a buffer between 2 audio pipelines ?

Patsjemoe, 2021-03-06:
Hi,
I have 2 pipelines in the VOIP app, and connection is made between the output of the multi output ringbuf and input_rb.
	ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
	audio_element_set_multi_output_ringbuf(i2s_stream_writer, input_rb, 0); 

If I now change this to
       audio_element_set_multi_output_ringbuf(i2s_stream_writer, "OUTPUT", 0); 
Can I put a buffer between the "OUTPUT" in one pipeline  and the input_rb in the other pipeline   ?
Regards
Ludo

HengYongChao, 2021-03-08:
Hi @Patsjemoe 

You certainly can use buf to link the two pipelines, like:
```
    ringbuf = rb_create(8192, 1);
    audio_element_set_input_ringbuf(previous_element, ringbuf);
    audio_element_set_output_ringbuf(next_element, ringbuf);
```

Usage example：
https://github.com/espressif/esp-adf/blob/master/examples/player/element_sdcard_mp3/main/element_sdcard_mp3_example.c#L81

Patsjemoe, 2021-03-08:
Hi,
thank you for your feedback.
I believe (but can be wrong ?) what you do is just connect two ringbuffers from 2 pipelines  together.
What I need to do is to put a buffer between the 2 ringbuffers, to create a delay of this signal.
To say it differently, I have a connection from i2s_stream_writer to input_rb from algorithm_stream with following : 
ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
audio_element_set_multi_output_ringbuf(i2s_stream_writer, input_rb, 0);
but need to cut that connection and put an additional buffer between.
Do I need to create a buffer as an audio_element, so I can define input and output ? Or can you 
create a ringbuf= rb_create(1024, 1) but then with the input and output defined to be ringbuffers from the 2 pipelines.

Regards
Ludo

HengYongChao, 2021-03-08:
> What I need to do is to put a buffer between the 2 ringbuffers, to create a delay of this signal.

In principle, inserting any number of buffers between them is no problem, but the application needs to ensure the correctness of data copy.

Patsjemoe, 2021-03-08:
Hi,
sorry , but still unclear :
the example you give is to put a ringbuf in/out to an audio element, which
connects the elements.
I have I2s with output ringbuf, and algorithm_stream input ringbuf,
elements with ringbuff defined.
But the buffer I need to place in between is not an audio element, how do
you define this buffer-element
with input and output ringbuffers ?

regards
Ludo

HengYongChao, 2021-03-08:
Please use raw stream element, it is essentially a buffer.

Patsjemoe, 2021-03-10:
hi,
I created a raw buffer element "raw_delay", pipeline "delay" and buffer "raw_bufferdelay" input coming from multi_output i2s, output is input_rd to algorithm stream.

Following error : 
I (13127) VOIP_EXAMPLE:  create audio pipeline delay buffer
I (13137) VOIP_EXAMPLE:  create RAW bufferdelay stream
I (13147) VOIP_EXAMPLE: link bufferdelay
W (13147) AUDIO_PIPELINE: There are no listener registered
I registered all 3 elements, so can you explain what this error means ?

thanks in advance
best regards
Ludo
******************************************************************************************
	ESP_LOGI(TAG, " create audio pipeline delay buffer ");
    audio_element_handle_t raw_delay;
    audio_pipeline_cfg_t delay_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    delay = audio_pipeline_init(&delay_cfg);
	mem_assert(delay);
    AUDIO_NULL_CHECK(TAG, delay, return ESP_FAIL);
	
	ESP_LOGI(TAG, " create RAW bufferdelay stream");
    raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_cfg.type = AUDIO_STREAM_READER;
    raw_delay = raw_stream_init(&raw_cfg);
	
	ESP_LOGI(TAG, "link bufferdelay ");
    audio_pipeline_register(delay, i2s_stream_writer, "i2s");
	audio_pipeline_register(delay, raw_delay, "rawdelay");
	audio_pipeline_register(delay, element_algo, "algo");
    const char *link_delay[3] = {"i2s" , "rawdelay", "algo"};
    audio_pipeline_link(player, &link_delay[0], 3);
    audio_pipeline_run(delay);
    ESP_LOGI(TAG, "delay created");
	
	ESP_LOGI(TAG, " create ringbuf ");
	ringbuf_handle_t raw_bufferdelay = rb_create(1024, 1);
		
	ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
	audio_element_set_multi_output_ringbuf(i2s_stream_writer, raw_bufferdelay, 0);
	audio_element_set_input_ringbuf(raw_delay, raw_bufferdelay);
	audio_element_set_output_ringbuf(raw_delay , input_rb);

	algo_stream_set_reference_rate(element_algo, 2, 16000); 
	algo_stream_set_record_rate(element_algo, 1, 16000); 
**********************************************************************************************

Patsjemoe, 2021-03-11:
Hi,
I found "There are no listener registered" in audio_pipeline.h. C....... But cannot find an example where it is initialise/defined. Is this because something is wrong in creating the pipeline? If so, what?
Regards
Ludo

Patsjemoe, 2021-03-11:
Hi,
found a problem in creating the pipeline , I used player instead of delay...
: audio_pipeline_link(player, &link_delay[0], 3);
                                  ------
regards
Ludo

closed 2021-03-11 by Patsjemoe