espressif/esp-adf #586: second pipeline stops when adding raw buffer between 2 audio pipelines

Patsjemoe, 2021-03-12:
Hi,
When running VOIP without the addition of the raw buffer, I get voice with Echo from the ESP32.
When adding a raw buffer "delay"  between 
 audio_element_set_multi_output_ringbuf(i2s_stream_writer, rb, 0);
 and 
 ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
the app will initialise. When pushing to call, I receive the call on my phone, but do not get voice.
Any suggestion where to look for ? 
Any errors in the delay buffer ? 

this is the result showing on the pc  : 
I (17033) VOIP_EXAMPLE:  create audio pipeline delay buffer
I (17033) VOIP_EXAMPLE:  create RAW bufferdelay stream
I (17043) VOIP_EXAMPLE: link bufferdelay
I (17043) AUDIO_PIPELINE: link el->rb, el:0x3f80bf30, tag:i2s, rb:0x3f811ad8
I (17053) AUDIO_PIPELINE: link el->rb, el:0x3f811960, tag:rawdelay, rb:0x3f813b18
I (17063) AUDIO_ELEMENT: [rawdelay-0x3f811960] Element task created
I (17073) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:4169528 Bytes, Inter:101196 Bytes, Dram:83856 Byt
es

I (17083) AUDIO_PIPELINE: Pipeline started
I (17083) VOIP_EXAMPLE: delay created
I (17093) VOIP_EXAMPLE:  create ringbuf
I (17093) SIP_RTP: receive task, local addr=192.168.0.151:39897
:1307803) SIP_RTP: send task, remote addr=192.168.0.203
I (17143) RSP_FILTER: sample rate of source data : 8000, channel of source data : 1, sample rate of destination data : 1
6000, channel of destination data : 2

....here, nothing happens, whereas without the buffer, the program continues with : 

I (10783) SIP: [1970-01-01/00:00:05]<<=====READ 0779 bytes==
I (10783) SIP:
SIP/2.0 200 OK

*********************************************************
 ESP_LOGI(TAG, " create audio pipeline delay buffer ");

    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    delay = audio_pipeline_init(&pipeline_cfg);
	mem_assert(delay);
    AUDIO_NULL_CHECK(TAG, delay, return ESP_FAIL);
	
	ESP_LOGI(TAG, " create RAW bufferdelay stream");
    raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_cfg.type = AUDIO_STREAM_READER;
    raw_delay = raw_stream_init(&raw_cfg);
		
	ESP_LOGI(TAG, "link bufferdelay ");
    audio_pipeline_register(delay, i2s_stream_writer, "i2s");
	audio_pipeline_register(delay, raw_delay, "rawdelay");
	audio_pipeline_register(delay, element_algo, "algo");
    const char *link_delay[3] = {"i2s" , "rawdelay", "algo"};
    audio_pipeline_link(delay, &link_delay[0], 3);
    audio_pipeline_run(delay);
    ESP_LOGI(TAG, "delay created");
	
	ESP_LOGI(TAG, " create ringbuf ");
	ringbuf_handle_t rb = rb_create(1024, 1);;
	ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
	audio_element_set_multi_output_ringbuf(i2s_stream_writer, rb, 0);
	
	audio_element_set_input_ringbuf(raw_delay, rb);
	audio_element_set_output_ringbuf(raw_delay , input_rb);
	algo_stream_set_reference_rate(element_algo, 2, 16000); 
	algo_stream_set_record_rate(element_algo, 1, 16000); 

Regards
Ludo

HengYongChao, 2021-03-16:
Hi @Patsjemoe 

The problem you are facing now is that you can receive phone call events, but no voice data?

Could you verify that your delay buffer is working properly?

@ahhfzhang   Do you have any suggestions for customer of this VOIP application?

Patsjemoe, 2021-03-16:
Hi,
Thank you for your feedback.
I am calling out with esp32 to linphone on android. The phone rings, I pick
up and should start voice, but then it blocks.
 I do NOT get following response from esp32 :
<<=====READ 0779 bytes==
I (10783) SIP:
SIP/2.0 200 OK
and I do not get voice....

How can I check if the delay buffer works?
Initialising seems to work, as it does not show errors :
I (17083) AUDIO_PIPELINE: Pipeline started
I (17083) VOIP_EXAMPLE: delay created
I (17093) VOIP_EXAMPLE: create ringbuf

Regards
Ludo

Patsjemoe, 2021-03-16:
Hi,
below, the pipelines complete : maybe this helps to test ?

static sip_handle_t sip;
static audio_element_handle_t raw_read, raw_write, raw_delay;
static audio_pipeline_handle_t recorder, player, delay;

static esp_err_t recorder_pipeline_open()
{

	ESP_LOGI(TAG, " **************recorder begin***************");
	
	ESP_LOGI(TAG, " create audio pipeline recorder for recording");
    audio_element_handle_t i2s_stream_reader;
    audio_pipeline_cfg_t pipeline_rd_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    recorder = audio_pipeline_init(&pipeline_rd_cfg);
	mem_assert(recorder);
    AUDIO_NULL_CHECK(TAG, recorder, return ESP_FAIL);

	ESP_LOGI(TAG, " create I2S stream to read from codec ");
    i2s_stream_cfg_t i2s_rd_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_rd_cfg.type = AUDIO_STREAM_READER;
    i2s_rd_cfg.uninstall_drv = false;
#if defined CONFIG_ESP_LYRAT_MINI_V1_1_BOARD  
    i2s_rd_cfg.i2s_port = 1;
#endif
    i2s_stream_reader = i2s_stream_init(&i2s_rd_cfg);
    audio_element_info_t i2s_rd_info = {0};
    audio_element_getinfo(i2s_stream_reader, &i2s_rd_info);
    i2s_rd_info.bits = I2S_BITS;
    i2s_rd_info.channels = I2S_CHANNELS;
    i2s_rd_info.sample_rates = I2S_SAMPLE_RATE;
    audio_element_setinfo(i2s_stream_reader, &i2s_rd_info);
	
	ESP_LOGI(TAG, "[10.2] Create algo stream ");
    algorithm_stream_cfg_t algo_config = ALGORITHM_STREAM_CFG_DEFAULT();
    algo_config.input_type = ALGORITHM_STREAM_INPUT_TYPE2;
    audio_element_handle_t element_algo = algo_stream_init(&algo_config);
	
	ESP_LOGI(TAG, "[] Create filter_rd ");
	rsp_filter_cfg_t rsp_rd_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_rd_cfg.src_rate = I2S_SAMPLE_RATE; //16000
    rsp_rd_cfg.src_ch = I2S_CHANNELS;
    rsp_rd_cfg.dest_rate = CODEC_SAMPLE_RATE;
    rsp_rd_cfg.dest_ch = CODEC_CHANNELS;
    rsp_rd_cfg.complexity = 5;
    audio_element_handle_t filter_rd = rsp_filter_init(&rsp_rd_cfg);

	ESP_LOGI(TAG, " Create sip encoder ");
    g711_encoder_cfg_t g711_rd_cfg = DEFAULT_G711_ENCODER_CONFIG();
    audio_element_handle_t sip_encoder = g711_encoder_init(&g711_rd_cfg);

		ESP_LOGI(TAG, "Create raw stream ");
    raw_stream_cfg_t raw_rd_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_rd_cfg.type = AUDIO_STREAM_READER;
    raw_read = raw_stream_init(&raw_rd_cfg);
    audio_element_set_output_timeout(raw_read, portMAX_DELAY);
	

    audio_pipeline_register(recorder, i2s_stream_reader, "i2s");
	audio_pipeline_register(recorder, element_algo, "algo");
	audio_pipeline_register(recorder, filter_rd, "filterrd");
    audio_pipeline_register(recorder, sip_encoder, "sip_enc");
    audio_pipeline_register(recorder, raw_read, "raw");
    const char *link_rec[5] = {"i2s", "algo", "filterrd", "sip_enc", "raw"};
    audio_pipeline_link(recorder, &link_rec[0], 5);
    audio_pipeline_run(recorder);
    ESP_LOGI(TAG, " SIP recorder has been created");
	

[... 63 lines cut ...]
		
	ESP_LOGI(TAG, "link bufferdelay ");
    audio_pipeline_register(delay, i2s_stream_writer, "i2s");
	audio_pipeline_register(delay, raw_delay, "rawdelay");
	audio_pipeline_register(delay, element_algo, "algo");
    const char *link_delay[3] = {"i2s" , "rawdelay", "algo"};
    audio_pipeline_link(delay, &link_delay[0], 3);
    audio_pipeline_run(delay);
    ESP_LOGI(TAG, "delay created");
	
	ESP_LOGI(TAG, " create ringbuf ");
	//Please reference the way of ALGORITHM_STREAM_INPUT_TYPE2 in "algorithm_stream.h"
	ringbuf_handle_t rb = rb_create(1024, 1);;
	ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
	audio_element_set_multi_output_ringbuf(i2s_stream_writer, rb, 0);
	
	audio_element_set_input_ringbuf(raw_delay, rb);
	audio_element_set_output_ringbuf(raw_delay , input_rb);

	algo_stream_set_reference_rate(element_algo, 2, 16000); 

	algo_stream_set_record_rate(element_algo, 1, 16000); 

    return ESP_OK;
}

Patsjemoe, 2021-03-22:
Hi,
Do you get the same result ?
As I do not get SIP/2.0 200 OK, it looks as if the sip ok  is not received, or handled anymore by the pipeline, and therefore sound not started... 
Regards
Ludo

Patsjemoe, 2021-03-23:
Maybe any suggestion how to add test-lines to find where it is stuck?

ahhfzhang, 2021-03-23:
Hi,
I think your delay buffer will not work, maybe you can set delay in this function, but we are not sure if this will cause other problems
https://github.com/espressif/esp-adf/blob/eafb7783ea9bd4274deb817e2e53e297d64e279e/components/audio_stream/algorithm_stream.c#L301

Patsjemoe, 2021-03-23:
Hi,
Thank you.
I am OK to try it, but can you make  any suggestion how to implement it ?

Regards
Ludo

Patsjemoe, 2021-03-24:
> Hi,
> I think your delay buffer will not work, maybe you can set delay in this function, but we are not sure if this will cause other problems
> https://github.com/espressif/esp-adf/blob/eafb7783ea9bd4274deb817e2e53e297d64e279e/components/audio_stream/algorithm_stream.c#L301

Hi,
Can you confirm why it will not work?
Did you test the sw I provided? Did you get the same problem?
The suggestion refers to a preprocessing part of aec, I fail to understand this is where a delay can be implemented. Can you elaborate on this?
Thanks in advance
Best regards
Ludo

se-k32, 2021-04-01:
Hi, 
I am also facing the same issue with echo. I also modified VoIP with AEC example to use type 2 algorithm and it is "working", but echo is pretty much the same. We are using audio board similar to Olimex one. @ahhfzhang could you please tell us in more detail how we could implement this delay or anything else that could help us.   

@Patsjemoe did you maybe already find any solution for this? 

Thank you!

Patsjemoe, 2021-04-05:
@se-k32: I do not have a solution for the buffer yet, still hoping for it.
Lyrat-mini solution provided seems to work, but I need lyrat or audiokit
for i2c interface.

Patsjemoe, 2021-04-14:
Hi,
I added a buffer only as pipeline, which seems not to disturb the other pipelines. 
I do get a lower (bit crackling) echo, but is a lot less, and seems to be on the right track for a solution....
I changed the ringbuf size to 256.
Is this something you can try too ?
regards
Ludo
***************************************************************************************************************
ESP_LOGI(TAG, " create audio pipeline delay buffer ");

    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    delay = audio_pipeline_init(&pipeline_cfg);
	mem_assert(delay);
    AUDIO_NULL_CHECK(TAG, delay, return ESP_FAIL);
	
	ESP_LOGI(TAG, " create RAW bufferdelay stream");
    raw_stream_cfg_t raw_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_cfg.type = AUDIO_STREAM_WRITER;
    raw_delay = raw_stream_init(&raw_cfg);
		
	ESP_LOGI(TAG, "link bufferdelay ");
	audio_pipeline_register(delay, raw_delay, "rawdelay");
	const char *link_delay[1] = {"rawdelay"};
    audio_pipeline_link(delay, &link_delay[0], 1);
    audio_pipeline_run(delay);
    ESP_LOGI(TAG, "delay created");
	
	
	ESP_LOGI(TAG, " create ringbuf ");
	ringbuf_handle_t rb = rb_create(256, 1);;
	ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
	audio_element_set_multi_output_ringbuf(i2s_stream_writer, rb, 0);
	
	audio_element_set_input_ringbuf(raw_delay, rb);
	audio_element_set_output_ringbuf(raw_delay , input_rb);

	//algo_stream_set_reference_rate(audio_element_handle_t algo_handle, int ref_ch, int ref_sample_rate);
	algo_stream_set_reference_rate(element_algo, 2, 16000); 
	//algo_stream_set_record_rate(audio_element_handle_t algo_handle, int rec_ch, int rec_sample_rate);

	algo_stream_set_record_rate(element_algo, 1, 16000);

se-k32, 2021-04-20:
Hi, thank Ludo for your reply (and sorry for the late answer ). I tried with your method, but the voice I got from ESP is terrible and you can understand just every 5th word or so. In comparison with Lyrat mini and VoIP example with algorithm 1, the results are like night and day. Does this also happens to you or do you get much better results? I tested this with Olimex board. Which board do you use? 

Thank you in advance!

Best Regards!

Patsjemoe, 2021-04-20:
Hi @se-k32
Are you referring to the 256 byte ringbuffer?

Regards
Ludo

se-k32, 2021-04-20:
Hi Ludo,

Yes I am referring to your last comment.

Best regards!

Patsjemoe, 2021-04-20:
Hi,
I could test it real live today, and it is not as I hoped.
With only the ringbuffer in the pipeline, audio functions (previously with
3 items in the pipeline, had no sound).
With ringbuffer of 256 bytes, I have good audio, but still echo,
unsatisfactory. (thought previously it was better, and still trying to do
better) .
I am actually using audiokit.
I still hope that Espressif can come up with a solution (lyrat or
audiokit), as I also use i2c for a display.  Lyrat mini has no i2c except
internal.
Regards
Ludo

se-k32, 2021-04-20:
Thank you for your reply. I also hope we get some kind of solution for this echo on other devices and not only for Lyrat Mini, otherwise we will have to use DSP chip with AEC.

Regards!

HengYongChao, 2021-04-21:
Hi @se-k32  @Patsjemoe,

The hardware difference between Lyrat v4.3 and Lyrat mini v1.2 is the design of the reference signal, not ringbuf or dsp chip. 

The reference signal ensures that the software AEC can work correctly. So your hardware can be designed with reference to the Lyrat mini v1.2, which can save a DSP chip for cost.

Patsjemoe, 2021-04-21:
Hi,
Thanks for your feedback.
Yes, the connection is hardwired in the lyrat mini , this is clear and
understood, but mini board does not have an i2c external interface to an
lcd display, therefore trying to find a solution with LyraT or audio kit.
Therefore, I am trying to solve it with a buffer to take into account
differences in timing of the audio output from the speaker to the micro and
the internal connection to the algo stream.
Unfortunately, the concept / block diagram you use for aec is not
described, and therefore difficult to know the impact of lyrat being stereo
versus your solution for lyrat mini which is mono.
Trying to switch off on channel, use a buffer and hope this is then close
to what your aec does in mini.
Appreciate your effort and help, just hoping for a lyrat solution.....
Regards
Ludo

HengYongChao, 2021-04-21:
> Hi, Thanks for your feedback. Yes, the connection is hardwired in the lyrat mini , this is clear and understood, but mini board does not have an i2c external interface to an lcd display, therefore [...]

If your problem is the lack of an I2C interface to drive the LCD, then it is not a big problem.

You can use jump lines connect to the esp32 module, which the I2C pins conected to the codec chip.

I2C itself supports multiple devices and communicates through different addresses.

Patsjemoe, 2021-04-22:
@HengYongChao
Thank you for the info. 
I retested lyrat mini today, and echo is good. Only the speaker sound is too weak, and if I increase it, esp speaker sound is distorted.
Any idea how to improve this?
Regards
Ludo

HengYongChao, 2021-04-25:
> @HengYongChao
> Thank you for the info.
> I retested lyrat mini today, and echo is good. Only the speaker sound is too weak, and if I increase it, esp speaker sound is distorted.
> Any idea how to improve this? Maybe decreasing mic gain and increasing speaker gain? (did not find how to decrease mic gain)
> Regards
> Ludo

Codec registers setting up dirver is here: https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver.

Take Lyrat v4.3 for example, es8388 mic gain API is [esp_err_t es8388_set_mic_gain(es_mic_gain_t gain)](https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es8388/es8388.h#L230).

Patsjemoe, 2021-04-25:
@HengYongChao
Thank you.
This is where I was looking, but
the es7243 datasheet I found, does not show registers, and in es7243.c used
in lyrat mini there is no mic gain description. Where can we find a
complete datasheet? Which register is for mic gain/settings for it?

Regards
Ludo

Patsjemoe, 2021-04-25:
@HengYongChao 
When searching "es7243 registers" , I found another datasheet with register information.... So, you can forget my previous comment/quesfion
Regards

Patsjemoe, 2021-05-01:
> > […](#)
> 
> If your problem is the lack of an I2C interface to drive the LCD, then it is not a big problem.
> 
> You can use jump lines connect to the esp32 module, which the I2C pins conected to the codec chip.
> 
> I2C itself supports multiple devices and communicates through different addresses.
*******************************************************************************************************
@HengYongChao : 
I hardwired I2C, and basically it works, can drive my display.
But, as I still get a number of errors in my debug screen in red, and cannot find where you init i2c config and where you load the i2c driver, can you confirm where this  is done ? (in my voip app, I added 	i2c_master_init(); , but since you must have done initialisation somewhere, it seems to conflict somehow...)
following errors : 
1) I (2771) VOIP_EXAMPLE: [ 2 ] Start codec chip
E (2771) i2c: i2c driver install error
E (2781) I2C_BUS: C:/msys32/home/ludok/esp/esp-adf-v2.2/components/esp_periphera
ls/driver/i2c_bus/i2c_bus.c:86 (i2c_bus_write_bytes):Handle error
2)I (2941) DRV8311: ES8311 in Slave mode
E (2941) I2C_BUS: C:/msys32/home/ludok/esp/esp-adf-v2.2/components/esp_periphera
ls/driver/i2c_bus/i2c_bus.c:86 (i2c_bus_write_bytes):Handle error
3)I (3291) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulld
own: 0| Intr:0
E (3301) I2C_BUS: C:/msys32/home/ludok/esp/esp-adf-v2.2/components/esp_periphera
ls/driver/i2c_bus/i2c_bus.c:127 (i2c_bus_read_bytes):Handle error
4)W (3421) I2C_BUS: i2c_bus_create:57: I2C bus has been already created, [port:0]
E (3431) I2C_BUS: C:/msys32/home/ludok/esp/esp-adf-v2.2/components/esp_periphera
ls/driver/i2c_bus/i2c_bus.c:88 (i2c_bus_write_bytes):I2C port error
5)I (3531) AUDIO_HAL: Codec mode is 3, Ctrl:1
E (3541) I2C_BUS: C:/msys32/home/ludok/esp/esp-adf-v2.2/components/esp_periphera
ls/driver/i2c_bus/i2c_bus.c:127 (i2c_bus_read_bytes):Handle error
(each error is shown several times)
regards
Ludo

Patsjemoe, 2021-05-02:
@HengYongChao : 
When I first initiated the codec chip (and its i2c config somewhere...), then left out my i2c_master_init() (below), I did not get any errors anymore.
What remains weard is that, despite of the errors, the display and voip worked fine. I feel like there is a little difference in initialising the codec chip on the mini board (based on esp-adf i2c ?) and the i2c_master_init() taken from the esp-idf......?

void i2c_master_init()
{
	i2c_config_t i2c_config = {
		.mode = I2C_MODE_MASTER,
		.sda_io_num = SDA_PIN,
		.scl_io_num = SCL_PIN,
		.sda_pullup_en = GPIO_PULLUP_ENABLE,
		.scl_pullup_en = GPIO_PULLUP_ENABLE,
		.master.clk_speed = 1000000
	};
	i2c_param_config(I2C_NUM_0, &i2c_config);
	i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);
}

Regards
Ludo

closed 2021-05-05 by Patsjemoe