espressif/esp-adf #587: please add a adc_stream_reader that samples data from the on board ADC

aguaviva, 2021-03-15:
I know this is not ideal but it would be helpful in certain situations. Would this works with a sample rate of at least 8K?

HengYongChao, 2021-03-16:
Hi @aguaviva 

It is feasible in principle to use general adc to collect sample data. sample rate is ok too.

But because of the noise floor, the actual usability will be reduced.

Thanks for your suggestion!

aguaviva, 2021-03-16:
I had in mind using this for a bluetooth electric guitar, so distorsion and noise would be very welcome ;) Who knows if this catches up and becomes trendy! :)

jason-mao, 2023-09-05:
This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

raulamd, 2023-09-05:
This is neither inactive nor complete :) This is still relevant and a feature worth having, and if this is not going to be implemented for technical reasons it would be good to explain why.

jason-mao, 2023-09-05:
@raulamd Indeed, thank you for mention this.  From the point of view of audio quality is not recommended, if you do not consider this factor is a different matter. In actual use, I rarely receive such a demand. Maybe I should reopen this issue to gather everyone's input.