espressif/esp-adf #588: implementation for esp_equalizer_process in esp-adf

sarimabbas, 2021-03-15:
We are trying to build a parametric equalizer and wished to find the definition for `esp_equalizer_process`. We found the header file in `esp-adf/components/esp-adf-libs/esp_codec/include/processing/esp_equalizer.h` but cannot find the implementation. Would you be able to point me in the right direction? Thank you so much!

HengYongChao, 2021-03-16:
Hi @sarimabbas 

This part of the code is encapsulated in lib, not open yet. 

Would you like to share your suggestions and ideas with us here?

sarimabbas, 2021-03-18:
Thank you for your response! Is it possible to create our own audio element to implement this or can you point us in the right direction on how to do so. Also adding the ability for a parametric equalizer would be helpful. Thanks again!

HengYongChao, 2021-03-18:
Hi @sarimabbas 

Please look here, does it meet your requirements?

https://github.com/espressif/esp-adf-libs/blob/59f531061f492eeb17894c100ac044afc554a4ab/esp_codec/equalizer.c

https://github.com/espressif/esp-adf-libs/blob/59f531061f492eeb17894c100ac044afc554a4ab/esp_codec/include/codec/equalizer.h

closed 2021-09-26 by jason-mao