espressif/esp-adf #589: [korvo_du1906]Compile error: ....phy_init_data.h:18:23: fatal error: sdkconfig.h: No such file or directory

jayjayhust, 2021-03-18:
I'm using the ADF v2.2 Release version with "audio/stack_on_psram_v3.3"  IDF v3.3.4 version to test the korvo_du1906 example, but still come across the error message like this:
[image]

Can any engineer put a ADF release with a compatiable IDF built-in? Just like the MDF v1.0 Release?

HengYongChao, 2021-03-19:
Hi @jayjayhust 

If your ADF and IDF are both pulled in completed, the compilation should not have such a problem.

Could you please use `git submodule update --init --recursive` in IDF the folder  to update the submodules, and  try rebuild again please?

jason-mao, 2021-09-26:
Please refer to korvo_du1906  readme https://github.com/espressif/esp-adf/tree/master/examples/korvo_du1906#idf-branch

closed 2021-09-26 by jason-mao