espressif/esp-adf #590: periph_bluetooth_passthrough_cmd not working

SinanAkkoyun, 2021-03-18:
Hello, I use the a2dp_sink_and_hfp example to connect my Mi 9T Pro phone to my LyraT.
The audio stream works but the pause, next, previous etc buttons do not work.
The touch gets registered but the phone does not receive the commands.

Log:
```--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM5 instead...
--- idf_monitor on \\.\COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
W (24389) BT_APPL: new conn_srvc id:27, app_id:1
E (24419) BT_HF: APP HFP event: CONNECTION_STATE_EVT
E (24419) BT_HF: --connection state connected, peer feats 0x0, chld_feats 0x0
E (24489) BT_HF: APP HFP event: CALL_IND_EVT
E (24489) BT_HF: --Call indicator NO call in progress
E (24489) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
E (24499) BT_HF: --Call setup indicator NONE
E (24499) BT_HF: APP HFP event: NETWORK_STATE_EVT
E (24529) BT_HF: --NETWORK STATE unavailable
E (24529) BT_HF: APP HFP event: SIGNAL_STRENGTH_IND_EVT
E (24529) BT_HF: -- signal strength: 0
E (24539) BT_HF: APP HFP event: ROAMING_STATUS_IND_EVT
E (24549) BT_HF: --ROAMING: inactive
W (24559) BT_APPL: bta_hf_client_send_at: No service, skipping 11 command
E (24589) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
E (24589) BT_HF: --battery level 2
E (24589) BT_HF: APP HFP event: CALL_HELD_IND_EVT
E (24599) BT_HF: --Call held indicator NONE held
E (24599) BT_HF: APP HFP event: CONNECTION_STATE_EVT
E (24609) BT_HF: --connection state slc_connected, peer feats 0x367, chld_feats 0x2b
E (24619) BT_HF: APP HFP event: SIGNAL_STRENGTH_IND_EVT
E (24629) BT_HF: -- signal strength: 1
I (24629) BT_LOG: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x0 bits:2 features:x8000

E (24639) BT_OSI: alarm_set failed to start timer, err 0x103

W (24639) BT_APPL: new conn_srvc id:19, app_id:0
I (24669) BLUETOOTH_EXAMPLE: [ * ] Receive music info from Bluetooth, sample_rates=44100, bits=16, ch=2
E (24699) BT_BTC: handle_rc_connect Connect failed with error code: 2
I (24719) AUDIO_ELEMENT: [i2s_w] AEL_MSG_CMD_PAUSE
I (24749) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (24759) AUDIO_ELEMENT: [i2s_w] AEL_MSG_CMD_RESUME,state:4
I (24759) I2S_STREAM: AUDIO_STREAM_WRITER
I (41569) BLUETOOTH_EXAMPLE: [ * ] [Play] touch tap event
I (42019) BLUETOOTH_EXAMPLE: [ * ] [Play] touch tap event```

HengYongChao, 2021-03-19:
Hi @SinanAkkoyun 

What IDF version did you use? Please use  built-in $ADF_PATH/ESP-IDF to compile the example and test again.

SinanAkkoyun, 2021-03-19:
Hello @HengYongChao 
I downloaded the latest ESP IDF version from within VSCode extension.

What do you mean with internal ADF path?

HengYongChao, 2021-03-19:
> Hello @HengYongChao
> I downloaded the latest ESP IDF version from within VSCode extension.
> 
> What do you mean with internal ADF path?

Yes, please use adf built-in esp-idf, or idf version below release/v4.2, then test you example again.

expresspotato, 2021-04-06:
@HengYongChao

Steps to reproduce (esp-idf v3.3.2):
```
git clone https://github.com/espressif/esp-adf.git
git submodule update --init --recursive
cd ~/esp-adf/esp-idf
./install.sh
. ./export.sh
cd ~/esp-adf/examples/get-started/pipeline_a2dp_sink_and_hfp
idf.py build
idf.py flash
```
This works but esp-idf that's part of esp-adf is only on 3.3.2... Lots of features and bug fixes missing. Volume control does not work because periph_bt_volume_up is only part of 4.0+... Not sure how other's got this working. It's really not ideal.

Checking out esp-idf v4.1 in the esp-adf folder, you start to get errors and AVRCP does not work at all.
```
cd ~/esp-adf/esp-idf
git checkout remotes/origin/releases/v4.1
git submodule update -f --init --recursive
rm -rf ~/.espressif
./install.sh
./ ./export.sh
cd ~/esp-adf/examples/get-started/pipeline_a2dp_sink_and_hfp
idf.py build
idf.py flash
```

`W (1523) BT_BTC: A2DP Enable without AVRC
W (1763) BT_BTC: AVRC Controller is expected to be initialized in advance of A2DP !!!`

Checking out v4.0 yields the same thing, but periph_bt_volume_up is missing before this version.