espressif/esp-adf #591: AUDIO_HAL_I2S_DSP usage

valioiv, 2021-03-25:
If an audio codec chip is configured as I2S Master device (the ESP32 is I2S Slave) and the Codec is working in so called "DSP" (or "multichannel") mode of operation as follows:
[image]
How does it expected to configured the I2S stream?
It is visible in the HAL is put **AUDIO_HAL_I2S_DSP** but what about the I2S stream initialization?
```c
i2s_stream_init(&i2s_cfg); // What to put inside the **i2s_cfg.i2s_config** substructure
```
[image]

HengYongChao, 2021-03-26:
Hi @valioiv 

 ESP32-LyraTD-MSC board has a DSP(zl38063) , similar to your current solution, you can refer to the implementation of MSC board.

valioiv, 2021-03-26:
@HengYongChao , great! Can you point me a link to an exact codebase to see..? This github or anything else..?

HengYongChao, 2021-03-26:
> @HengYongChao , great! Can you point me a link to an exact codebase to see..? This github or anything else..?

https://github.com/espressif/esp-adf/tree/master/components/audio_hal/driver/zl38063

valioiv, 2021-03-26:
@HengYongChao , I went through the code and I don't see anywhere **i2s_stream_init()** invocation... Can you share an example (I believe it should be in the application/examples domain) how to configure the I2S stream itself or at least the I2S hardware in "DSP" mode. It is crystal clear to me how to enable the codec chip in "DSP" mode but the ZL**** code you have shared does not contain any direct I2S configuration but only codec-related configurations...

HengYongChao, 2021-03-29:
> @HengYongChao , I went through the code and I don't see anywhere **i2s_stream_init()** invocation... Can you share an example (I believe it should be in the application/examples domain) how to configure the I2S stream itself or at least the I2S hardware itself in "DSP" mode. It is crystal clear to me how to enable the codec chip in "DSP" mode but I2S is unclear. The ZL38068 code you have shared does not contain any direct I2S configuration but only ZL38068-related configurations...

Hi @valioiv 

To be Clarify, AUDIO_HAL_I2S_DSP format is PCM format actually. There are instructions in the comments of the code.
```
/**
 * @brief Select I2S interface format for audio codec chip
 */
typedef enum {
    AUDIO_HAL_I2S_NORMAL = 0,  /*!< set normal I2S format */
    AUDIO_HAL_I2S_LEFT,        /*!< set all left format */
    AUDIO_HAL_I2S_RIGHT,       /*!< set all right format */
    AUDIO_HAL_I2S_DSP,         /*!< set dsp/pcm format */
} audio_hal_iface_format_t;
```
Regarding this format, only es8311 uses it.  https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es8311/es8311.c#L514

valioiv, 2021-03-30:
> Hi @valioiv
>
> To be Clarify, AUDIO_HAL_I2S_DSP format is PCM format actually. There are instructions in the comments of the code.
>
> ```
> /**
>  * @brief Select I2S interface format for audio codec chip [...]

@HengYongChao, I perfectly understand the idea of **AUDIO_HAL_I2S_DSP** from the codec chip point of view. But now it is just something floating in the air without real usage if it is not in direct relation to the **i2s_stream_init()** which actually is "the guy" really configuring the I2S hardware (no one else). And there in the **i2s_stream_init()** there is nothing about DSP configuration. Even I'm not talking for my original question how ESP32 supports that **standard** I2S "DSP/multichannel" stream. How do you think?
[image]

HengYongChao, 2021-03-31:
Hi @valioiv 

Just like the  dsp zl38063 mentioned earlier, for example, in the asr example, we can choose msc v2.2 board to setup this DSP chip to work in menuconfig.

In fact, the initialization of dsp is done in /components/audio_board/lyratd_msc_v2_2 folder. The audio board calls the dirvers under /audio hal/dirvers/zl38063.

The point that may need to be noted is that when communicating with dsp, we need to select esp32 as the master and provide mclk clock to slave dsp.

valioiv, 2021-04-01:
@HengYongChao, I'll be short.
Can you point me exact line in the code where on a  **AUDIO_HAL_I2S_DSP** selection you guys configure ESP32's I2S hardware to work in "DSP" mode? Not the audio chip registers but the ESP32 I2S itself?
I expect to see **AUDIO_HAL_I2S_DSP** in switch-case statement or something and the **i2s_stream_init()** invocation to configure I2S in "DSP" mode. Where is that?

HengYongChao, 2021-04-06:
Hi @valioiv 

> Can you point me exact line in the code where on a AUDIO_HAL_I2S_DSP selection you guys configure ESP32's I2S hardware to work in "DSP" mode?

I was said "AUDIO_HAL_I2S_DSP format is PCM format actually", and the code is here:https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es8311/es8311.c#L514
```
        case AUDIO_HAL_I2S_DSP:
            ESP_LOGD(TAG, "ES8311 in DSP-A Format");
            adc_iface &= 0xDC;
            dac_iface &= 0xDC;
            adc_iface |= 0x03;
            dac_iface |= 0x03;
            break;
```

> Not the audio chip registers but the ESP32 I2S itself? 

Audio chip config use i2c or spi interface not i2s port,  i2s port is used to transmit raw data.

> I expect to see AUDIO_HAL_I2S_DSP in switch-case statement or something and the i2s_stream_init() invocation to configure I2S in "DSP" mode next to it. Where is that?

I said that maybe the name "AUDIO_HAL_I2S_DSP" caused your confusion. In fact, the dsp configuration is through the spi or i2c interface, and the audio data is transmitted through the i2s interface.

valioiv, 2021-04-12:
@HengYongChao, okay, the answer is acceptable from point of ESP-ADF usage. But it is still strange to have AUDIO_HAL_**I2S_DSP** because I2S has a DSP format of operation and whenever you configure it you are actually configuring the the audio chip but there is nothing in common with I2S ESP32 channel.
If you ask me you should configure the I2S ESP32 hardware in one pass (within one and a same function) together with the audio chip configuration (register write over I2C or SPI, or etc.), but not having separate function calls in the application - one for the **i2s_stream_init()** another for the audio chip. Should be something like that as a function one of the top level functions on ESP-ADF:
```c
audio_interface_config(audio_interface_config_t* aiface)
{
    // Depending on the **aiface** structure content configure both the ESP32 audio interface and audio chip interface on one pass here. Should have some evaluation of what is supported in a switch-case statement or multi-dimensional table...

    // Configure the ESP32 I2S interface
    // ...
    i2s_cfg.i2s_config.communication_format = I2S_COMM_FORMAT_PCM;
    audio_element_handle_t i2s_stream = i2s_stream_init(&i2s_cfg);

    // Configure the audio chip interface here:
    // Call some audio chip specific functions here to send register settings over I2C, SPI, etc...
    // ...
}
```

valioiv, 2021-04-12:
For my original request I've open an issue in the ESP-IDF:
https://github.com/espressif/esp-idf/issues/6784