espressif/esp-adf #592: VOIP example   problem

ahmadfayed7, 2021-03-27:
i installed asterisk server on ubuntu i could communicate 2 phone  together 
i downloaded the VOIP_example on lyrat v4.3   but i got  this
[image]

please advice !!!!

HengYongChao, 2021-03-29:
Hi @ahmadfayed7  
I don't know the problem you encountered. You mean the yellow warning code in the log?
I think you can ignore it for the time being, and know from the log that your device has been connected to the WiFi network.
Please update the question or description.

ahmadfayed7, 2021-03-31:
Hi @HengYongChao 

according to VOIP example the steps as following :
[1.0] Initialize peripherals management
[1.1] Initialize and start peripherals
[1.2] Start and wait for Wi-Fi network
[ 2 ] Start codec chip
[ 3 ] Create and start input key service
[ 4 ] Create SIP Service

But the SW stuck on   [1.2] Start and wait for Wi-Fi network  step  .

ahhfzhang, 2021-04-01:
Hi @ahmadfayed7  

Seems your board did not get the IP, please check if DHCP is enabled on your AP.

ahmadfayed7, 2021-04-02:
Hi @ahhfzhang 

could you tell me steps  to  check if DHCP is enabled on your AP.