espressif/esp-adf #593: esp_sip_uac_bye doesn't work

szucsi23, 2021-03-29:
Hey!
I'm having an issue with the function esp_sip_uac_bye().
When I'm in a call, and hang up by calling this function, nothing happens on the other side. The logs show, that the function is called, but on the other end there is no sign of the hangup action, the call still goes on.

IDF version: v4.2
ADF version: v2.2

Logs:

`I (76125) STATEMACHINE: State machine state changed to: gsVOIP
W (76475) SIP: CHANGE STATE FROM 16, TO 32, :func: sip_uas_process_req:884
I (76485) SIP_RTP: receive task, local addr=192.168.4.247:32731:1290485) SIP_RTP: send task, remote addr=159.65.95.129
W (102695) SIP: CHANGE STATE FROM 32, TO 2, :func: _sip_request_bye:1408
I (102695) SIP_RTP: send task stopped
I (102715) SIP_RTP: receive task stopped
I (102715) SIP_RTP: RTP stopped`

ahhfzhang, 2021-03-30:
Can you provide a complete log?

szucsi23, 2021-03-30:
Yes, I'll provide the full logs tomorrow, I don't have the device with me right now.

szucsi23, 2021-04-06:
Sorry, I got caught up with other things.
So here is the log for a full run.
I used the adf example voip app on a LyraT v4.3, but instead of antisip, I use signalwire, since that's the provider I'm using in my main application as well.

This is how the test goes:
- I flash the example VoIP code to my LyraT v4.3.
- I call my LyraT from the Zoiper VoIP app on Windows (other VoIP apps produce the same issue). The call is detected on line 184 of the log file.
- The call connects, audio is working.
- I press the Set button (for hangup) on the LyraT, and it should hang up the call. (line 364)
- The LyraT hangs up the call, but it's still alive in the Zoiper app on my PC.
- I press hangup on the Zoiper app on the PC (line 443)

Thank you for your help in advance!

[VoIP_example_logs.txt](https://github.com/espressif/esp-adf/files/6264894/VoIP_example_logs.txt)

szucsi23, 2021-05-28:
Hey!
Just an update, it seems like if the LyraT initiates the call, the hangup works.
But if I call the LyraT from an other device, and hang up on the LyraT, it doesn't work.

Did you have a chance to check te logs?

ahhfzhang, 2021-06-03:
Hi @szucsi23 ,

I have checked the log, 'BYE' is alerady send to the server, but i don't know why the server responded with a 404 (Not here).
[image]

Maybe you can try the latest adf, cause we fixed some sip bug, hope you can try again.

BR.

szucsi23, 2021-06-03:
I'll check tomorrow, and write with an update, thanks!

raulrodriguez1, 2021-06-11:
It seems I may have a similar problem with BYE. Although slightly different.

I have two ESPs with an Asterisk Server running on Arch. If I call ESP 2 with ESP 1 and hang up the call with ESP 2, then BYE works correctly. However, when I call ESP 2 with ESP 1 and hang up the call with ESP 1, then I get a SIP/2.0 481 Call leg/transaction does not exist and the call continues.

In the attached log, 
[SIP_failed_BYE.txt](https://github.com/espressif/esp-adf/files/6635097/SIP_failed_BYE.txt)

I have seen that when hanging up from the same ESP that made the call causes the TO and FROM fields to switch. Meaning that the TO field contains the extension of the sending ESP and the FROM field contains the extension of the receiving ESP. You can see the difference in the two logs. Below is a successful hangup.
 
[SIP_successful_BYE.txt](https://github.com/espressif/esp-adf/files/6635101/SIP_successful_BYE.txt)

IDF Version: v4.0.2
ADF Version: v2.3-2-g7203f86

jason-mao, 2021-09-26:
@szucsi23 How is your issue?

closed 2022-02-08 by jason-mao