espressif/esp-adf #596: VOIP Example LyraT Mini v1.2 Mic not working

nexstem-siddhant, 2021-04-04:
Using the example I am able to register the ESP LyraT Mini v1.2 as a SIP Phone, and am able to make calls as well as accept calls. I am able to hear audio using the earphones jack, but the mic on the LyraT isn't working at all. I tried running other examples to test out the mic, and they worked perfectly.

Can anybody tell me what might be the potential issue here?

HengYongChao, 2021-04-06:
Hi @nexstem-siddhant 

VOIP example can run on lyrat v4.3 board, except for AEC function cannot be used.

> https://github.com/espressif/esp-adf/tree/master/examples/advanced_examples/voip

Is it correct that the setting of your menuconfig development board? please update log here.

nexstem-siddhant, 2021-04-10:
I got it to work by just using the example from the master branch for the ADF.

closed 2021-04-10 by nexstem-siddhant

fk1000iq, 2024-10-16:
> I got it to work by just using the example from the master branch for the ADF.

what is your env? idf version? adf version? I steel get the problem with idf(v5.2.3), adf(v2.7 , currently as master) , board LyraT Mini v1.2.   LinPhone on both PC work well. But Linphone + LyraT Mini, mic not work.