espressif/esp-adf #598: Target bt unable to connect to the user device

erango-git, 2021-04-07:
**Issue description**
I have been running the pipeline_bt_sink example on the ESP32-LyraT: 
https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_sink.
I have just modified the behaviour related to the buttons Vol+/Vol-. I used the api to connect/disconnect from target to user device. After trying to connect and disconnect to the user device I have noticed that sometimes after a disconnection, the connection will not be successful.
Below the log:
[log_sink_pipeline.txt](https://github.com/espressif/esp-adf/files/6273246/log_sink_pipeline.txt)

**Step to reproduce the error**
- Clone esp-idf: git clone -b v4.0.1 --recursive https://github.com/espressif/esp-idf.git
- Clone esp-adf: git clone -b v2.1 --recursive https://github.com/espressif/esp-adf.git
- Copy the example pipeline_bt_sink from esp-adf
- Set from menuconfig ESP32-LyraT V 4.3 in Audio HAL configuration
- Apply yhe following modifications to play_bt_music_example.c code:
  * Add #include "esp_a2dp_api.h"
  * Define bda (the bt id device to which you want to connect)
  * Use the api esp_a2d_sink_connect(bda) istead of periph_bluetooth_next(bt_periph)
  * Use the api esp_a2d_sink_disconnect(bda) istead of periph_bluetooth_prev(bt_periph)
  * Comment the break where the bt disconnection bt occurs (to avoid the deinitialization of bluetooth service)
- Load the fw
- Try to connect/disconnect to the user device(S8 Android 9)

Below the play_bt_music_example.c code modified:
[play_bt_music_example.txt](https://github.com/espressif/esp-adf/files/6273284/play_bt_music_example.txt)

HengYongChao, 2021-04-14:
Hi @erango-git 
Would you like to try [IDF release v4.2](https://github.com/espressif/esp-idf/releases/tag/v4.2), this version fixed some bt issue.

erango-git, 2021-04-20:
Hi @HengYongChao ,
I used esp-idf v4_2 as you suggested, but i have the same issue.
Below the log:
[log_pipeline_esp_idf_4_2.txt](https://github.com/espressif/esp-adf/files/6341480/log_pipeline_esp_idf_4_2.txt)

The behavior is very casual: the cycle connection/disconnection sometimes works for many times, sometimes it doesn't even after only one connection / disconnection cycle as you can see on the log.
When I have these lines on the log, the connection seems established, but the audio can't be reproduced.
E (22725) BT_APPL: bta_dm_pm_btm_status hci_status=12
E (22725) BT_APPL: bta_dm_pm_btm_status hci_status=12

What do you suggest to overcome the issue?
Thank you

ahhfzhang, 2021-04-30:
Hi @erango-git , 

Would you like to try another BT sink example : https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream 
and recommended that you can use ADF master branch and IDF release v4.2. 

We tested your usage scenario, and the log you described also appeared, but Audio can still be played normally.

hope you can try again, thanks.

erango-git, 2021-04-30:
Hi @ahhfzhang,
Thank you for your help.
Unfortunatly I see the same problem. May it depend on the device used? 
I am using Samsung S8 (Android 9) for my test. While if I use S21 with Android 11 everything works correctly.
I am using ADF master branch, IDF release v4.2 and the a2dp example as you suggested but nothing with S8.

The last time the connection happens, after two or three cycle of connection/disconnection, i can't enable to reproduce audio and after more or less 20 seconds this message appears:
W (140828) BT_SDP: SDP - Rcvd conn cnf with error: 0xeeee  CID 0x47
E (140828) BT_BTC: handle_rc_connect Connect failed with error code: 2

I attach the log for completeness
[pipeline_a2dp_sink_stream_log.txt](https://github.com/espressif/esp-adf/files/6405175/pipeline_a2dp_sink_stream_log.txt)

What do you suggest? Am I wrong something?

ahhfzhang, 2021-05-06:
Hi @erango-git,

We have tested with Samsung S8 on IDF 4.0 and 4.2, and found the the disconnection will be very slow, which will affect the connection.

But based on some of our previous experience, We've test on a specific branch `audio/stack_on_psram_v3.3` , and the current results looks great, so hope you can use this branch and test again.
```

cd $IDF_PATH
git checkout master
git pull
git checkout audio/stack_on_psram_v3.3
git submodule update --init --recursive

```
ADF : master 
Example :  https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream

btw: the specific branch `audio/stack_on_psram_v3.3` should use lyrat development board with ESP32-WROVER-E module 
and you must set `CONFIG_ESP32_REV_MIN=3` in your sdkconfig.

ahhfzhang, 2021-05-12:
Hi @erango-git,

Yes, i got same error when using Cmake to compile, but make is good. 
ref : https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started/index.html
hope you can try again.

erango-git, 2021-05-13:
Hi @ahhfzhang,
I build the app with make as you suggest and I used a board with ESP32-WROVER-E to verify that don't crash and I confirm you it doesen't happen.
But I can't test the feature because I have lyrat development board with ESP32-WROVER (Rev 1) and an other board with ESP32-WROVER-B (Rev 1) and i can set from menuconfig until CONFIG_ESP32_REV_MIN=1. Using this configuration the sink example resets after starting scheduler:

I (26) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6812
load:0x40078000,len:12072
load:0x40080400,len:6708
entry 0x40080778
I (113) boot: Chip Revision: 1
I (131) boot_comm: chip revision: 1, min. bootloader chip revision: 0

Is it possible to use this specific branch  audio/stack_on_psram_v3.3 with Revision 1?

ahhfzhang, 2021-05-17:
Hi @erango-git,

no, this specific branch audio/stack_on_psram_v3.3 can only be used on ESP32-v3 chip.
But you can directly replace the module of the lyrat development board with ESP32-WROVER-E or WROOM-32E.

BR.

erango-git, 2021-05-18:
Hi @ahhfzhang,
I am organizing to integrate the Wrover-E on the EVB and on my target, Which are the main differences between Wrover-B and Wrover-E chips? Have I to expect different behavior about module interfaces like i2s, gpio and ADC/DAC and for Bluetooth Audio specifications?

Thank you
Erango

ahhfzhang, 2021-05-20:
Hi @erango-git,

The only difference is that the main chip is replaced with the V3 version, all module interfaces are the same.

closed 2022-02-08 by jason-mao