espressif/esp-adf #599: VOIP : mute microphone when voice in the speaker ?

Patsjemoe, 2021-04-08:
Hi,
As AEC in VOIP with LyraT and audiokit is not working, and no solution for a buffer  is available, I wonder if following is possible : 
when voice is going trough the output, mute the microphohne (in SW), so dynamically mute the mic ?
regards
Ludo

HengYongChao, 2021-04-09:
Hi @Patsjemoe 

First of all, thank you for your suggestion!

When one person is talking, the microphones of other people will be muted. This is called half-duplex communication and is not a very satisfying experience.

You cannot hear the far-end voice when you are speaking. This means that if you want to insert a point of view or want to clarify, you cannot communicate with the other person until the other person stops talking, because your microphone is It is muted.

Therefore, in most cases, half-duplex conferences are unacceptable.

Patsjemoe, 2021-04-09:
Hi,
Thank you for your feedback.
I understand, but the reason why I asked this is because  AEC does not work
with LyraT and audiokit and the echo makes voip not usable.
Maybe another idea : could we use the "SPEEX" codec ? This codec seems to
have AEC integrated...
I was also thinking to mute one channel on the stereo boards, as maybe AEC
is processed on 1 channel only (?? How is it really done?? ) and the other
channel still picks up the speakers sound and echos it. It is just an
idea....
Regards
Lludo

expresspotato, 2021-04-11:
> Hi @Patsjemoe
> 
> First of all, thank you for your suggestion!
> 
> When one person is talking, the microphones of other people will be muted. This is called half-duplex communication and is not a very satisfying experience.
> 
> You cannot hear the far-end voice when you are speaking. This means that if you want to insert a point of view or want to clarify, you cannot communicate with the other person until the other person stops talking, because your microphone is muted.
> 
> Therefore, in most cases, half-duplex conferences are unacceptable.

Completely agree. Although the Algorithm isn't very usable either...
 - Type 2 should be more easily set with an i2s stream that has 2 channels! I'm not even sure how to do it with the external ring buffer. Even if there is a delay because the two mics share the same i2s path, it's preferable to zero AEC... Calls without it are just as unbearable as half-duplex.
 - AEC, AGC and NS need some tunable parameters for the given application. I tried to increase the window (even just doubled all values) in algorithm_steam.c in hopes of reasonable quality, but it just crashes. 
 - NS usually is supposed to take a secondary microphone and cancel out what is heard on both, so not too sure how it's doing Noise Suppression with just one mic.
 - Calls are very soft with the Algorithm enabled, this is most likely because of AGC, when I tap on the microphone it seems to be louder afterward, but this is far from ideal.

closed 2021-04-15 by Patsjemoe