espressif/esp-adf #602: MiTuner / Firmware for LayraTD-MSC ZL38063

expresspotato, 2021-04-12:
Since AEC, AGC and NS aren't really very good with the current software algorithm, I'm actively investing doing it with a DAC that supports these in hardware, namely the Espressif Dev Kit LayraTD-MSC with the MicroSemi ZL38063.

https://github.com/espressif/esp-adf/blob/c3b792c1e129f0ccd8877fc082a253ded78e984f/components/audio_hal/driver/zl38063/Readme_twolf.txt

The software links don't work, does anyone know where I can download the firmware / MiTuner / Guide PDF to properly tune / enable these functions?

HengYongChao, 2021-04-12:
Hi @expresspotato 

I think MSC dsp [firmware was updated](https://github.com/espressif/esp-adf/commit/81541bd4f1901e892cb3f238e18801c06c4744a7) in ADF, please have a try.

expresspotato, 2021-04-12:
Thanks, I see it was updated 17 days ago! Any documentation though on the chip itself (parameters, AEC, AGC, etc)? Seems like the functionality is set via i2c and set in registers once the firmware is loaded.

HengYongChao, 2021-04-13:
> Thanks, I see it was updated 17 days ago! Any documentation though on the chip itself (parameters, AEC, AGC, etc)? Seems like the functionality is set via i2c and set in registers once the firmware is loaded but I can find zero documentation on which registers to set for what functionality...

That DSP firmware contains AEC, NS, BF, AGC functions.
If you want to get customized firmware, you need to configure it with MSC software.

expresspotato, 2021-04-13:
Okay I'll buy a LayraTD-MSC and give it a go. I've reached out to MSC but just like Qualcomm they're tight lipped about how to actually configure the thing.

Patsjemoe, 2021-04-13:
Hi,
I found following :
https://www.microchip.com/wwwproducts/en/ZL38063
Then view datasheet... 63 pages..
Regards
Ludo

expresspotato, 2021-04-13:
Thanks, it seems all of it is configured in firmware, then flashed to the device on initialization. There is not much to configure other than switching between pre-configured modes. I'm trying to get the software from micro semi to check if we can evaluate this chip for mass production but it's also taking time. 

What happens when the host audio application is configurated differently for music playback vs a call? That's another thing I don't know.

jason-mao, 2021-09-26:
@expresspotato `What happens when the host audio application is configurated differently for music playback vs a call? That's another thing I don't know.`  The music may deformation. This issue closed  by solved  main topic.

closed 2021-09-26 by jason-mao

NathanSweet, 2023-03-11:
@expresspotato Did you ever get the MiTuner ZLS38508 software? I'm in a similar situation, waiting on their approval.

expresspotato, 2023-03-11:
> @expresspotato Did you ever get the MiTuner ZLS38508 software? I'm in a similar situation, waiting on their approval.

We found it difficult to work with Microchip, they didn't provide any information and their sales person was quite rude. So we moved onto an older but well working chip from Taiwan.

NathanSweet, 2023-03-11:
Ah, damn. Any chance you can share the chip in case I need to go that route?

rduell, 2023-07-01:
I received the MSC software (MiTuner Gui). However, i found the ZL38063 firmware inside the adf repository [https://github.com/espressif/esp-adf/tree/master/components/audio_hal/driver/zl38063/firmware](url) in form of a static lib (*.a) which contains the compiled object file of the timberwolf firmware.
the original timberwolf firmware and config files (*.s3 and *.cr2) are are not available. is there any chance to receive receive them from Espressif , so I can fine tune the filters?