espressif/esp-adf #603: Bluetooth A2DP sink SBC, limit sample rate to 44.1KHz

iloop2020, 2021-04-12:
Hi,

Is it possible to limit A2DP sink SBC to 44.1KHz, even smartphone can support 48KHz.

Thank you.

HengYongChao, 2021-04-14:
Hi @iloop2020 

I find the same question here: https://www.esp32.com/viewtopic.php?t=1834, Please check it if helps you.

To define sample rate I do check in callback for A2DP sink: in case of ESP_A2D_AUDIO_CFG_EVT 

        a2d = (esp_a2d_cb_param_t *)(param);
                 uint8_t sbc0 = a2d->audio_cfg.mcc.cie.sbc[0];
One of 4 bits in sbc0 indicates sample rate:

	if (sbc0 & 0x80)	sample_rate = 16000;
	if (sbc0 & 0x40)	sample_rate = 32000;
	if (sbc0 & 0x20)	sample_rate = 44100;
	if (sbc0 & 0x10)	sample_rate = 48000;

HengYongChao, 2021-04-14:
Hi @iloop2020 
For the decoder in the SNK the sampling frequencies 44.1 kHz and 48 kHz are mandatory to support.
If phone use 48k, sink use 48k. Now answer your question directly. You need to add resample to output 44.1k data.

iloop2020, 2021-04-21:
Hi @HengYongChao ,

Thank you, and got your point, but my question can we limited the phone use 44.1KHz ?

Thank you again.

HengYongChao, 2021-04-22:
> Hi @HengYongChao ,
> 
> Thank you, and got your point, but my question can we limited the phone use 44.1KHz ?
> 
> Thank you again.

Hi @iloop2020 
Sorry, I don’t know how to limited the phone to only use 44.1k sampling rate. Maybe its' likely to depend on the behavior of the phone or app (source role)?

iloop2020, 2021-04-25:
Hi @HengYongChao 

I am not familiar with A2DP, but I expect there a is negotiation process for number of channes/ codec and Sample Rate etc.

Can the team check with it ?

Thank you

HengYongChao, 2021-04-25:
There is indeed a negotiation mechanism. 

Generally, the sink role must support two sampling rates of 44.1k and 48k, but the source role does not necessarily need to support all sampling rates, so the negotiation result depends on the source role.

closed 2022-02-08 by jason-mao