espressif/esp-adf #604: How to visualise content of a created buffer

Patsjemoe, 2021-04-15:
Hi,
Is it possible to visualise the content of a created buffer? This to see if the buffer really functions
Regards
Ludo

HengYongChao, 2021-04-16:
Hi @Patsjemoe 

I want to say that visualization is a big project, currently we are using [these interfaces](https://github.com/espressif/esp-adf/blob/master/components/audio_pipeline/include/ringbuf.h) to print the status of ringbuf.

Patsjemoe, 2021-04-18:
hi,
if I would make the buffer with 2 outputs, can I then link 1 of them to an I/O pin of the board ? 
If possible, I could put a scope or logic analyzer to the I/o and visuasize the buffer.
regards
ludo

HengYongChao, 2021-04-19:
> hi,
> if I would make the buffer with 2 outputs, can I then link 1 of them to an I/O pin of the board ?
> If possible, I could put a scope or logic analyzer to the I/o and visuasize the buffer.
> regards
> ludo

If you like, it is possible in principle to use GPIOs to output the state of the internal element.

But I think serial printing is better way, after all, printing contains more information, such as line number, timestamp, error code, etc.

Patsjemoe, 2021-04-19:
Hi,
Thank you.
Do you have an example on how to print the content of the buffer?  That is
what I meant with visualize....
Regards
Ludo
*********************

HengYongChao, 2021-04-19:
Hi @Patsjemoe 

I still don't know what `an example on how to print the content of the buffer` you are looking for means. But generally speaking, you can directly use the printf function to print raw data. Or you can also use APIs such as `get_xx x` to get the status of element or pipeline.

Patsjemoe, 2021-04-19:
hi @HengYongChao , 
I created a ringbuffer : ringbuf_handle_t rb = rb_create(512, 1)
How do I prinf the content/data of that buffer ?
regards
ludo

HengYongChao, 2021-04-20:
> hi @HengYongChao ,
> I created a ringbuffer : ringbuf_handle_t rb = rb_create(512, 1)
> How do I prinf the content/data of that buffer ?
> regards
> ludo

As I mentioned you can use `rb_read` read the rb data to a buffer and then print them. 

When we talk about buffer I assume you are talk rb, because the array buffer or pointer malloc is easy to see the real data, ringbuffer need to read to static buffer then operate the same as a normal array.

michalber, 2021-04-29:
Hi @Patsjemoe 

I am not sure if I understand your question properly but if you want to print content of ringbuffer you have three options:
1. You need to make sure other tasks won't write to ring buffer during this process. Next you can read all RB to buffer (with size of RB) and print content using for loop or ESP_LOG_BUFFER_CHAR / ESP_LOG_BUFFER_HEX. After this you need to write readed data to RB because you have readed it all (RB is empty).
2. You can create copy of RB component and create function such as `rb_get_data`, which will be based on `rb_read` function but you need to modify it to not change ringbuffer's read pointer. In this case you can get ringbuffer's data to separate buffer without need of writing it back to RB.
3. Similarly to 2 you can create function such as 'rb_print_data' to iterate through conent of RB and print it directly without copying RB data to separate buffer.

Hope this help you.

closed 2021-05-05 by Patsjemoe