espressif/esp-adf #605: Speech recognition with ESP32-S2?

barakv75, 2021-04-21:
I'm trying to compile esp-sr with ESP32-S2 and I'm getting linker error:
**undefined reference to `esp_sr_wakenet5_quantized'**
I believe it's regarding the extern variables in rec_eng_helper.c file.
Were is the declaration of this variable (esp_sr_wakenet5_quantized)?
Did I miss something using the ESP32-S2? (with ESP32 it's working good, without compilation error)
Does anyone succeeded to work with ESP32-S2 and Speech recognition?

Thanks

HengYongChao, 2021-04-22:
Hi @barakv75 
esp-sr is only a lib, You can got more details here:  [esp-skainet](https://github.com/espressif/esp-skainet)

barakv75, 2021-04-26:
Hi HengYongChao,
I understand that it's a lib, but I try to use it in my project (the same as esp-skainet does), and I'm getting the same error:
"undefined reference to `esp_sr_wakenet5_quantized'.
Do you know if it's possible to use ESP32-S2 module with esp-skainet or esp-adf speech recognition?
Thanks

jason-mao, 2021-09-26:
@barakv75 I am afraid the esp-sr does not support ESP32-S2 chip.

closed 2021-09-26 by jason-mao