espressif/esp-adf #606: 这是什么错误

wangdaiop, 2021-04-23:
/home/wangda/esp/esp-adf/esp-idf/components/lwip/lwip/src/core/init.c:144:2: error: #error "TCP_WND is bigger than the configured LWIP_WND_SCALE allows!"
 #error "TCP_WND is bigger than the configured LWIP_WND_SCALE allows!"
  ^~~~~
esp-idf/lwip/CMakeFiles/__idf_lwip.dir/build.make:328: recipe for target 'esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/init.c.obj' failed
make[2]: *** [esp-idf/lwip/CMakeFiles/__idf_lwip.dir/lwip/src/core/init.c.obj] Error 1
make[2]: *** 正在等待未完成的任务....
CMakeFiles/Makefile2:4286: recipe for target 'esp-idf/lwip/CMakeFiles/__idf_lwip.dir/all' failed
make[1]: *** [esp-idf/lwip/CMakeFiles/__idf_lwip.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
make failed with exit code 2

wangdaiop, 2021-04-23:
wangda@wangda-K43BY:~/esp/korvo_du1906$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/wangda/esp/korvo_du1906/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/wangda/esp/korvo_du1906"...
fatal: 不是一个 git 仓库（或者任何父目录）：.git
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/wangda/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/wangda/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/wangda/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/wangda/esp/korvo_du1906/sdkconfig
Loading defaults file /home/wangda/esp/korvo_du1906/sdkconfig.defaults...
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:11 CONFIG_TOOLPREFIX was replaced with CONFIG_SDK_TOOLPREFIX
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:12 CONFIG_PYTHON was replaced with CONFIG_SDK_PYTHON
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:13 CONFIG_MAKE_WARN_UNDEFINED_VARIABLES was replaced with CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:29 CONFIG_LOG_BOOTLOADER_LEVEL_INFO was replaced with CONFIG_BOOTLOADER_LOG_LEVEL_INFO
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:30 CONFIG_LOG_BOOTLOADER_LEVEL was replaced with CONFIG_BOOTLOADER_LOG_LEVEL
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:170 CONFIG_FLASHMODE_QIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_QIO
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:181 CONFIG_MONITOR_BAUD_921600B was replaced with CONFIG_ESPTOOLPY_MONITOR_BAUD_921600B
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:182 CONFIG_MONITOR_BAUD_OTHER_VAL was replaced with CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER_VAL
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:183 CONFIG_MONITOR_BAUD was replaced with CONFIG_ESPTOOLPY_MONITOR_BAUD
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:203 CONFIG_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_SIZE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:204 CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED was replaced with CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_ENABLE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:205 CONFIG_STACK_CHECK_NONE was replaced with CONFIG_COMPILER_STACK_CHECK_MODE_NONE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:210 CONFIG_ESP32_APPTRACE_DEST_NONE was replaced with CONFIG_APPTRACE_DEST_NONE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:211 CONFIG_ESP32_APPTRACE_LOCK_ENABLE was replaced with CONFIG_APPTRACE_LOCK_ENABLE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:221 CONFIG_BTDM_CONTROLLER_MODE_BTDM was replaced with CONFIG_BTDM_CTRL_MODE_BTDM
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:222 CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN was replaced with CONFIG_BTDM_CTRL_BLE_MAX_CONN
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:223 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:224 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:226 CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF was replaced with CONFIG_BTDM_CTRL_BLE_MAX_CONN_EFF
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:227 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN_EFF
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:228 CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF was replaced with CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN_EFF
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:230 CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE was replaced with CONFIG_BTDM_CTRL_PINNED_TO_CORE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:231 CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI was replaced with CONFIG_BTDM_CTRL_HCI_MODE_VHCI
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:236 CONFIG_BLE_SCAN_DUPLICATE was replaced with CONFIG_BTDM_BLE_SCAN_DUPL
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:237 CONFIG_SCAN_DUPLICATE_BY_ADV_DATA_AND_DEVICE_ADDR was replaced with CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:238 CONFIG_SCAN_DUPLICATE_TYPE was replaced with CONFIG_BTDM_SCAN_DUPL_TYPE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:239 CONFIG_DUPLICATE_SCAN_CACHE_SIZE was replaced with CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:240 CONFIG_BLE_MESH_SCAN_DUPLICATE_EN was replaced with CONFIG_BTDM_BLE_MESH_SCAN_DUPL_EN
/home/wangda/esp/korvo_du1906/sdkconfig.defaults:241 CONFIG_MESH_DUPLICATE_SCAN_CACHE_SIZE was replaced with CONFIG_BTDM_MESH_DUPL_SCAN_CACHE_SIZE
[... cut ...]
[ 13%] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32/chip_info.c.obj
[ 13%] Linking C static library libesp_hw_support.a
[ 13%] Built target __idf_esp_hw_support
Scanning dependencies of target __idf_soc
[ 13%] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/memory_layout_utils.c.obj
[ 13%] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj
[ 13%] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/soc_include_legacy_warn.c.obj
[ 13%] Buildi

HengYongChao, 2021-04-23:
Hi @wangdaiop 

你的这个错误是 menuconfig 中 lwip 的窗口配置错误， 除非你知道自己在改什么，否则一般用默认的配置即可。
当前你的问题解决：删掉 sdkconfig ，然后重新配置必要的 menuconfig 即可。

wangdaiop, 2021-04-23:
什么都没有改  安装完第一次运行就报这样的错误

决绝方案能说具体点吗？

HengYongChao, 2021-04-23:
> 什么都没有改 安装完第一次运行就报这样的错误
> 
> 决绝方案能说具体点吗？

DU1906 的例程说明文档里已经提示：需要使用特殊的 IDF 分支 ` audio/stack_on_psram_v3.3`，不能使用 adf 内置的 idf 分支。
请按照 README.MD 文档的说明来操作，https://github.com/espressif/esp-adf/blob/master/examples/korvo_du1906/README.md

closed 2021-04-30 by jason-mao