espressif/esp-adf #607: esp-sr for esp32-s2

andrej102, 2021-04-28:
I moved a working project from ESP 32 to ESP32-S2. I use the stable environment ESP-IDF-v4.2 and acoustic algorithms from ESP-ADF-v2.2 (only components in esp-sr (only MASE and AGC)). The project is running on ESP32, but when trying to build it for ESP32-S2 this error occurs:

FAILED: voca-esp-gsm.elf
cmd.exe /C "cd . && C:\Users\andrej\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin\xtensa-esp32s2-elf-g++.exe  -mlongcalls   @CMakeFiles\voca-esp-gsm.elf.rsp  -o voca-esp-gsm.elf  && cd ."
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:/esp-adf-v2.2/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(analog_agc.o)(.text.WebRtcAgc_ProcessAnalog+0xd1): could not decode instruction; possible configuration mismatch
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:/esp-adf-v2.2/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(analog_agc.o)(.text.WebRtcAgc_Init+0xa6): could not decode instruction; possible configuration mismatch
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:/esp-adf-v2.2/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(digital_agc.o)(.text.WebRtcAgc_ProcessVad+0x1d): could not decode instruction; possible configuration mismatch
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:/esp-adf-v2.2/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(digital_agc.o)(.text.WebRtcAgc_ProcessVad+0x61): could not decode instruction; possible configuration mismatch
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:/esp-adf-v2.2/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(digital_agc.o): in function `WebRtcAgc_InitVad':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/automatic_gain_control/digital_agc.c:655:(.text.WebRtcAgc_InitDigital+0x53): dangerous relocation: cannot decode instruction opcode: (.text.WebRtcAgc_InitDigital+0x5c)
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/automatic_gain_control/digital_agc.c:655:(.text.WebRtcAgc_InitDigital+0x85): dangerous relocation: cannot decode instruction opcode: (.text.WebRtcAgc_InitDigital+0x8c)
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/automatic_gain_control/digital_agc.c:655:(.text.WebRtcAgc_InitVad+0x2f): dangerous relocation: cannot decode instruction opcode: (.text.WebRtcAgc_InitVad+0x36)
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:/esp-adf-v2.2/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(analog_agc.o): in function `WebRtcAgc_Init':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/automatic_gain_control/analog_agc.c:1662:(.text.WebRtcAgc_Init+0xa6): dangerous relocation: cannot decode instruction opcode: (.text.WebRtcAgc_Init+0xad)
c:/users/andrej/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/automatic_gain_control/analog_agc.c:1675:(.text.WebRtcAgc_Init+0xe1): dangerous relocation: cannot decode instruction opcode: (.text.WebRtcAgc_Init+0xec)
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

What does this error mean, and how do I fix it?

HengYongChao, 2021-04-29:
Hi @andrej102 

Currently esp-sr not support esp32s2, but only the tts function is available.

andrej102, 2021-04-29:
It would be good to specify this in the ESP-ADF description. It is very unpleasant for users to find out this after the start of the project.
Are you planning to introduce support esp32-s2?

HengYongChao, 2021-05-06:
> It would be good to specify this in the ESP-ADF description. It is very unpleasant for users to find out this after the start of the project.

I suggest use esp320s3 chip, instead of the current s2 chip.

> Are you planning to introduce support esp32-s2?

Please pay attention to the update of this [esp-sr](https://github.com/espressif/esp-sr) repo.

andrej102, 2021-05-14:
1.
The ESP32-S2 was chosen because it consumes little power and is available for purchase in the ESP32-S2-MINI-1 format. This is a great option for portable use. 
ESP32-S3 is not available to order now, and I have not even found documentation for it and for modules based on it. Which version of the IDF and ADF support it?

2.
> Please pay attention to the update of this esp-sr repo.

What is this link for, I did not find any information on the issues discussed on it.

3.
Why is there no ESP32-S2 support in the Aqustic Algorithm, is it due to a lack of performance or insufficient memory?

4.
We manufacture and sell thousands of devices annually with ESP32, and in order to determine the possibility of using ESP32-S2 in our projects in the future, I need to know exactly whether support for the Aqustic Algorithm will be added or not?

Thanks.

jason-mao, 2021-09-26:
@andrej102 Sorry for late.

> The ESP32-S2 was chosen because it consumes little power and is available for purchase in the ESP32-S2-MINI-1 format. This is a great option for portable use.
ESP32-S3 is not available to order now, and I have not even found documentation for it and for modules based on it. Which version of the IDF and ADF support it?

Now the latest master have support ESP32-S3. ADF developing .

>Please pay attention to the update of this esp-sr repo.
> What is this link for, I did not find any information on the issues discussed on it.

Here is https://github.com/espressif/esp-sr

> Why is there no ESP32-S2 support in the Aqustic Algorithm, is it due to a lack of performance or insufficient memory?

ESP32-S2 is sigle core, I think performance is not enough to do more algorithm. what is exactly Aqustic Algorithm?

> We manufacture and sell thousands of devices annually with ESP32, and in order to determine the possibility of using ESP32-S2 in our projects in the future, I need to know exactly whether support for the Aqustic Algorithm will be added or not?

How is your products? Would you please send your requests to our mail by  sales@espressif.com, we are glad to help you.

closed 2022-02-08 by jason-mao