espressif/esp-adf #610: esp_sleep_enable_ext0_wakeup problem

Patsjemoe, 2021-05-05:
Hi,
I use following line in VOIP to wake up from deep sleep.
esp_sleep_enable_ext0_wakeup(GPIO_NUM_33, 1); //1 = High, 0 = Low 
But get following error : 
 error: expected ')' before numeric constant
 esp_sleep_enable_ext0_wakeup(GPIO_NUM_33, 1); //1 = High, 0 = Low
                                                                             ^
The  1 (or 0 for the level to wake up) should be defined, but it complains for it.
any idea ?
regards
Ludo

michalber, 2021-05-05:
Hi @Patsjemoe 

I assume the problem lies in GPIO_NUM_33 definition. Due to IDF documentation this function expects `gpio_num_t`. Maybe try something like this:
`esp_sleep_enable_ext0_wakeup((gpio_num_t)GPIO_NUM_33, 1);`
or this if you are using C++
`esp_sleep_enable_ext0_wakeup(static_cast<gpio_num_t>(GPIO_NUM_33), 1);`

Patsjemoe, 2021-05-06:
Hi @michalber 
Thank you for your quick reply.
I used your suggestion, but now get following error : 

 expected declaration specifiers or '...' before '(' token
 esp_sleep_enable_ext0_wakeup((gpio_num_t)GPIO_NUM_33, 1);
                                                     ^
regards
Ludo

michalber, 2021-05-06:
@Patsjemoe 

I tested `esp_sleep_enable_ext0_wakeup(GPIO_NUM_33, 1);` function invocation with my application and it is working without any errors. I am using ADF v2.1 with delivered IDF.

Also I found example of deep sleep in IDF repo included in ADF master.
https://github.com/espressif/esp-idf/blob/722043f734fa556d66d57473ac266fb1d0ec5ad2/examples/system/deep_sleep/main/deep_sleep_example_main.c

Check lines 86-96, 140-146 and 177.

Patsjemoe, 2021-05-06:
@michalber,
Thank you.
Can you share your test example?
I found that example, but is not using ext0.
What do you mean by "delivered IDF"?
Regards
Ludo

michalber, 2021-05-06:
@Patsjemoe 

I tested it with voip example where I added two lines:
```
esp_sleep_enable_ext0_wakeup(GPIO_NUM_33, 1);
esp_deep_sleep_start();
```
to the end of main function and it is compiling without any errors.

By delivered IDF I mean specific IDF copy which is included in ADF repository.

Patsjemoe, 2021-05-06:
@michalber
thank you. 
When adding it , as mentioned, indeed the error is gone.
I had put the  esp_sleep_enable_ext0_wakeup(GPIO_NUM_33, 1); at the beginning, after my "defines", as I thought this was a setting required before the main program.
Now I can see how to use it properly.
thanks again for your help
regards
Ludo

closed 2021-05-07 by Patsjemoe