espressif/esp-adf #613: Issue in pipeline_hfp_stream example."SCO xmit Q overflow, pkt dropped" after ESP_HF_CLIENT_AUDIO_STATE_EVT connects / disconnects multiple times

spaceMan00, 2021-05-09:
Hello, 
I am facing this problem when using the pipeline_hfp_stream example. 
ADF version master (latest)
IDF version: v4.1.1-250-ga9218526
Hardware: Lyrat-v4.3

I connect to HFP through my iphone and I use the Voice Memo app to record from the Lyrat.
First time I start a recording everything is fine. 
If I stop and start a recording a second time I notice there is a lag. 
If I do it for the 3rd time the lag becomes unbearable and Lyrat throws the `<BT_BTM: SCO xmit Q overflow, pkt dropped>` warning. 

You can check my log file [here](https://github.com/espressif/esp-adf/files/6446909/lyratLogs.txt).

After investigating I noticed that `<fixed_queue_length(p_ccb->xmit_data_q)>` in BTM_WriteScoData function inside btm_sco.c remains zero when I do the first recording but increases in the second recording and reaches the 30 threshold when I start the third recording which is triggering the overflow warning. 

Please help!

HengYongChao, 2021-05-11:
Hi @spaceMan00 

I recommend that you use this branch [bugfix/stack_on_psram_v3.3](https://github.com/espressif/esp-idf/tree/audio/stack_on_psram_v3.3) instead of v4.1.1. The current branch fixes some problems.

At the same time, this branch allows tasks run on PSRAM, It is recommended to use WROVER-E (eco3) module, please check  the module on your development board.

Any questions, please feedback here.

spaceMan00, 2021-05-13:
Thank you @HengYongChao for your response. 

I did as you said and reverted back IDF to branch "audio/stack_on_psram_v3.3" and kept ADF on master. After flashing the example pipeline_hfp_stream the Lyrat board kept resetting (I believe due to WDT). Check the logs [here](https://github.com/espressif/esp-adf/files/6471923/log.txt).
Clearly the example didn't work. 
I checked the module on my Lyrat and it is indeed the ESP32-WROVER-E

I have a few questions as a follow up:
1- Does the branch "audio/stack_on_psram_v3.3"  work with ADF master? or do I have to checkout an old version of ADF? 
2- It seems that "audio/stack_on_psram_v3.3" branch is much older than 4.1. It will be a shame that I have to revert to such an old version of IDF to be able to make the example work! (if it works).

HengYongChao, 2021-05-13:
Hi @spaceMan00 

Please check the menuconfig  "(Top) → Component config → ESP32-specific → Minimum Supported ESP32 Revision → Rev  3"
1. Yes, It works and some examples build with it, like DU-1906.
2. The audio branch has a different optimization focus from the ordinary branch. This is the reason why I suggest you use the special version. In addition, you can try use the latest IDF V4.2.1 if you mind the older branch.

spaceMan00, 2021-06-23:
Hello @HengYongChao 
Thank you for following up but moving to IDF v4.2.1 did not help. 

We still have the lag problem. 

To repeat the issue: 
I am facing this problem when using the pipeline_hfp_stream example.
I connect to HFP through my iPhone and I use the Voice Memo app to record from the Lyrat.
First time I start a recording everything is fine.
If I stop and start a recording on iPhone a second time we start noticing  a lag.
If I do it for the 3rd time the lag becomes unbearable. 
And then Lyrat crashes and resets. 

These are the steps we followed:
1. Installed ESP-IDF version 4.2.1 from the vs code idf extension.
2. Made sure all submodules are up to date and all necessary libraries are installed (install.bat).
3. Applied the $ADF_PATH/idf_patches/idf_v3.3_freertos.patch to esp-idf
4. In pipeline_hfp_stream example, moved lines 38 and 39 (i2s_stream_set_clk) to lines 110-111. After this step, the example no longer gives "SCO xmit Q overflow, pkt dropped" error when recording for the third time.
5. Even switching to rev3 as suggested did not solve the issue. The behavior is the same.

You can check my log file [here]([[LOGS_HFP.txt](https://github.com/espressif/esp-adf/files/6699652/LOGS_HFP.txt)](url))