espressif/esp-adf #614: AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.

DarekLin, 2021-05-09:
Hi, 
I have a small problem with VOIP example.  During the SIP connection  can't establish the audio stream.

```
I (73829) AUDIO_ELEMENT: [raw-0x3f81240c] Element task created
I (73839) AUDIO_THREAD: The sip_dec task allocate stack on external memory
E (73849) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first

E (73869) AUDIO_THREAD: Error creating RestrictedPinnedToCore sip_dec
E (73869) AUDIO_ELEMENT: [sip_dec] audio_thread_create failed
```
But when I try to run patch I received message:
```
C:\esp\esp-idf>git apply c:/esp/esp-adf/idf_patches/idf_v3.3_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:756
error: components/freertos/tasks.c: patch does not apply
```

```
C:\esp\esp-adf>git log
commit 9d62c41742ea3ab94ac7552ed765990751db2d09 (HEAD -> master, origin/master, origin/HEAD)
Merge: 9525ede 4003e0e
Author: Jason-mao <maojianxin@espressif.com>
Date:   Thu Apr 29 03:54:26 2021 +0000

C:\esp\esp-idf>git log
commit 21ecef5c56e01d3cccf0648c869e3fa9248bf3d4 (HEAD, tag: v4.2.1)
Author: He Yin Ling <heyinling@espressif.com>
Date:   Thu Apr 29 10:13:07 2021 +0800

    versions: Update version to 4.2.1

```
LyraT board 4.3

DarekLin, 2021-05-09:
Meanwhile, I found a solution of the problem.
The problem can be solved via add two parameters to git apply command

C:\esp\esp-idf>git apply --ignore-space-change --ignore-whitespace c:/esp/esp-adf/idf_patches/idf_v3.3_freertos.patch

closed 2021-09-26 by jason-mao

Vxs14, 2022-01-27:
Hi,

with master branch and idf 4.4, impossible to patch !

C:\esp\esp-idf>git apply --ignore-space-change --ignore-whitespace %ADF_PATH%/idf_patches/idf_v3.3_freertos.patch
error: patch failed: components/freertos/include/freertos/task.h:674
error: components/freertos/include/freertos/task.h: patch does not apply
error: patch failed: components/freertos/tasks.c:3899
error: components/freertos/tasks.c: patch does not apply

Vxs14, 2022-01-27:
On the documentation you specify that ESP-ADF provides support for specific ESP-IDF versions (v3.3, v4.0, v4.1, v4.2, v4.3 and v4.4). 
but on some posts you recommend the v4.2.

which is the correct version?

Vxs14, 2022-01-28:
sorry i didn't see the idf_v4.4_freertos.patch 

but the error message when running IDF 4.4 recommended to use the idf_v3.3_freertos.patch.