espressif/esp-adf #616: Can ESP-ADF play multiple MP3/WAV files simultaneously?

Theagainmen, 2021-05-10:
I've looked at the GitHub docs and the [espressif docs](https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html ), however I couldn't find an answer to my question.

As the title says: is it possible with the ESP32 (Lyrat board for example) to play multiple audio files simultaneously?

I have a game running on the ESP32 where I need to run some sound effects from that can sometimes be ~10 (short) sounds playing at the same time. If this is not possible, could anyone point me into the right direction? 
I've found something else called the [WAV Trigger Board](https://www.sparkfun.com/products/13660), that can play up to 14 files at the same time, however this it too expensive to use for my project.

Any help would be welcome, thanks in advance 😄.

Theagainmen, 2021-05-12:
Hello @HengYongChao, thanks for your response again. That sounds very promising! I will buy an ESP32 Lyrat module and get my head around it, you will probably hear more from me in a few weeks when the board is arrived and I've played with it.

Since this will probably be very hard for a new comer trying this out, and I have to wait for a few weeks, and you probably know every detail, I will take my shots not trying to be rude, but can you maybe show an example of what playing 2+ .wav files looks like (and maybe even sounds like 😄 )? Only if you have the time available of course.

Theagainmen, 2021-05-19:
Any updates @HengYongChao ?

HengYongChao, 2021-05-20:
Hi @Theagainmen 

1. replace the mp3 decder, use wav decoder.
2. add others input wav audio to downmix pipeline.

robert-alfaro, 2021-07-13:
downmixer supports up to 8 if youd like.   There is a #define to control it.  You will need psram and probably lower sampling rate like 12 or 16kHz.  Also ensure no bottleneck buffering the audio samples from sdcard or whichever memory type you're using.

closed 2023-09-05 by jason-mao