espressif/esp-adf #618: MCLK output by GPIO 0 not always desired

ropg, 2021-05-13:
`components/audio_stream/i2s_stream.c` hardcodes:

```c
i2s_mclk_gpio_select(i2s->config.i2s_port, GPIO_NUM_0);
```

Some boards have no use for this signal or, worse, use this pin for something that does not like to get MCLK there instead. (Such is the case on the M5Stack Core device, see [here](https://github.com/ropg/m5core2_adf#gpio_0-used-for-mclk-output) for details.

> Fortunately the `i2s_mclk_gpio_select` lives in the audio board code, so a custom board driver can replace this with a dummy function. Still weird to hardcode it like this.

(See also #617)

HengYongChao, 2021-06-03:
Hi @ropg 

Thanks for your advice!

felixcollins, 2022-08-26:
I ran into this too... https://www.esp32.com/viewtopic.php?f=20&t=29167&sid=88cdb8efb3afca9974eddca2dca965c0

It seems to me that the audio board layer should be built on top of the adf. Therefore i2s_stream should have no dependency on the audio board driver. My fix for this was to just duplicate the i2s_stream code, fix it and rename the public function  so it does not clash. I'll rip it out when this bug is fixed.

felixcollins, 2022-08-26:
A backward compatible fix would be to add a function like this.
audio_element_handle_t i2s_stream_init_with_pins(i2s_stream_cfg_t *config, i2s_pin_config_t* i2s_pin_cfg)

felixcollins, 2023-06-06:
Hello ADF Team? Is this ticket ever going to be assigned? If it is never getting fixed shall we close it?

jason-mao, 2023-09-05:
It fixed on 57282dde2100ca98f4e48ee3daf8daa7d3406713

closed 2023-09-05 by jason-mao